srsran/srsRAN_Project #709: Scheduling strategies for gNB

RogerLin88124, 2024-07-06:
## Issue Description ##
[image]
I have found the following files related to RR (Round Robin) and PF (Proportional Fair) scheduling in the path /srsRAN_Project/lib/scheduler/policy/:

scheduler_policy.h
scheduler_policy_factory.cpp
scheduler_policy_factory.h
scheduler_time_pf.cpp
scheduler_time_pf.h
scheduler_time_rr.cpp
scheduler_time_rr.h
ue_allocator.h

Could you please provide guidance on how to use these scheduling strategies ? 

## Setup Details ##
I am following the instructions from [this documentation](https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html) to construct and set up my simulation environment.

## Expected Behavior ##
I expect to be able to add the scheduler type in the gnb_zmq.yaml file as follows: scheduler: type="pf" or scheduler: type="rr".

herlesupreeth, 2024-07-06:
By default Round-Robin scheduler is being used in gNB. But if you want to use the proportional fair scheduler then you need to add the following to your gNB configuration file

```
cell_cfg:
  sched_expert_cfg:
    policy_sched_cfg:
      pf_sched:
        pf_sched_fairness_coeff: 2.0
```

RogerLin88124, 2024-07-06:
@herlesupreeth 
Thank you very much for your help.

I have one more question. If I want to add another scheduling strategy, such as WF, where should I create the file to allow the gNB to obtain the parameters of the new strategy?

herlesupreeth, 2024-07-07:
If you want to add another scheduling strategy you would need to implement the interface specified in `scheduler_policy.h`. For start, I would suggest initializing the parameters of the new scheduling strategy in its constructor

RogerLin88124, 2024-07-07:
@herlesupreeth I got it .   Thank you

closed 2024-07-07 by RogerLin88124