srsran/srsRAN_Project #712: Server Crash Issue with srsRAN 24.04 and LiteON RU Integration

Xinya25, 2024-07-07:
## Issue Description ##
I have encountered a critical issue while executing the gNB after installing srsRAN. I installed the latest version of srsRAN 24.04 according to [this documentation](https://docs.srsran.com/projects/project/en/latest/user_manuals/source/installation.html) (Installation Guide). Additionally, I followed the instructions provided in the [srsRAN documentation](https://docs.srsran.com/projects/project/en/latest/tutorials/source/oranRU/source/rus/liteon.html#initializing-the-network) (O-RAN 7.2 RU Guide - LITEON FlexFi) for this process.

## Setup Details ##
1. I have confirmed that the LiteON RU is started and has PTP lock.
[image]

2. However, when I execute the command `sudo ./gnb -c gnb_ru_liteon_tdd_n78_100mhz.yml`, **it causes the entire server to crash.**
- Error logs from our attempts
[image]

## Expected Behavior ##
The correct log is below
[image]

## Additional Information ##
- The model of the server we are using
[image]
[image]

- Our current gNB configuration file
[gnb_ru_liteon_tdd_n78_100mhz.yml](https://mail.google.com/mail/u/0?ui=2&ik=128fbdbfdd&attid=0.1&permmsgid=msg-a:r-284172473641659113&view=att&disp=safe&realattid=f_ly77d3bd6)
 
- I also run the [srsran_performance](https://github.com/srsran/srsRAN_Project/blob/main/scripts/srsran_performance) first.
[image]

- I would like to know the cause of the entire server crash and how to resolve this issue. Thank you for your help!

andrepuschmann, 2024-07-07:
I don't see a crash/segfault. Please post full console output. Also note that you need a new firmware of the RU to make it work.

pgawlowicz, 2024-08-08:
@Xinya25 any update on this issue?

Xinya25, 2024-08-08:
@pgawlowicz After we replaced the server hardware and changed the OS to Ubuntu 22.04, the srsRAN gNB was able to run successfully. Therefore, we have resolved that issue. Thank you!

closed 2024-08-08 by pgawlowicz

AdityaKoranga, 2024-10-10:
> ## Issue Description
> I have encountered a critical issue while executing the gNB after installing srsRAN. I installed the latest version of srsRAN 24.04 according to [this documentation](https://docs.srsran.com/projects/project/en/latest/user_manuals/source/installation.html) (Installation Guide). Additionally, I followed the instructions provided in the [srsRAN documentation](https://docs.srsran.com/projects/project/en/latest/tutorials/source/oranRU/source/rus/liteon.html#initializing-the-network) (O-RAN 7.2 RU Guide - LITEON FlexFi) for this process.
> 
> ## Setup Details
> 1. I have confirmed that the LiteON RU is started and has PTP lock.
> 
> <img alt="image" width="300" src="https://private-user-images.githubusercontent.com/80584940/346345797-7862314b-19a7-46d1-a0d5-84d2ce7c5af1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg1NjI2MjEsIm5iZiI6MTcyODU2MjMyMSwicGF0aCI6Ii84MDU4NDk0MC8zNDYzNDU3OTctNzg2MjMxNGItMTlhNy00NmQxLWEwZDUtODRkMmNlN2M1YWYxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDEwVDEyMTIwMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFhZmQ5OGRmY2JiYWY1MTUxZTU0NjNkNmIyOTMxNjkwODRhZTEwODAyMjZkZDA1MGRmNDdmYWIwN2Y5MWJiYzcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.yypGrqMbeyzopJDf1ssYW2SVAk5R0FSr8NJtonZJ2gw">
> 2. However, when I execute the command `sudo ./gnb -c gnb_ru_liteon_tdd_n78_100mhz.yml`, **it causes the entire server to crash.**
> 
> * Error logs from our attempts
>   ![image](https://private-user-images.githubusercontent.com/80584940/346345861-ac787d00-c730-431f-aef9-41378ea8f0de.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg1NjI2MjEsIm5iZiI6MTcyODU2MjMyMSwicGF0aCI6Ii84MDU4NDk0MC8zNDYzNDU4NjEtYWM3ODdkMDAtYzczMC00MzFmLWFlZjktNDEzNzhlYThmMGRlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDEwVDEyMTIwMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBjOWMxYmYzZjlhM2ZhZGE4NDBlODRiMTBjN2Q0MWIyNDJmMjMyZmJjZTk2NDE2MzUxNjBhYzBiZTk0ZjRhYTAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.A854PsTLZnK7j1PGXglsCJKsftOU66JFNc7eL1OQQAc)
> 
> ## Expected Behavior
> The correct log is below ![image](https://private-user-images.githubusercontent.com/80584940/346345891-a39c514d-feda-4cb1-8d2d-df860b6928fc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg1NjI2MjEsIm5iZiI6MTcyODU2MjMyMSwicGF0aCI6Ii84MDU4NDk0MC8zNDYzNDU4OTEtYTM5YzUxNGQtZmVkYS00Y2IxLThkMmQtZGY4NjBiNjkyOGZjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDEwVDEyMTIwMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY2YWEyMDJhMGE1MzEwYzgwNTM0ZGVjNmRlN2MxYjM0M2Y4MzJiNjYzZTk3ZDgyZjU3ZjEwMzkwMmNiNWYxYzgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.cOPlKL9ON756m35kEYXuIhnkZMKr8bio-ueC6BC7Ejo)
> 
> ## Additional Information
> * The model of the server we are using
> 
> <img alt="image" width="400" src="https://private-user-images.githubusercontent.com/80584940/346346360-807f81e9-0152-4188-9ec4-13acc23e72cb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg1NjI2MjEsIm5iZiI6MTcyODU2MjMyMSwicGF0aCI6Ii84MDU4NDk0MC8zNDYzNDYzNjAtODA3ZjgxZTktMDE1Mi00MTg4LTllYzQtMTNhY2MyM2U3MmNiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDEwVDEyMTIwMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZiYWZhYTllNzI3M2EyNmM1MWExMDE5YjhlZTBiYTFlYmY5ODRlMTAwOTRiNGYxZTYyYmI4OTUyYmM1ODY3ZTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.N9LvpxdEIk0_QW4kyUA3ZP9o-m_PipsO8yqFhi9uqwk"> <img alt="image" width="400" src="https://private-user-images.githubusercontent.com/80584940/346346370-12f56cd9-42ae-485d-95a9-37888b5c930c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg1NjI2MjEsIm5iZiI6MTcyODU2MjMyMSwicGF0aCI6Ii84MDU4NDk0MC8zNDYzNDYzNzAtMTJmNTZjZDktNDJhZS00ODVkLTk1YTktMzc4ODhiNWM5MzBjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDEwVDEyMTIwMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA1MmFjMjQzOTRlYzVjYWYyMzM3NmRkZjA0NTM5YmU3YmYwMDBlZGRmY2YzZmMyYTk2ZGViYTIwZTkxZDM1YzEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.-ZtE6QpF8oy3_IRiczyVB6v_P696B0Q8Itn6KKnzEK0">
> * Our current gNB configuration file
>   [gnb_ru_liteon_tdd_n78_100mhz.yml](https://mail.google.com/mail/u/0?ui=2&ik=128fbdbfdd&attid=0.1&permmsgid=msg-a:r-284172473641659113&view=att&disp=safe&realattid=f_ly77d3bd6)
> * I also run the [srsran_performance](https://github.com/srsran/srsRAN_Project/blob/main/scripts/srsran_performance) first.
> 
[... cut ...] 2lnbmVkSGVhZGVycz1ob3N0In0.60F4_zgyXeHfaxbtKjknxFqaJGNHPjTZJSbnUfNWP-g">
> * I would like to know the cause of the entire server crash and how to resolve this issue. Thank you for your help!

Hi @Xinya25  I am unable to download the `gnb_ru_liteon_tdd_n78_100mhz.yaml` file that you have attached in this issue.
Can you please reshare the file and also if possible could you share the output of show running-config on the RU.
That will be very helpful. Currently I am facing this issue: #848 

Thanks