srsran/srsRAN_Project #721: SRSUE stuck with attaching UE

iamniki01, 2024-07-10:
## Issue Description ##
I'm following the recent tutorial 'srsRAN gNB with srsUE' also using DPDK to avoid packet loss from here https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html. The UE is not attaching after many tries and I have no clue where I am going wrong, and when I check the logs of UE (attached as an UE.log) got CRC Failed warning and loops via it forever.

## Setup Details ##
Using 2 USRP of x410 as gNB AND SRSue with high compatible machines AMD with 16 cores

## Expected Behavior ##
srsue connecting to the gNB.

## Actual Behaviour ##
[image]
[image]

## Steps to reproduce the problem ##
Here I attached the config files of both gnb and Ue. Also its Logs and its pcaps.

## Additional Information ##
NOTE: I created the issue at srsran_4g. then I noticed people are more active in this repository so I am moving this here.
https://github.com/srsran/srsRAN_4G/issues/1358, one can obsolete either of this. 

Any insights and assistance will be appreciated. I am looking forward for your quick response.

P.S: Using both srsran and srsue with latest release (and updates)
[Logs_Pcaps.zip](https://github.com/user-attachments/files/16158698/Logs_Pcaps.zip)

pgawlowicz, 2024-07-11:
could you try to add the following parameters to the srsue:
```
[expert]
lte_sample_rates = true

```

see here: https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#usrp-x300-x310

iamniki01, 2024-07-12:
> could you try to add the following parameters to the srsue:
> 
> ```
> [expert]
> lte_sample_rates = true
> ```
> 
> see here: https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#usrp-x300-x310

Hi Thanks, It passed the stage atleast, thanks for an eye opener. However, it is not connecting to the AMF and not getting its IP too. 
1. UE connectivity 
[image]
 
2. This amf logs shows that only about gnb, ue is not connecting.
[image]

pgawlowicz, 2024-07-12:
Could you put gnb logs and ngap pcap?

iamniki01, 2024-07-12:
> Could you put gnb logs and ngap pcap?

That was quick, Thanks:) Here is the complete logs and pcap of both gnb and ue.
[logs_pcaps.zip](https://github.com/user-attachments/files/16189871/logs_pcaps.zip)

pgawlowicz, 2024-07-16:
could you post the gnb and srsue logs?

iamniki01, 2024-07-16:
> could you post the gnb and srsue logs?

Hi again,

I played around with `freq_offset` and `tx & Rx gain`, and it worked. I started receiving the IP and can now see the traffic between the gNB and UE without network breaks. However, if I run the UE five times, it might fail once with an RRC release after a successful connection of about five seconds. I realize that the cause of the break is if the UE gets the `Late` status. If you know how to solve this minor issue, that would be helpful; if not, this issue can be closed now. Thanks for the great help!

closed 2024-08-08 by pgawlowicz