srsran/srsRAN_Project #723: srsRAN ERROR: Failed to connect DU to CU-CP

Moussa-Guemdani, 2024-07-10:
## Issue Description ##
After setting up and running the CU, the DU fails to connect to CU-CP 

## Setup Details ##
Followed the tutorial in "https://docs.srsran.com/projects/project/en/latest/tutorials/source/cu_du_split/source/index.html#"

## Actual Behaviour ##
[image]

pgawlowicz, 2024-07-11:
Hi @Moussa-Guemdani, this discussion seems to be related could you check it?
https://github.com/srsran/srsRAN_Project/discussions/707

pgawlowicz, 2024-07-11:
did you create a new virtual interface with the requested IP address?

Moussa-Guemdani, 2024-07-11:
No, I haven't created a virtual interface for the address 127.0.10.1 or any other address, Could please explain further?

pgawlowicz, 2024-07-11:
you can try:

sudo ip addr add 127.0.10.1/8 dev lo
sudo ip addr add 127.0.10.2/8 dev lo
ip addr show lo

Moussa-Guemdani, 2024-07-12:
Yse,
Still having the same issue
[image]

Moussa-Guemdani, 2024-07-12:
Hello @pgawlowicz ,

I have rebuilt and used the minimal configuration with different IP addresses, the "amf: bind_addr:" and the "cu_cp: bind_addr:", and it works very well,

[CU_TEST.txt](https://github.com/user-attachments/files/16195692/CU_TEST.txt)[DU_TEST.txt](https://github.com/user-attachments/files/16195702/DU_TEST.txt)

I will close this issue,
Thank you for your help.

closed 2024-07-12 by Moussa-Guemdani