srsran/srsRAN_Project #725: srsRAN gNB Handover

apperezG, 2024-07-11:
## Issue Description ##
Good morning! I am trying out the tutorial https://docs.srsran.com/projects/project/en/latest/tutorials/source/handover/source/index.html#hardware-and-software-overview. My setup is very similar to the one given in the tutorial except that I am working with the srsUE. My problem is that the UE does not connect to the gNB but I do not see any errors or strange behavior in the gNB log file.

The gNB deploys without problem with the two cells and I don't get any errors, either on the terminal or in the logs. On the part of the srsrUE I am running it in the same way as in another deployment I have with a gNB with only 1 cell in a B210 and in that setup it works without problem. I attach the configuration I am applying in case you see something that does not make sense to you. 

Thank you very much!

## Setup Details ##
gNB configuration (I tested both the n3 and n50 bands -> I can't use the n78 due to frequency limitation):

            gnb_id: 411
            gnb_id_bit_length: 32
            
            amf:
              addr: 10.100.0.9                  
              bind_addr: 10.100.0.8          
            
            ru_sdr:
              device_driver: uhd
              device_args: type=x300
              tx_gain: 20
              rx_gain: 15
              srate: 23.04
              lo_offset: 45
              clock: internal
              sync: internal
              time_alignment_calibration: 0
            
            cell_cfg:
              dl_arfcn: 294900 #368500                                              # ARFCN of the downlink carrier (center frequency).
              band: 50 #3                                                      # The NR band.
              channel_bandwidth_MHz: 20                                       # Bandwith in MHz. Number of PRBs will be automatically derived.
              common_scs: 30 #15                                                   # Subcarrier spacing in kHz used for data.
              plmn: "00101"                                                   # PLMN broadcasted by the gNB.
              tac: 7                                                          # Tracking area code (needs to match the core configuration).
              nof_antennas_dl: 1
              nof_antennas_ul: 1
            
            cells:
              -
                pci: 1
                prach:
                  prach_root_sequence_index: 0
              -
                pci: 2
                prach:
                  prach_root_sequence_index: 64
            
            cu_cp:
              mobility:
                trigger_handover_from_measurements: true
                cells:
                  - nr_cell_id: 0x19B0
                    periodic_report_cfg_id: 1
                    ncells:
                      - nr_cell_id: 0x19B1
                        report_configs: [ 2 ]
                  - nr_cell_id: 0x19B1
                    periodic_report_cfg_id: 1
                    ncells:
                      - nr_cell_id: 0x19B0
                        report_configs: [ 2 ]
                report_configs:
                  - report_cfg_id: 1
                    report_type: periodical
                    report_interval_ms: 480
                  - report_cfg_id: 2
                    report_type: event_triggered
                    a3_report_type: rsrp
                    a3_offset_db: 3
                    a3_hysteresis_db: 0
                    a3_time_to_trigger_ms: 100
            
            log:
              filename: ./gnb.log            # Path of the log file.
              all_level: debug
            
            pcap:
              mac_enable: true
              mac_filename: /tmp/gnb_mac.pcap
              ngap_enable: true
              ngap_filename: /tmp/gnb_ngap.pcap
              f1ap_enable: true
              f1ap_filename: /tmp/gnb_f1ap.pcap
              gtpu_enable: true
              gtpu_filename: /tmp/gnb_gtpu.pcap
              e1ap_enable: true
              e1ap_filename: /tmp/gnb_e1ap.pcap

UE configuration (I tested both the n3 and n50 bands):

          [rf]
          freq_offset = 0
          tx_gain = 30
          rx_gain = 20
          srate = 23.04e6
          nof_antennas = 1
          
          device_name = uhd
          device_args = type=b200, serial=31D4A7B
          #,num_recv_frames=64,num_send_frames=64
          time_adv_nsamples = 300
          
          [rat.eutra]
          dl_earfcn = 2850
          nof_carriers = 0
          
          [rat.nr]
          bands = 50 #3 
          nof_carriers = 1
          max_nof_prb = 106
          nof_prb = 106
          
          [pcap]
          enable = none
          mac_filename = /tmp/ue_mac.pcap
          mac_nr_filename = /tmp/ue_mac_nr.pcap
          nas_filename = /tmp/ue_nas.pcap
          
          [log]
          all_level = info
          phy_lib_level = none
          all_hex_limit = 32
          filename = /tmp/ue.log
          file_max_size = -1
          
          [usim]
          mode = soft
          algo = milenage
          opc  = 63BFA50EE6523365FF14C1F45F88737D
          k    = 00112233445566778899aabbccddeeff
          imsi = 001010123456780
          imei = 353490069873319
          
          [rrc]
          release = 15
          ue_category = 4
          
          [nas]
          apn = internet
          apn_protocol = ipv4
          
          [gw]
          # netns = ue1
          # ip_devname = tun_srsue
          # ip_netmask = 255.255.255.0
          
          [gui]
          enable = false

## Expected Behavior ##
UE conects with the gNB.

## Actual Behaviour ##
The UE does not connect to the gNB but I do not see any errors.

pgawlowicz, 2024-07-11:
Hi,
Band 50 uses TDD, srsUE supports only FDD.

apperezG, 2024-07-12:
I tested both the n3 and n50 bands (the n3 with the values commented, which is FDD) and the result is the same in both cases.

apperezG, 2024-07-12:
I attached the logs of the gNB and the UE (in n3 band) in case they are helpful:
[gnb.log](https://github.com/user-attachments/files/16191801/gnb.log)
[ue.log](https://github.com/user-attachments/files/16191807/ue.log)

pgawlowicz, 2024-07-12:
could you increase UE tx gain? the gnb did not receive any PRACH

apperezG, 2024-07-16:
By adjusting the gain value so that I can see the tx of the gNB on the B210 that has the srsUE I get the following:

**gNB**
[image]

**UE**
[image]

Logs: 
[logs.zip](https://github.com/user-attachments/files/16253312/logs.zip)

I already ran the srsran_performance script as well just in case. Besides the gain can you think of anything else to try changing in the configuration file @pgawlowicz ?

Thank you!

pgawlowicz, 2024-07-16:
CFO seems to be high, do you use external clock source?

apperezG, 2024-07-16:
Not in this case. Do you recommend me to set a specific value or add an external sync generating a 1 Hz signal that will serve the x310 as a reference for the clock?

pgawlowicz, 2024-07-16:
you should connect both to a single clock source, if it works then you can try if connecting only b200 (srsUE) is enough

pgawlowicz, 2024-07-17:
you also need to connect 10MHz clock

apperezG, 2024-07-18:
Hi! I tried putting in an octoclock that would generate a 1 PPS signal and a 10 MHz clock signal for each USRP. I no longer have the problem of the UE giving me sync failure but, with the same configuration as before I only get Overflows and the UE does not connect. Attached are the logs --> 
[logs.zip](https://github.com/user-attachments/files/16282016/logs.zip)

I have the UE and the gNB on two (quite capacious) different servers and I already ran the srsran_performance.sh just in case it was processing problems but I don't think it could be from that.

pgawlowicz, 2024-07-18:
could you share your current configs? could you try to run 5MHz cell and set srate to 5.76 in both gnb and srsue configs?

pgawlowicz, 2024-08-08:
@apperezG any update on this issue?

apperezG, 2024-08-26:
@pgawlowicz I couldn't go any further with this issue at the moment, if I make any progress I'll let you know!

closed 2024-10-16 by pgawlowicz