srsran/srsRAN_Project #730: Test mode max number of ues can't exceed 160

aibtw, 2024-07-14:
## Issue Description ##
According to configuration reference, up to 1024 UEs is supported in test mode. I can only see 96 ues on band78 and 160 ues on band71.

## Setup Details ##
Commit: [4cf7513](https://github.com/srsran/srsRAN_Project/commit/4cf7513e9fe988449aadedce3278063de761a4ac) (also tried the release version 24.04)
Tried on different servers with different CPUs including i7-13700k-32GB RAM.

## Expected Behavior ##
1024 UEs created

## Actual Behaviour ##
Only 160 UEs created (N71) or 96 (N78)
When I open log files it has the following message repeated for each of the missing ues:
`2024-07-14T12:08:00.171035 [DU-MNG  ] [W] ue=160 rnti=0x46a1 proc="UE Create": Procedure failed. Cause: Failed to create DU UE context. Cause: Unable to allocate dedicated PUCCH resources for cell=0.` 

[gnb.txt](https://github.com/user-attachments/files/16219320/gnb.txt)

## Steps to reproduce the problem ##
the config file shared below
```
amf:
  addr: 127.0.1.100                                             # The address or hostname of the AMF.
  bind_addr: 127.0.0.1                                          # A local IP that the gNB binds to for traffic from the AMF.
  no_core: true
  
cell_cfg:
  # dl_arfcn: 640000                                              # ARFCN of the downlink carrier (center frequency).
  # band: 78                                                      # The NR band.
  # channel_bandwidth_MHz: 20                                     # Bandwith in MHz. Number of PRBs will be automatically derived.
  # common_scs: 30                                                # Subcarrier spacing in kHz used for data.
  dl_arfcn: 126900 # n71
  band: 71                                                      # The NR band.
  channel_bandwidth_MHz: 10                                     # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 15                                                # Subcarrier spacing in kHz used for data.
  plmn: "99970"                                                 # PLMN broadcasted by the gNB.
  tac: 7                                                        # Tracking area code (needs to match the core configuration).
  pci: 1                                                        # Physical cell ID.
  nof_antennas_dl: 1
  nof_antennas_ul: 1
    
test_mode: 
 test_ue: 
   rnti: 0x4601                               # Optional ENUM (0). Sets the C-RNTI of the UE. Supported: [0 - 65519].
   nof_ues: 1000                            # Optonal UINT (1). Sets the number of test UE(s) created. Supported [0 - 1024].
   pdsch_active: true                    # Optional BOOLEAN (true). Enables the PDSCH of the UE. Supported: [false, true].
   pusch_active: true                    # Optional BOOLEAN (true). Enables the PUSCH of the UE. Supported: [false, true].
   cqi: 15                               # Optional UINT (15). Sets the Channel Quality Information to be forwarded to the test UE. Supported: [1 - 15].
   pmi: 0                                # Optional UINT (0). Sets the Precoder Matrix Indicator to be forwarded to test UE. Supported: [0 - 3].
   ri: 1                                 # Optional UINT (1). Sets the Rank Indicator to be forwarded to the test UE. Supported: [1 - 4].
   i_1_1: 0                              # Optional INT (0). Sets the Precoder Matrix codebook index "i_1_1" to be forwarded to test UE, in the case of more than 2 antennas. Supported: [0 - 7].
   i_1_3: 0                              # Optional INT (0). Sets the Precoder Matrix codebook index "i_1_3" to be forwarded to test UE, in the case of more than 2 antennas. Supported: [0 - 1].
   i_2: 0                                # Optional INT (0). Sets the Precoder Matrix codebook index "i_2" to be forwarded to test UE, in the case of more than 2 antennas. Supported: [0 - 3].

log: 
 all_level: info                    # Optional TEXT (warning). Sets a common log level across PHY, MAC, RLC, PDCP, RRC, SDAP, NGAP and GTPU layers

metrics:
  stdout_metrics_period: 1000 
  enable_json_metrics: true

```

herlesupreeth, 2024-07-15:
You should use the configuration `cell_cfg_max_128_ues.yml` under configs folder in addition to above gNB configuration file so that maximum number of UEs are able to attach.

aibtw, 2024-07-16:
Thanks for the clarification. I can now get up to 820 UEs in test mode.

closed 2024-07-16 by aibtw