srsran/srsRAN_Project #731: Enable tracing mode in docker setup?

dominikheinz, 2024-07-15:
## Issue Description ##
I am unsure about the method to enable tracing mode for gNB in order to view the usual table of statistics when UEs are connected. In a normal installation, tracing mode can be activated by typing `t` after starting gNB. However, I could not figure out how to do this in a Docker setup. Is there a command line flag available for this? I couldn't find any.
It is mainly this part of the config that I hope to modify somehow:

```
command: gnb -c /gnb_config.yml log --all_level debug amf --addr ${OPEN5GS_IP:-10.53.1.2} --bind_addr ${GNB_IP:-10.53.1.3}
```

## Setup Details ##
Run srsRAN via the provided docker-compose file with some minor customizations.

`docker-compose -f myconfig.yml up`

## Expected Behavior ##
Maybe a way to automatically enable tracing with a commandline flag?

## Actual Behaviour ##
I can not enable tracing mode

## Steps to reproduce the problem ##
1. Run `docker-compose -f myconfig.yml up`
2. Realize that typing `t` in the docker logs (obviously) has no effect.

## Additional Information ##

**myconfig.yml**
```yaml
services:
  5gc:
    container_name: open5gs_5gc
    build:
      context: open5gs
      target: open5gs
      args:
        OS_VERSION: "22.04"
        OPEN5GS_VERSION: "v2.7.0"
    env_file:
      - ${OPEN_5GS_ENV_FILE:-open5gs/open5gs.env}
    privileged: true
    ports:
      - "9999:9999/tcp"
    command: 5gc -c open5gs-5gc.yml
    healthcheck:
      test: [ "CMD-SHELL", "nc -z 127.0.0.20 7777" ]
      interval: 3s
      timeout: 1s
      retries: 60
    networks:
      ran:
        ipv4_address: ${OPEN5GS_IP:-10.53.1.2}

  gnb:
    container_name: srsran_gnb
    image: srsran/gnb
    build:
      context: ..
      dockerfile: docker/Dockerfile
      args:
        OS_VERSION: "24.04"
    privileged: true
    cap_add:
      - SYS_NICE
      - CAP_SYS_PTRACE
    volumes:
      - /dev/bus/usb/:/dev/bus/usb/
      - /usr/share/uhd/images:/usr/share/uhd/images
      - gnb-storage:/tmp
      - ./gnb_custom.yml:/gnb_config.yml:ro
    networks:
      ran:
        ipv4_address: ${GNB_IP:-10.53.1.3}
      metrics:
        ipv4_address: 172.19.1.3
    depends_on:
      5gc:
        condition: service_healthy
    command: gnb -c /gnb_config.yml log --all_level debug amf --addr ${OPEN5GS_IP:-10.53.1.2} --bind_addr ${GNB_IP:-10.53.1.3}

  metrics-server:
    container_name: metrics_server
    image: srsran/metrics_server
    build:
      context: metrics_server
    environment:
      - PORT=${METRICS_SERVER_PORT}
      - BUCKET=${DOCKER_INFLUXDB_INIT_BUCKET}
      - TESTBED=default
      - URL=http://${DOCKER_INFLUXDB_INIT_HOST}:${DOCKER_INFLUXDB_INIT_PORT}
      - ORG=${DOCKER_INFLUXDB_INIT_ORG}
      - TOKEN=${DOCKER_INFLUXDB_INIT_ADMIN_TOKEN}
    ports:
      - 55555:${METRICS_SERVER_PORT}/udp
    networks:
      metrics:
        ipv4_address: 172.19.1.4

  influxdb:
    container_name: influxdb
    image: influxdb:${DOCKER_INFLUXDB_VERSION}
    volumes:
      - influxdb-storage:/var/lib/influxdb2:rw
    env_file:
      - .env
    restart: on-failure:10
    networks:
      metrics:
        ipv4_address: 172.19.1.5

  grafana:
    container_name: grafana
    image: srsran/grafana
    build:
      context: grafana
    volumes:
      - grafana-storage:/var/lib/grafana:rw
    env_file:
      - .env
    depends_on:
      - influxdb
      - metrics-server
    ports:
      - 3300:${GRAFANA_PORT}
    networks:
      metrics:
        ipv4_address: 172.19.1.6

volumes:
  gnb-storage:
  grafana-storage:
  influxdb-storage:
  gnb-config:

networks:
  ran:
    ipam:
      driver: default
      config:
        - subnet: 10.53.1.0/24
  metrics:
    ipam:
      driver: default
      config:
        - subnet: 172.19.1.0/24
```

andrepuschmann, 2024-07-31:
Just enable the stdout metrics in the config file or pass via cmd line `metrics --autostart_stdout_metrics=true`

closed 2024-07-31 by andrepuschmann

dominikheinz, 2024-08-15:
@andrepuschmann 

The solution you provided works, which is good news. However, there's an unexpected behavior with the tracelog output. Instead of continuous printing, it's coming out in bursts at irregular intervals.
I've attempted to adjust this by setting the `stdout_metrics_period` to 1000 (aiming for output every second), but this change doesn't seem to have any effect. The current pattern is long periods of silence (15-20 seconds) followed by a sudden burst of trace output. This cycle then repeats.

Any idea why that is? When running a non-dockerized setup, I see the tracelog being printed every second or so, line by line.

pgawlowicz, 2024-08-15:
What is the behaviour if you run the gnb container in the deattached mode (`-d`) and connect to the container with:
```
docker logs -t -f <name-of-service>
```

dominikheinz, 2024-08-15:
> What is the behaviour if you run the gnb container in the deattached mode (`-d`) and connect to the container with:
> 
> ```
> docker logs -t -f <name-of-service>
> ```

Same problem.

pgawlowicz, 2024-08-15:
which docker version do you have?
could you try to set `export PYTHONUNBUFFERED=1` and try again?

dominikheinz, 2024-08-15:
@pgawlowicz 
Docker Compose version 2.29.1
Docker version 27.1.1, build 63125853e3

Setting `export PYTHONUNBUFFERED=1` has not effect. Problem remains.

dominikheinz, 2024-08-20:
@pgawlowicz Any other ideas what I could try?

pgawlowicz, 2024-08-20:
Unfortunately no, seems the stdout is somewhere buffered, not sure where. 
what about receiving the metric json thought the metric udp port?

herlesupreeth, 2024-08-20:
If the docker is in detached state then you can do the following

`docker container attach <container_name>`

NOTE:

you need to have the following in your docker compose file for that service (gnb:)

```
    stdin_open: true
    tty: true
```