srsran/srsRAN_Project #732: srsUE cannot connect to gNb

CrashxZ, 2024-07-16:
## Issue Description ##
I followed this tutorial - https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html

I am  trying to use the over-the-air setup to establish connection between the srsUE and gNb, but the UE gets stuck  after attaching UE, with the following message - "RF status: O=10, U=0, L=0". There are overflows and underflows on the gNb side too. 

My gNb and UE start and here are the screenshots (will also attach log files)
[image]
[image]

## Setup Details ##
PC1 - gNb and Open5GC
Lenovo Legion  5 (AMD Ryzen 5 4600h + Nvidia GTX 1660  + 8GB RAM)
UHD Version - UHD_4.1.0.5-3
Radio - B210

Here is my configuration file - 
[gnb.yaml.txt](https://github.com/user-attachments/files/16241540/gnb.yaml.txt)

PC2 - srsUE
Raspberry Pi4 8GB varient. 
UHD Version - UHD_4.1.0.5-3
Radio - B210

Here is my configuration file - 
[ue.conf.txt](https://github.com/user-attachments/files/16241687/ue.conf.txt)

I have ensured that both setups have the cpu's set to performance and have also run the performance script.

## Additional Information ##
I ran the usrp benchmarks scripts on both the pcs with the selected frequencies and had no issues there. 

Here are my log files -
[amf.log.txt](https://github.com/user-attachments/files/16241680/amf.log.txt)
[gnb.log.txt](https://github.com/user-attachments/files/16241681/gnb.log.txt)
[ue.log.txt](https://github.com/user-attachments/files/16241682/ue.log.txt)

pgawlowicz, 2024-07-16:
gnb log says:
```
2024-07-15T03:18:27.564308 [GNB     ] [W] CPU0 scaling governor is not set to performance, which may hinder performance. You can set it to performance using the "srsran_performance" script
```

pgawlowicz, 2024-07-16:
also RPi5 might be too slow for srsUE, you could try lower BW maybe it helps.

CrashxZ, 2024-07-16:
Sorry my bad, I reran the performance script and also this command - "sudo cpupower frequency-set --governor performance" , and ensured that the CPUs are set to performance mode. I also lowered the channel bandwidth on the gNb side to 5mhz (i don't know how to explicitly set it on the srsUE side). The behaviour is still the same, here are the new logs - 
[ue.log.txt](https://github.com/user-attachments/files/16255557/ue.log.txt)
[gnb.log.txt](https://github.com/user-attachments/files/16255558/gnb.log.txt)

I will also run the UE on the Nvidia Jetson TX2, since it might be more powerful, but is there any way to run the UE on the RPi 4 ?

CrashxZ, 2024-07-16:
Installed srsran4g on the Jetson TX2 and tweaked the config a bit but still i cant establish connection.   (also rerunning the performance script helped a bit, gNb overflows and underflows are less frequent)
Note that the tx2 has ubuntu 18.04 and UHD 3.15. Here are the screenshots -

UE (TX2) - 
[image]

gNb - 
[image]

Here are the config files - 
[gnb.yaml.txt](https://github.com/user-attachments/files/16256261/gnb.yaml.txt)
[mainak_ue.conf.txt](https://github.com/user-attachments/files/16256262/mainak_ue.conf.txt)

and here are the logs - 

[ue.log.txt](https://github.com/user-attachments/files/16256266/ue.log.txt)
[gnb.log.txt](https://github.com/user-attachments/files/16256267/gnb.log.txt)

pgawlowicz, 2024-07-16:
could you use `srate: 30.72 ` in gnb config?

CrashxZ, 2024-07-16:
Still the same - here are the logs. (also my Samsung S23 ultra cannot see the network when I do a network search, I understand it cant connect (without a programmed sim and a different config), but it should be able to see it right ?)

Logs - 
[ue.log.txt](https://github.com/user-attachments/files/16257346/ue.log.txt)
[gnb.log.txt](https://github.com/user-attachments/files/16257347/gnb.log.txt)

CrashxZ, 2024-07-17:
I also tried running the UE from another x86 system (Core i7 9th gen and 16gb of RAM), just to check if its a performance issue. But it was the same response. 

Then I thought it would be a Tx/Rx gain tuning issue, so i connected the 2 B210s using SMA cables, and tested it again both x86 machines. This time the out of sync message wasn't present, but instead it showed - "PBCH-MIB: CRC failed" in the UE logs.

here are the logs- 

[ue.log.txt](https://github.com/user-attachments/files/16257930/ue.log.txt)
[gnb.log.txt](https://github.com/user-attachments/files/16257931/gnb.log.txt)

pgawlowicz, 2024-07-17:
with 5MHz cell, could you use `srate: 5.76` in gnb conifg? could you also increase log level to debug in the gnb config?

CrashxZ, 2024-07-17:
@pgawlowicz , thanks a lot for taking the time to help me out - 
Here are the logs with the suggested changes. I still cant establish connection with the b210s connected via sma cables (on x86 machines)

[ue.log.txt](https://github.com/user-attachments/files/16273654/ue.log.txt)
[gnb.log.txt](https://github.com/user-attachments/files/16273656/gnb.log.txt)

pgawlowicz, 2024-07-18:
please also change srate: 5.76 in ue config. Seems it is too slow to process signal samples with 23.04MHz

CrashxZ, 2024-07-18:
That makes sense, I changed both srate to 5.76 , now it shows RRC connected, but then it is released right after - 
[image]

The ip address is not assigned, there is nothing on the amf log too. Also should there be a network adapter created on the UE machine? where the ipaddress is assigned ? because i dont see any adapter that was created, when i use the command ifconfig.

here are the logs - 
[gnb.log.txt](https://github.com/user-attachments/files/16281263/gnb.log.txt)
[ue.log.txt](https://github.com/user-attachments/files/16281357/ue.log.txt)

I had to delete some lines from the ue log , it was 89 mb, hopefully, this will be or i'll provide a google drive link.

pgawlowicz, 2024-07-18:
if ue is not getting an IP assigned, an interface will not be created. 
Could you try to apply this patch and send logs again?
[0002-nr-rrc-fix-apply-t_offset-only-once-after-SIB1.patch.txt](https://github.com/user-attachments/files/16282309/0002-nr-rrc-fix-apply-t_offset-only-once-after-SIB1.patch.txt)

CrashxZ, 2024-07-18:
Ohh, I understand, here are the new logs with the patch - 

[gnb.log.txt](https://github.com/user-attachments/files/16288283/gnb.log.txt)
[ue.log.txt](https://github.com/user-attachments/files/16288284/ue.log.txt)

pgawlowicz, 2024-07-18:
Hi, are you sure the patch is applied correctly, srsUE commit is the same in both cases (i.e., commit ec29b0c1f) but maybe because the changes from patch are not commited.

CrashxZ, 2024-07-18:
Hi I did apply it locally, but now I forked it and applied the patch (ensured the changes) and here is the new log file - 
[gnb.log.txt](https://github.com/user-attachments/files/16289792/gnb.log.txt)
[ue.log.txt](https://github.com/user-attachments/files/16289793/ue.log.txt)

pgawlowicz, 2024-07-19:
hmm, PUSCH SNR is very small:
```
2024-07-18T22:36:24.696728 [PHY     ] [D] [  1010.6] PUSCH: rnti=0x4601 h_id=0 prb=[3, 6) symb=[0, 14) mod=QPSK rv=0 tbs=11 crc=KO iter=6.0 sinr=-30.3dB t=306.8us uci_t=0.0us ret_t=0.0us
2024-07-18T22:36:24.706472 [PHY     ] [D] [  1011.6] PUSCH: rnti=0x4601 h_id=0 prb=[3, 6) symb=[0, 14) mod=QPSK rv=0 tbs=11 crc=KO iter=6.0 sinr=-30.7dB t=148.4us uci_t=0.0us ret_t=0.0u
```

could you try to tune tx/rx gains on both gnb and ue?
you could try the following config in both:

```
  tx_gain: 60
  rx_gain: 30
```

CrashxZ, 2024-07-19:
Thanks for that , I tuned the tx/rx gains for both the ue and the gnb (tried a couple of configs).
Whats seems to be working is - 

[gnb]
tx_gain: 80
rx_gain:50
[ue]
tx_gain: 50
rx_gain:70

Now the UE connects for a file and then it fails the scheduling request - "Scheduling request failed: releasing RRC connection..."

The AMF log aslo shows that it recieves the request. Here are the logs - 

[amf.log.txt](https://github.com/user-attachments/files/16315892/amf.log.txt)
[gnb.log.txt](https://github.com/user-attachments/files/16315893/gnb.log.txt)
[ue.log.txt](https://github.com/user-attachments/files/16315894/ue.log.txt)

do i have to adjust the time_adv_nsamples on the ue side? If yes can you tell me how to calculate it correctly ?

CrashxZ, 2024-07-20:
Update : I saw that there was thread where you suggested another user to add the inactivity timer in their gnb config- 
cu_cp:
  inactivity_timer: 7200

So I added it, and still the RRC is released.

pgawlowicz, 2024-07-21:
PDSCH at UE seems to be more or less OK, but PUSCH at gnb has a huge cfo. Do you have a possibility to connect your b200s to a better clock source?

CrashxZ, 2024-07-22:
I don't have access to a better clock source like a GPS DO, at the moment. Is there anything that I can turn to make the connection more stable to get a IP address ( and make it last for a few mins ) ?

Because if I do that , I can reasonably convince my lab to buy the external clocks.

CrashxZ, 2024-07-24:
I managed to connect for a second, using different B210s. But it disconnected right after.

[image]

Erick5G, 2024-08-02:
Hello, 
How can I add this patch to the SrsRan Project folder?
I have the same error with X310 as gnb and a b210 as UE and I do not have an assigned IP or an interface created.
I already tried downloading the BW, srate in both gnb and ue scripts, and applying the performance.
[image]

CrashxZ, 2024-08-02:
Fork the code and use the git apply command. 

Something like -> git apply name-of-file.patch

Or use a vs code extension, which does it for you. ( Its pretty simple) 

Let me know if it works for you ? 
Are you using external clocks ? 

It would be a great help if you can tell me about your setup and findings .

CrashxZ, 2024-08-03:
@pgawlowicz 
Hey, Luckily we have ordered 2 units of the GPSDO for using as external reference clocks.

While it arrives, I wanted to get a more stable connection, but i left it running for a few hours, and now it wont connect at all - 
Can you please suggest what happened? and how to fix that ?

The UE log - 

2024-08-03T18:12:13.715850 [UE     ] [I] Using binary srsue with arguments: mainak_ue.conf 
2024-08-03T18:12:13.744771 [UE     ] [I] Built in Release mode using commit 4e49269f7 on branch CPS_LAB.
2024-08-03T18:12:13.780048 [RF     ] [I] Configuring physical DL channel 0 with band-pass filter (0.0, 0.0)
2024-08-03T18:12:13.780049 [RF     ] [I] Configuring physical UL channel 0 with band-pass filter (0.0, 0.0)
2024-08-03T18:12:15.302913 [STCK   ] [E] Unknown PCAP option true
2024-08-03T18:12:15.305275 [PHY-SA ] [D] [    0] SYNC:  state=IDLE, tti=0
2024-08-03T18:12:15.305277 [PHY-SA ] [D] [    0] Discarding samples and sending tx_end
2024-08-03T18:12:15.310992 [MAC    ] [I] [    0] Resetting MAC
2024-08-03T18:12:15.310996 [MAC    ] [D] [    0] Msg3 buffer flushed
2024-08-03T18:12:15.310998 [MAC    ] [I] [    0] Logical Channel Setup: LCID=0, LCG=0, priority=0, PBR=-1, BSD=50ms, bucket_size=0
2024-08-03T18:12:15.311001 [MAC    ] [I] [    0] Logical Channel Setup: LCID=1, LCG=0, priority=1, PBR=-1, BSD=50ms, bucket_size=0
2024-08-03T18:12:15.311012 [MAC    ] [D] [    0] After MAC reset:
- lcid=0, lcg=0, prio=0, Bj=0, PBR=-1, BSD=50, buffer_len=0, sched_len=0
- lcid=1, lcg=0, prio=1, Bj=0, PBR=-1, BSD=50, buffer_len=0, sched_len=0
2024-08-03T18:12:15.311012 [MAC    ] [I] [    0] Setting configuration
2024-08-03T18:12:15.311014 [MAC    ] [I] [    0] BSR:   Configured timer reTX 2560 ms
2024-08-03T18:12:15.311020 [RLC    ] [I] Added LTE radio bearer with LCID 0 in Transparent Mode
2024-08-03T18:12:15.311033 [USIM   ] [I] Read Home PLMN Id=00101
2024-08-03T18:12:15.316014 [RLC-NR ] [I] Added LTE radio bearer with LCID 0 in Transparent Mode
2024-08-03T18:12:15.316019 [MAC-NR ] [D] [    0] cc_idx=0, set number of HARQ processes for DL to 8
2024-08-03T18:12:15.316020 [MAC-NR ] [I] [    0] Logical Channel Setup: LCID=0, LCG=0, priority=0, PBR=0, BSD=0ms, bucket_size=0
2024-08-03T18:12:15.316028 [RRC    ] [I] using srand seed of 316028
2024-08-03T18:12:15.344405 [MAC    ] [D] [    0] DL retx: 0.00 %, perpkt: 0.00, UL retx: 0.00 % perpkt: 0.00
2024-08-03T18:12:15.344409 [RLC    ] [D] lcid=0, rx_rate_mbps=0.00 (real=0.00), tx_rate_mbps=0.00 (real=0.00)
2024-08-03T18:12:15.344429 [GW     ] [D] gw_rx_rate_mbps=0.00 (real=0.00), gw_tx_rate_mbps=0.00 (real=0.00)
2024-08-03T18:12:15.345909 [STCK   ] [I] Triggering NAS switch on
2024-08-03T18:12:15.345957 [NAS5G  ] [I] Switching on
2024-08-03T18:12:15.345961 [NAS5G  ] [D] Changed to mm5g state: DEREGISTERED, with substate PLMN-SEARCH
2024-08-03T18:12:15.406510 [PHY-SA ] [D] [    0] run_stack_tti: from recv
2024-08-03T18:12:15.406511 [PHY-SA ] [D] [    0] SYNC:  Setting initial TTI time to 0.901121
2024-08-03T18:12:15.406512 [PHY-SA ] [D] [    0] run_stack_tti: calling stack tti=0, tti_jump=1
2024-08-03T18:12:15.406512 [PHY-SA ] [D] [    0] run_stack_tti: stack called
2024-08-03T18:12:15.406513 [PHY-SA ] [D] [    0] SYNC:  received 5760 samples from radio
2024-08-03T18:12:15.406515 [PHY-SA ] [D] [    0] SYNC:  state=IDLE, tti=0
2024-08-03T18:12:15.406516 [PHY-SA ] [D] [    0] Discarding samples and sending tx_end
2024-08-03T18:12:15.406556 [MAC    ] [D] [    0] Running MAC tti=0
2024-08-03T18:12:15.406557 [MAC    ] [D] [    0] Update Bj: lcid=0, Bj=0
2024-08-03T18:12:15.406557 [MAC    ] [D] [    0] Update Bj: lcid=1, Bj=0
2024-08-03T18:12:15.406561 [MAC    ] [D] [    0] RA:    IDLE:    rar_window_start=-1, rar_window_length=-1
2024-08-03T18:12:15.406562 [MAC-NR ] [D] [    0] Running MAC tti=0
2024-08-03T18:12:15.406567 [MAC-NR ] [D] [    0] nof_lcids_with_data=0, nof_lcgs_with_data=0, last_non_zero_lcg=-1 [lcg0=0, ] [lcid0=0, ]
2024-08-03T18:12:15.406570 [RRC    ] [D] State IDLE
2024-08-03T18:12:15.406598 [USIM   ] [I] Read Home PLMN Id=00101
2024-08-03T18:12:15.406599 [NAS5G  ] [I] Requesting IMSI attach (IMSI=001010123456780)
2024-08-03T18:12:15.406603 [NAS5G  ] [D] Starting T3410. Timeout in 15000 ms.
2024-08-03T18:12:15.406604 [NAS5G  ] [I] Sending Registration Request
2024-08-03T18:12:15.406604 [NAS5G  ] [D] Initiating RRC NR Connection
2024-08-03T18:12:15.406606 [RRC-NR ] [I] Proc "Setup Request" - Initiation of Setup request procedure
2024-08-03T18:12:15.406607 [RRC-NR ] [I] Proc "Cell Selection" - Starting...
2024-08-03T18:12:15.406612 [NAS5G  ] [D] Changed to mm5g state: REGISTERED-INITIATED
2024-08-03T18:12:15.406613 [STCK   ] [D] tti_tprof: duration=0 msec
2024-08-03T18:12:15.406615 [PHY-SA ] [I] [    0] Cell Search: Going to IDLE
2024-08-03T18:12:15.406616 [PHY-SA ] [I] [    0] Tuning Rx channel 0 to 1842.50 MHz
2024-08-03T18:12:15.406620 [RF     ] [I] Mapping RF channel 0 (device=0, channel=0) to logical carrier 0 on f_rx=1842.5 MHz
2024-08-03T18:12:15.407577 [PHY-SA ] [D] [    0] run_stack_tti: from recv
2024-08-03T18:12:15.407577 [PHY-SA ] [D] [    0] run_stack_tti: calling stack tti=0, tti_jump=1
2024-08-03T18:12:15.407578 [PHY-SA ] [D] [    0] run_stack_tti: stack called
2024-08-03T18:12:15.407578 [PHY-SA ] [D] [    0] SYNC:  received 5760 samples from radio
2024-08-03T18:12:15.407579 [PHY-SA ] [D] [    0] SYNC:  state=IDLE, tti=0
2024-08-03T18:12:15.407579 [PHY-SA ] [D] [    0] Discarding samples and sending tx_end
[... cut ...]
2024-08-03T18:12:15.410831 [MAC    ] [D] [    0] Update Bj: lcid=0, Bj=0
2024-08-03T18:12:15.410831 [MAC    ] [D] [    0] Update Bj: lcid=1, Bj=0
2024-08-03T18:12:15.410832 [MAC    ] [D] [    0] RA:    IDLE:    rar_window_start=-1, rar_window_length=-1
2024-08-03T18:12:15.410832 [MAC-NR ] [D] [    0] Running MAC tti=0
2024-08-03T18:12:15.410833 [MAC-NR ] [D] [    0] nof_lcids_with_data=0, nof_lcgs_with_data=0, last_non_zero_lcg=-1 [lcg0=0, ] [lcid0=0, ]
2024-08-03T18:12:15.410834 [STCK   ] [D] tti_tprof: duration=0 msec

CrashxZ, 2024-08-03:
I am not even receiving any underflow/overflow messages on the UE side.

frankist, 2024-08-04:
Isn't there a manual command on the UE side to set a starting frequency offset (I believe it is `rf.freq_offset`)? If the difference between gnb and ue is too high, the UE won't be able to compensate it

Erick5G, 2024-08-05:
Hi @CrashxZ @pgawlowicz 
I have already made the changes of the new patch, but still no IP address is assigned. Now I get the following errors: 2024-08-05T19:51:57.158745 [RRC-NR ] [I] Proc "Cell Selection" - Completed with failure.
2024-08-05T19:51:57.158751 [RRC-NR ] [W] Could not finish setup request. **Deallocating dedicatedInfoNAS PDU**
Do you know how can I fix them?

Attached logs: 
[gnb_20M.log](https://github.com/user-attachments/files/16502662/gnb_20M.log)
[ue.log](https://github.com/user-attachments/files/16502664/ue.log)

@CrashxZ 
Yes, I'm using GPSDO for both radios X310 as GNB and B210 as UE. But I have also performed tests with the internal clock of both radios, and I have the same error result with and without it.

**Late: 0; Underflow: 1; Overflow: 0;**

pgawlowicz, 2024-08-06:
@Erick5G could you share your configs?

Erick5G, 2024-08-06:
Sure, @pgawlowicz 

I've tested band n3 and n78 with this configuration for GNB and it's worked perfectly for Cost UE. Also with internal clock or gpsdo got the same result. 
But no sucess for SrsUE with B210.

Attached config:
[ue_rf.conf.txt](https://github.com/user-attachments/files/16514566/ue_rf.conf.txt)
[gnb_x310_20MHz_UECOST.yml.txt](https://github.com/user-attachments/files/16514567/gnb_x310_20MHz_UECOST.yml.txt)

Erick5G, 2024-08-07:
I've tried new config for GNB, adding the following lines:
  pdcch:
    dedicated:
      ss2_type: common                                            
      dci_format_0_1_and_1_1: false                               
    common:
      ss0_index: 0                                                
      coreset0_index: 12                                          
  prach:
    prach_config_index: 1 

Found Issues:
1. [PHY-SA ] [E] [    0**] prach_cfg.freq_offset=1 is not compatible with LTE**
2. PDCCH DMRS Rx: Measure L=2; ncce=4; RSRP=-30.2 dBfs; EPRE=-5.7 dBfs; Corr=0.004
    Discarded PDCCH candidate L=2;ncce=4; Correlation is too low (0.0<0.5); EPRE=-5.68; RSRP=-30.17;
3. [PHY-SA ] [E] [ 6360] SYNC: detected out-of-sync... skipping slot ...
4. [PHY0-NR] [E] [ 6380] PBCH-MIB: NR SFN (639) does not match current SFN (638)
5. [PHY0-NR] [W] [ 6440] PBCH-MIB: CRC failed

Ubuntu Terminal for the UE 
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=3694
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=3854
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=4014
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=4174

Attached Conf & logs:
[gnb_x310_20MHz_SRSUE.yml.txt](https://github.com/user-attachments/files/16516872/gnb_x310_20MHz_SRSUE.yml.txt)
[ue.conf.txt](https://github.com/user-attachments/files/16516873/ue.conf.txt)
[ue.log](https://github.com/user-attachments/files/16516874/ue.log)

As additional information: srsRan_Project is being used for the GNB and srsRan_4G for the UE. I hope this information is of some use. Regards

pgawlowicz, 2024-08-07:
could you try to use this config with gnb running on x310
```
ru_sdr:
  device_driver: uhd                  
  device_args: type=n3xx              
  clock: gpsdo                        
  sync: gpsdo                         
  srate: 30.72                        
  tx_gain: 35                         
  rx_gain: 60                         

cell_cfg:
  dl_arfcn: 368640                    
  band: 3                             
  channel_bandwidth_MHz: 20           
  common_scs: 15                      
  plmn: "00101"                       
  tac: 7                              
  pci: 1 
```

CrashxZ, 2024-08-07:
Hey @pgawlowicz ,
I have a question , do you think I messed something up by letting it run for hours ? Because I don't see rf underflow /overflow messages anymore . 

It gets stuck at attaching UE.

CrashxZ, 2024-08-07:
How much should I set that value to ? 

> Isn't there a manual command on the UE side to set a starting frequency offset (I believe it is `rf.freq_offset`)? If the difference between gnb and ue is too high, the UE won't be able to compensate it

Erick5G, 2024-08-07:
Hi @pgawlowicz 

> could you try to use this config with gnb running on x310
> 
> ```
> ru_sdr:
>   device_driver: uhd                  
>   device_args: type=x300,addr=X.X.X.X,master_clock_rate=184.32e6,send_frame_size=8000,recv_frame_size=8000        
>   clock: gpsdo                        
>   sync: gpsdo                         
>   srate: 30.72                        
>   tx_gain: 35                         
>   rx_gain: 60                         
> 
> cell_cfg:
>   dl_arfcn: 368640                    
>   band: 3                             
>   channel_bandwidth_MHz: 20           
>   common_scs: 15                      
>   plmn: "00101"                       
>   tac: 7                              
>   pci: 1 
> ```

Hi @pgawlowicz
With this config for X310, maximum values accepted for this radios are 31.5 & 37.5. 
**Terminal UE**
Error: setting gain for transmitter 0. Tx gain (i.e., 35.0 dB) is out-of-range. Range is [0.0, 31.5] dB in steps of 0.5 dB.
Error: setting gain for receiver 0. Rx gain (i.e., 60.0 dB) is out-of-range. Range is [0.0, 37.5] dB in steps of 0.5 dB.
srsGNB ERROR: Unable to create radio session.

Applying the changes did not produce any improvement, only PDU errors.

[gnb_x310_20MHz_UECOST.yml.txt](https://github.com/user-attachments/files/16534232/gnb_x310_20MHz_UECOST.yml.txt)
[uePLMNtac change.log](https://github.com/user-attachments/files/16534233/uePLMNtac.change.log)
[gnb_20M.log](https://github.com/user-attachments/files/16534234/gnb_20M.log)

pgawlowicz, 2024-08-08:
seems there are some issues on the gnb side as well:
```
2024-08-07T20:55:30.002583 [RF      ] [W] Real-time failure in RF: underflow
2024-08-07T20:55:34.002587 [RF      ] [W] Real-time failure in RF: underflow
2024-08-07T20:55:46.002601 [RF      ] [W] Real-time failure in RF: underflow
2024-08-07T20:55:53.002523 [RF      ] [W] Real-time failure in RF: underflow
2024-08-07T20:55:58.002561 [RF      ] [W] Real-time failure in RF: underflow
2024-08-07T20:56:22.002599 [RF      ] [W] Real-time failure in RF: underflow
2024-08-07T20:56:25.002658 [RF      ] [W] Real-time failure in RF: underflow
2024-08-07T20:56:30.002634 [RF      ] [W] Real-time failure in RF: underflow
2024-08-07T20:56:34.002560 [RF      ] [W] Real-time failure in RF: underflow
```

could you try to run this script on both PC? https://github.com/srsran/srsRAN_Project/blob/main/scripts/srsran_performance

Erick5G, 2024-08-09:
Done!
We still receive this info on terminal. 

**Late: 0; Underflow: 1; Overflow: 0;**
And the same issues of PDU.

pgawlowicz, 2024-08-09:
could you try with uhd-4.3?

CrashxZ, 2024-08-09:
I am using UHD 4.3 with a core i7 11900k for one of the UEs , I still get the underflow/late messages.

pgawlowicz, 2024-08-12:
hmm, that is strange. Could you try with a smalle BW, like 10MHz and post your configs ang logs here?

pgawlowicz, 2024-10-16:
@CrashxZ any updates on this issue?

CrashxZ, 2024-10-16:
Hi @pgawlowicz !
We got 2 units of GPSDOs and 2 active GPS antennas for the B210s. I tried connecting it using them by setting clock=gpsdo and sync=gpsdo, but it didn't help. I have similar connection issues. It connects intermittently for a few seconds and disconnects (if it even connects).  Since we have to use this as an indoor solution, I'm not sure on what to do .  

I'll post the logs and the config files shortly .

pgawlowicz, 2024-10-16:
could you try the patch in this comment https://github.com/srsran/srsRAN_Project/issues/826#issuecomment-2370588502

CrashxZ, 2024-10-23:
Here are the logs for the indoor setup. Unfortunately, i cannot use clock=gpsdo and sync=gpsdo on the gNb since, its needs to be outdoors, for a gps signal. But after the from the comment #826  the connection is established every time (unlike intermittently like before). but only for a few times.

[gnb.log.txt](https://github.com/user-attachments/files/17497088/gnb.log.txt)
[ue.log.txt](https://github.com/user-attachments/files/17497089/ue.log.txt)

CrashxZ, 2024-10-23:
I plan to test outdoors soon, but we need this for indoor application, so I have to come up with a work around.

CrashxZ, 2024-10-28:
Hi,
I pulled the newest srsRAN-Project, commit - 9d5dd742a. Updated to the latest UHD Drivers (4.7) and still running into the same issue - 
`2024-10-28T23:00:39.845798 [DU-MNG  ] [I] ue=0: RLF timer expired with cause="RLC max ReTxs reached". Requesting a UE release...
2024-10-28T23:00:39.845802 [PHY     ] [D] [   421.7] PDCCH: rnti=0x4601 ss_id=2 format=1_0 cce=0 al=2 t=9.9us`

I have also noticed that it says UE Context Release [Action:2] in the open5gc logs (There was once i saw it was action 1) - 
`open5gs_5gc  | 10/29 00:00:38.433: [amf] INFO: InitialUEMessage (../src/amf/ngap-handler.c:401)
open5gs_5gc  | 10/29 00:00:38.433: [amf] INFO: [Added] Number of gNB-UEs is now 1 (../src/amf/context.c:2550)
open5gs_5gc  | 10/29 00:00:38.433: [amf] INFO:     RAN_UE_NGAP_ID[0] AMF_UE_NGAP_ID[1] TAC[7] CellID[0x66c000] (../src/amf/ngap-handler.c:562)
open5gs_5gc  | 10/29 00:00:38.433: [amf] INFO: [suci-0-001-01-0000-0-0-0123456780] Unknown UE by SUCI (../src/amf/context.c:1835)
open5gs_5gc  | 10/29 00:00:38.433: [amf] INFO: [Added] Number of AMF-UEs is now 1 (../src/amf/context.c:1616)
open5gs_5gc  | 10/29 00:00:38.433: [gmm] INFO: Registration request (../src/amf/gmm-sm.c:1165)
open5gs_5gc  | 10/29 00:00:38.433: [gmm] INFO: [suci-0-001-01-0000-0-0-0123456780]    SUCI (../src/amf/gmm-handler.c:166)
open5gs_5gc  | 10/29 00:00:39.999: [amf] INFO: UE Context Release [Action:2] (../src/amf/ngap-handler.c:1698)
open5gs_5gc  | 10/29 00:00:39.999: [amf] INFO:     RAN_UE_NGAP_ID[0] AMF_UE_NGAP_ID[1] (../src/amf/ngap-handler.c:1699)
open5gs_5gc  | 10/29 00:00:39.999: [amf] INFO:     SUCI[suci-0-001-01-0000-0-0-0123456780] (../src/amf/ngap-handler.c:1702)
open5gs_5gc  | 10/29 00:00:39.999: [amf] INFO: [Removed] Number of gNB-UEs is now 0 (../src/amf/context.c:2557)`

 I believe that the TX and RX gains are good with the observed sinr, so can it be an authentication issue from the open5gc? I am using the dockerised version provided in the repository.

Here are the logs  and configs - 

[gnb.yaml.txt](https://github.com/user-attachments/files/17549692/gnb.yaml.txt)
[mainak_ue.conf.txt](https://github.com/user-attachments/files/17549693/mainak_ue.conf.txt)
[ue.log3.txt](https://github.com/user-attachments/files/17549694/ue.log3.txt)
[gnb.log3.txt](https://github.com/user-attachments/files/17549695/gnb.log3.txt)

pgawlowicz, 2024-10-29:
hmm, PUSCH cfo is high and variable. 
```
2024-10-28T23:00:38.563891 [PHY     ] [D] [   293.0] PUSCH: rnti=0x4601 h_id=0 prb=[3, 22) symb=[0, 14) mod=16QAM rv=0 tbs=800 crc=KO iter=6.0 sinr=3.9dB t=193.9us uci_t=0.0us ret_t=0.0us
  cfo=+1354.1Hz
2024-10-28T23:00:38.574148 [PHY     ] [D] [   294.0] PUSCH: rnti=0x4601 h_id=0 prb=[3, 22) symb=[0, 14) mod=16QAM rv=0 tbs=800 crc=KO iter=6.0 sinr=-2.6dB t=182.5us uci_t=0.0us ret_t=0.0us
  cfo=-1381.8Hz
2024-10-28T23:00:38.584059 [PHY     ] [D] [   295.0] PUSCH: rnti=0x4601 h_id=0 prb=[3, 22) symb=[0, 14) mod=16QAM rv=0 tbs=800 crc=KO iter=6.0 sinr=-2.2dB t=177.4us uci_t=0.0us ret_t=0.0us
  cfo=-1356.4Hz
2024-10-28T23:00:38.593971 [PHY     ] [D] [   296.0] PUSCH: rnti=0x4601 h_id=0 prb=[3, 22) symb=[0, 14) mod=16QAM rv=0 tbs=800 crc=KO iter=6.0 sinr=-1.8dB t=174.1us uci_t=0.0us ret_t=0.0us
  cfo=-1360.4Hz
2024-10-28T23:00:38.603893 [PHY     ] [D] [   297.0] PUSCH: rnti=0x4601 h_id=0 prb=[3, 22) symb=[0, 14) mod=16QAM rv=0 tbs=800 crc=KO iter=6.0 sinr=-3.1dB t=177.5us uci_t=0.0us ret_t=0.0us
  cfo=-1320.9Hz
```
 
Do you have an external clock to connect with both gnb and UE USRPs?

CrashxZ, 2024-10-30:
Hi,
I tested it with a GPSDO connected to both the USRPs, and now it works. It gets connected. and I do get an ip address. However we have to place it outside (the active GPS antennas). 

Is there a possibility to use it indoor, i.e. without a gps antena? Since our UE USRP has to be mounted on a moving robot, we have to use it without gps. 

The logs were too big so i made a onedrive link - 
 - 
[https://uconn-my.sharepoint.com/:t:/g/personal/mainak_mondal_uconn_edu/EZ5vnsJaWvtJhfgroA6Ve8MB2A7Ns2pUWA4wI6iti_PRwQ?e=1Jdwcy](UE Log)
- [https://uconn-my.sharepoint.com/:t:/g/personal/mainak_mondal_uconn_edu/ERzvaxlXLtVEpivqcGgE8h8BDdqv6r3pWJv0dGQ3ElPIbA?e=RcY5ag](gNb Log)

pgawlowicz, 2024-11-05:
You can try connecting to external clocks.

pgawlowicz, 2024-11-11:
You might try to synchronize the GPSDO clocks using a GPS antenna and then disconnect the antenna. The clocks should continue to operate accurately for an extended period.

CrashxZ, 2024-11-11:
Thats a good solution for the timebeing, and I am doing that at the moment for running benchmarks and running ROS over the 5G Link. But unfortunately, the link disconnects after 30 - 50 seconds. 

[image]

and sometimes - 
[image]

[GnB Log](https://uconn-my.sharepoint.com/:t:/g/personal/mainak_mondal_uconn_edu/EfWb7hVLAzxLi0Iltn1GwPMB2Vmp0of9n2kuQGlUGF2TwQ?e=fbTS5I) 

[UE Log](https://uconn-my.sharepoint.com/:t:/g/personal/mainak_mondal_uconn_edu/Edfa5SIVVYZIvGvR13WCFVYB_4OA2afH5Cci5jP34q7Jcg?e=1FuSwP)

pgawlowicz, 2024-11-13:
Uplink SNR is quite low, could you try to tune the srsUE tx_gain?

newt2105, 2025-03-26:
Hi @CrashxZ ,
Did you solve this problem?

CrashxZ, 2025-03-26:
I actually haven't ! I'm trying to test outdoors now, Which works intermittently.  I need to try out different SCS settings but the SRS UE doesn't support it.

mochammadfikri28, 2025-07-12:
> Sebenarnya belum! Saya sedang mencoba mengujinya di luar ruangan, yang berfungsi sesekali. Saya perlu mencoba pengaturan SCS yang berbeda, tetapi SRS UE tidak mendukungnya.

HI @CrashxZ , Has the srsRAN gNB and srsUE worked? Can you share the required files?
Thank you.

CrashxZ, 2025-07-15:
Hi @mochammadfikri28  ,
The srsRAN gNb and srsUE , werent working as expected with my testbed setup, thats why I have moved to the OpenAirInterface repo and i am working with that (It synchronizes without gpsdo or active gps connection, quite consistently.)

mochammadfikri28, 2025-07-15:
> Hi [@mochammadfikri28](https://github.com/mochammadfikri28) , The srsRAN gNb and srsUE , werent working as expected with my testbed setup, thats why I have moved to the OpenAirInterface repo and i am working with that (It synchronizes without gpsdo or active gps connection, quite consistently.)

@CrashxZ Okay, thats great. im still trying out gnB and srsUE . Have you ever run it with docker-compose.yml from srsRAN Project?
thankyou