srsran/srsRAN_Project #737: Very low throughput with RAN550

melkael, 2024-07-17:
Hello!
## Issue Description ##
We are trying to run SRS with a Benetel RAN550. We manage to attach the UE reliably. However the throughput is extremely low (downlink around 1.5mbps and uplink around 5mbps, see the attached iperf screenshots). The issue reminded us an issue we had of too high transmitted power so we tried playing with the gain (varying it to several values between 4 and 24db) but it did not help. We also tried modifying downlink_scaling which did not help either.

## Setup Details ##
* UE: OnePlus Nord AC2003
* RU: benetel RAN 550
* Firmware: 1.0.4
* DU server: Xeon(R) Gold 6348 CPU @ 2.60GH with 256gB RAM

* srs configuration file:
```
gnb_id: 1
amf:
  addr: 172.20.25.74                                               # The address or hostname of the AMF.
  bind_addr: 172.20.25.61                                            # A local IP that the gNB binds to for traffic from the AMF.

ru_ofh:
  t1a_max_cp_dl: 500                                              # Maximum T1a on Control-Plane for Downlink in microseconds.
  t1a_min_cp_dl: 450                                              # Minimum T1a on Control-Plane for Downlink in microseconds.
  t1a_max_cp_ul: 350                                              # Maximum T1a on Control-Plane for Uplink in microseconds.
  t1a_min_cp_ul: 290                                              # Minimum T1a on Control-Plane for Uplink in microseconds.
  t1a_max_up: 360                                                 # Maximum T1a on User-Plane in microseconds.
  t1a_min_up: 300                                                 # Minimum T1a on User-Plane in microseconds.
  ta4_max: 200                                                    # Maximum Ta4 on User-Plane in microseconds.
  ta4_min: 0                                                      # Minimum Ta4 on User-Plane in microseconds.
  is_prach_cp_enabled: true                                      # Configures if Control-Plane messages should be used to receive PRACH messages.
  compr_method_ul: bfp                                            # Uplink compression method.
  compr_bitwidth_ul: 9                                            # Uplink IQ samples bitwidth after compression.
  compr_method_dl: bfp                                            # Downlink compression method.
  compr_bitwidth_dl: 9                                            # Downlink IQ samples bitwidth after compression.
  compr_method_prach: bfp                                         # PRACH compression method.
  compr_bitwidth_prach: 9                                         # PRACH IQ samples bitwidth after compression.
  enable_ul_static_compr_hdr: false                                # Configures if the compression header is present for uplink User-Plane messages (false) or not present (true).
  enable_dl_static_compr_hdr: false                                # Configures if the compression header is present for downlink User-Plane messages (false) or not present (true).
  iq_scaling: 4.5                                                 # IQ samples scaling factor applied before compression, should be a positive value smaller than 10.
  cells:
  - network_interface: enp179s0f0                                   # Ethernet interface name used to communicate with the RU.
    ru_mac_addr: 70:b3:d5:e1:5d:ca                                # RU MAC address.
    du_mac_addr: 64:9d:99:b2:14:b4                                # DU MAC address.
    vlan_tag_cp: 2                                                # VLAN tag value for C-Plane.
    vlan_tag_up: 2                                                # VLAN tag value for U-Plane.
    prach_port_id: [4,5]                                          # PRACH eAxC port value.
    dl_port_id: [0,1]                                             # Downlink eAxC port values.
    ul_port_id: [0,1]                                             # Uplink eAxC port values.

cell_cfg:
  dl_arfcn: 640000                                                # ARFCN of the downlink carrier (center frequency).
  band: 78                                                        # The NR band.
  channel_bandwidth_MHz: 20                                       # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 30                                                  # Subcarrier spacing in kHz used for data.
  plmn: "00101"                                                   # PLMN broadcasted by the gNB.
  tac: 1                                                          # Tracking area code (needs to match the core configuration).
  pci: 1                                                          # Physical cell ID.
  nof_antennas_dl: 2
  nof_antennas_ul: 2
  prach:
    prach_config_index: 159                                       # PRACH configuration index.
    prach_root_sequence_index: 1                                  # PRACH root sequence index.
    zero_correlation_zone: 0                                      # Zero correlation zone.
    prach_frequency_start: 2                                      # Offset in PRBs of lowest PRACH transmission occasion in frequency domain respective to PRB 0.
    preamble_trans_max: 50
    power_ramping_step_db: 4
  tdd_ul_dl_cfg:
    dl_ul_tx_period: 10                                           # Optional INT (10). Sets the TDD pattern periodicity in slots. The combination of this value and the chosen numerology must lead to a TDD periodicity of 0.5, >
    nof_dl_slots: 7                                               # Optional INT (6). Number of consecutive full Downlink slots. Supported: [0-80].
    nof_dl_symbols: 6                                             # Optional INT (0). Number of Downlink symbols at the beginning of the slot following full Downlink slots. Supported: [0-13].
    nof_ul_slots: 2                                               # Optional INT (3). Number of consecutive full Uplink slots. Supported: [0 - 80].
    nof_ul_symbols: 4
  ssb:
    ssb_block_power_dbm: -20
  pdsch:
    mcs_table: qam64
  pusch:
    p0_nominal_with_grant: -76
log:
  filename: log.log                                          # Path of the log file.
  all_level: warning
  f1ap_level: info                   # Optional TEXT (warning). Sets F1AP log level.
  f1u_level: info
```
* ru_config.cfg:

```
# Config file for Benetel RANx50 radio units
# Please place this file in /etc

# MIMO mode - the radio can operate in 1_3, 2_4, 1_2_3_4_4x2 or 1_2_3_4_4x4 mode
# Variable name : mimo_mode
# Valid options : 1_3
#                 2_4
#                 1_2_3_4_4x2
#                 1_2_3_4_4x4
mimo_mode=1_2_3_4_4x2

# FPGA downlink scaling in steps of 6 dB (0, 6, 12 and 18 can be set)
# Variable name : downlink_scaling
# Valid options : 0
#                 6
#                 12
#                 18
downlink_scaling=6

# PRACH format
# Variable name : prach_format
# Valid options : short
#                 long
prach_format=short

# Short form PRACH compression settings (note uncompressed mode is not supported when RU is in 4x2 or 4x4 mode)
# Variable name : compression
# Valid options : static_uncompressed
#                 dynamic_uncompressed
#                 static_compressed
#                 dynamic_compressed
compression=dynamic_compressed

# Long form PRACH compression settings
# Variable name : lf_prach_compression_enable
# Valid options : true
#                 false
lf_prach_compression_enable=false

# C-Plane per-symbol workaround
# Variable name : cplane_per_symbol_workaround
# Valid options : enabled
#                 disabled
cplane_per_symbol_workaround=disabled

# CUPLANE dl couplling sectionID
# Variable name : cuplane_dl_coupling_sectionI
# Valid options : enabled
#                 disabled
cuplane_dl_coupling_sectionID=disabled

# FlexRAN PRACH workaround
# Variable name : flexran_prach_workaround
# Valid options : enabled
#                 disabled
flexran_prach_workaround=disabled

#DL_special slot tunning
#Variable name : dl_tuning_special_slot
#Valid range   : 0x0:0x13bc
dl_tuning_special_slot=0xfd00000
```
* Uplink throughput
[image]

* Downlink throughput
[image]

Thank you

andrepuschmann, 2024-07-18:
Hey - can you share console traces please? I assume timing is good - so it's very likely a power calibration matter.

melkael, 2024-07-18:
Sure, please find the whole file for a DL test attached: [log.log](https://github.com/user-attachments/files/16288523/log.log). I tried varying various power parameters on both SRS and the benetel, but no success for now.

The benetel indicates TX power of around 15-17dBm currently (with 24dB gain and the attenuation parameter in /etc/ru_config.cfg set to 0). 

Also, I think the environment is not the issue because we have a Foxconn radio which works fine in the same room.

ismagom, 2024-07-18:
Hi @melkael, it seem the tx power is too low or DL signal is very low. You can see it by the CQI=2 reported by the UE. The UL power is also very low, but you should look at it later. Can you try the following?

1) If you set 4x2 mode in the RU, set 4 DL antennas in the gNB config too
2) Try increasing iq_scale until you see dBFS ~-15/-13 dB in the oru_stats command in the RU

melkael, 2024-07-19:
Thank you guys for your help, the issue was that the TDD patterns were wrong.

Side question: will you make a call for demos for the srs workshop in october?

closed 2024-07-19 by melkael