srsran/srsRAN_Project #742: Network connectivity issues with docker setup

dominikheinz, 2024-07-23:
## Issue Description ##
I am using srsran and open5gs in the docker compose setup. 1 system runs gnb+core and the other system runs the UE.
I get the UE to connect, and from what I can tell also establish a PDN session, but I can't perform a speedtest with `iperf3` because of connectivity problems.
I also can't reach the librespeed instance that is running on the gnb system under `192.168.42.79:80`.
I assume my network routing is configured incorrectly. I hope someone can help me.

## Setup Details ##
- Ubuntu 22.04
- USRP B210
- Quectel RM520N-GL

## Expected Behavior ##
I would assume that I can perform a speedtest of the connection with iperf3.

## Actual Behaviour ##
I can't reach the gnb (i think?). I am not entirely sure what IP relates to what.

## Steps to reproduce the problem ##
1. Run `docker-compose -f docker-compose-custom.yml up`
2. Observe the UE establishing a connection
3. Configure ip on UE system like so:
```
sudo ip link set dev wwan0 up
sudo ip a add 10.45.0.3/16 dev wwan0
```
4. Try to `ping 10.53.1.3 -I wwan0` (I think thats the core?). Can't reach.
5. Try to `ping 192.168.42.79 -I wwan0` (gnb host system where speedtest is running). Can't reach.

## Additional Information ##

**Routing table gnb+core system:**
```
╰─❯  route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         192.168.42.1    0.0.0.0         UG    101    0        0 enp0s31f6
10.53.1.0       0.0.0.0         255.255.255.0   U     0      0        0 br-bd331f1276aa
169.254.0.0     0.0.0.0         255.255.0.0     U     1000   0        0 enx00e02265f824
172.17.0.0      0.0.0.0         255.255.0.0     U     0      0        0 docker0
172.18.0.0      0.0.0.0         255.255.0.0     U     0      0        0 br-e89b9b16d236
172.19.1.0      0.0.0.0         255.255.255.0   U     0      0        0 br-dc96ae153b4a
192.168.2.0     0.0.0.0         255.255.255.0   U     100    0        0 enx00e02265f824
192.168.42.0    0.0.0.0         255.255.255.0   U     101    0        0 enp0s31f6
```

**Routing table gnb system:**
```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: enp0s31f6: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 4c:d7:17:84:e2:e9 brd ff:ff:ff:ff:ff:ff
    inet 192.168.42.79/24 brd 192.168.42.255 scope global noprefixroute enp0s31f6
       valid_lft forever preferred_lft forever
    inet6 fe80::8416:f39b:5897:ee54/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
3: enx00e02265f824: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 1000
    link/ether 00:e0:22:65:f8:24 brd ff:ff:ff:ff:ff:ff
    inet 192.168.2.10/24 brd 192.168.2.255 scope global dynamic noprefixroute enx00e02265f824
       valid_lft 860128sec preferred_lft 860128sec
    inet6 fe80::c840:b501:6201:10f2/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
4: ogstun: <NO-CARRIER,POINTOPOINT,MULTICAST,NOARP,UP> mtu 1400 qdisc fq_codel state DOWN group default qlen 500
    link/none 
    inet6 fe80::175:c16d:3c3a:5f1f/64 scope link stable-privacy 
       valid_lft forever preferred_lft forever
5: docker0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default 
[... 249 lines cut ...]
open5gs_5gc       | 07/23 14:25:00.432: [sbi] WARNING: [SMF] (NRF-discover) NF has already been added [76b06046-48ee-41ef-9bbb-719bdd4c4934:1] (../lib/sbi/nnrf-handler.c:1057)
open5gs_5gc       | 07/23 14:25:00.432: [sbi] WARNING: NF EndPoint(addr) updated [127.0.0.4:80] (../lib/sbi/context.c:2174)
open5gs_5gc       | 07/23 14:25:00.432: [sbi] WARNING: NF EndPoint(addr) updated [127.0.0.4:7777] (../lib/sbi/context.c:1917)
open5gs_5gc       | 07/23 14:25:00.432: [sbi] INFO: [SMF] (NF-discover) NF Profile updated [76b06046-48ee-41ef-9bbb-719bdd4c4934:1] (../lib/sbi/nnrf-handler.c:1095)
open5gs_5gc       | 07/23 14:25:00.432: [smf] INFO: [Added] Number of SMF-UEs is now 1 (../src/smf/context.c:1019)
open5gs_5gc       | 07/23 14:25:00.432: [smf] INFO: [Added] Number of SMF-Sessions is now 1 (../src/smf/context.c:3068)
open5gs_5gc       | 07/23 14:25:00.434: [sbi] INFO: [SMF] (SCP-discover) NF registered [76b06046-48ee-41ef-9bbb-719bdd4c4934:1] (../lib/sbi/path.c:211)
open5gs_5gc       | 07/23 14:25:00.435: [smf] INFO: UE SUPI[imsi-999700000066491] DNN[srsapn] IPv4[10.45.0.3] IPv6[] (../src/smf/npcf-handler.c:539)
open5gs_5gc       | 07/23 14:25:00.435: [upf] INFO: [Added] Number of UPF-Sessions is now 1 (../src/upf/context.c:208)
open5gs_5gc       | 07/23 14:25:00.435: [gtp] INFO: gtp_connect() [127.0.0.4]:2152 (../lib/gtp/path.c:60)
open5gs_5gc       | 07/23 14:25:00.435: [upf] INFO: UE F-SEID[UP:0xb86 CP:0x3d2] APN[srsapn] PDN-Type[1] IPv4[10.45.0.3] IPv6[] (../src/upf/context.c:485)
open5gs_5gc       | 07/23 14:25:00.435: [upf] INFO: UE F-SEID[UP:0xb86 CP:0x3d2] APN[srsapn] PDN-Type[1] IPv4[10.45.0.3] IPv6[] (../src/upf/context.c:485)
open5gs_5gc       | 07/23 14:25:00.435: [gtp] INFO: gtp_connect() [10.53.1.2]:2152 (../lib/gtp/path.c:60)
open5gs_5gc       | 07/23 14:25:00.435: [smf] WARNING: Unknown PCO ID:(0x17) (../src/smf/context.c:3011)
open5gs_5gc       | 07/23 14:25:00.435: [smf] WARNING: Unknown PCO ID:(0x23) (../src/smf/context.c:3011)
open5gs_5gc       | 07/23 14:25:00.435: [smf] WARNING: Unknown PCO ID:(0x24) (../src/smf/context.c:3011)
open5gs_5gc       | 07/23 14:25:00.509: [gtp] INFO: gtp_connect() [10.53.1.3]:2152 (../lib/gtp/path.c:60)
open5gs_5gc       | 07/23 14:25:00.510: [amf] INFO: [imsi-999700000066491:1:11][0:0:NULL] /nsmf-pdusession/v1/sm-contexts/{smContextRef}/modify (../src/amf/nsmf-handler.c:837)
^CGracefully stopping... (press Ctrl+C again to force)
Stopping srsran_gnb     ... done
Stopping grafana        ... done
Stopping metrics_server ... done
Stopping influxdb       ... done
Stopping open5gs_5gc    ... done
```

herlesupreeth, 2024-07-23:
First, have you done the following??

`sudo ip link set wwan0 up`

Second,

> ping 10.53.1.3 -I wwan0 (I think thats the core?). Can't reach.

This is wrong. The Core is at 10.53.1.2

dominikheinz, 2024-07-23:
@herlesupreeth Thanks for the reply.

Yes I did set the interface up.
I also can't reach the core at 10.53.1.2

:/

What could I try?

herlesupreeth, 2024-07-23:
In the gNB machine is see the following, do you have open5gs installed via apt-get install also on that machine? 

```
4: ogstun: <NO-CARRIER,POINTOPOINT,MULTICAST,NOARP,UP> mtu 1400 qdisc fq_codel state DOWN group default qlen 500
    link/none 
    inet6 fe80::175:c16d:3c3a:5f1f/64 scope link stable-privacy 
       valid_lft forever preferred_lft forever
```

Also, can you send me the output of `ip r `on UE machine?

dominikheinz, 2024-07-24:
@herlesupreeth Thanks for the reply :) 
On the same machine, I have tested a native setup before (by that I mean not dockerized) too, yes. 
For that, I installed Open5GS via apt, which is probably why this interface exists. However, I made sure to stop all Open5GS services on the host, before running the dockerized setup. But of course this interface still exists, is that a problem?

Running ip r on the UE machine returns:
```
10.45.0.0/16 dev wwan0 proto kernel scope link src 10.45.0.3
```

dominikheinz, 2024-08-01:
@herlesupreeth Any ideas what could be the issue?

dominikheinz, 2024-08-09:
@pgawlowicz Maybe you have an idea on what the problem with the network configuration could be?

pgawlowicz, 2024-08-09:
check here: 
https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#nd-open5gs-instance-installed-manually

also make sure to enable IP forwarding on the open5gs host machine.

dominikheinz, 2024-08-12:
> check here: https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#nd-open5gs-instance-installed-manually
> 
> also make sure to enable IP forwarding on the open5gs host machine.

That progress. 
I added the route `sudo ip route add 10.45.0.0/16 via 10.53.1.2 dev br-b2eb147102f3`

This now allows me to connect to a local service running on the GNB system. 
However, I still are not able to connect directly to the internet via the UE.
E.g. `ping 8.8.8.8 -I wwan0` doesn't return any anwers.
The idea is to utilize the UE to browser the internet. Am I missing a route?

`ip r` on GNB system:

```
default via 192.168.42.1 dev enp0s31f6 proto static 
10.45.0.0/16 via 10.53.1.2 dev br-53fbb211a551 
10.53.1.0/24 dev br-53fbb211a551 proto kernel scope link src 10.53.1.1 
172.17.0.0/16 dev docker0 proto kernel scope link src 172.17.0.1 linkdown 
172.18.0.0/16 dev br-1b65988aed95 proto kernel scope link src 172.18.0.1 
172.19.1.0/24 dev br-cb545e302083 proto kernel scope link src 172.19.1.1 
192.168.2.0/24 dev enp0s20f0u1 proto kernel scope link src 192.168.2.10 
192.168.42.0/24 dev enp0s31f6 proto kernel scope link src 192.168.42.79 
```

`ip r` on UE system:

```
default via 192.168.42.1 dev enp0s31f6 proto static metric 100 
10.45.0.0/16 dev wwan0 proto kernel scope link src 10.45.0.3 
169.254.0.0/16 dev enp0s31f6 scope link metric 1000 
192.168.24.79 dev wwan0 scope link 
192.168.42.0/24 dev enp0s31f6 proto kernel scope link src 192.168.42.13 metric 100 
192.168.42.79 via 10.45.0.1 dev wwan0 
```

pgawlowicz, 2024-08-12:
looks ok, do you see the ping in open5gs with wireshark? do you run open5gs in docker on the same machine as gnb? Did you disable the ogstun interface?

dominikheinz, 2024-08-12:
@pgawlowicz I run open5gs and gnb both in docker (see initial post for the docker file).
I set everything up on a new machine, so the ogstun interface is not even present anymore (well, it technically is inside the open5gs container, but that is expected. It is not present not on the host anymore tho, so should not interfere).

Yes, I can see the ping inside the open5gs container. 
I checked with tcpdump:
```
root@1f5584c8e6a5:/open5gs# sudo tcpdump -i any icmp
bash: sudo: command not found
root@1f5584c8e6a5:/open5gs#  tcpdump -i any icmp
tcpdump: data link type LINUX_SLL2
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on any, link-type LINUX_SLL2 (Linux cooked v2), snapshot length 262144 bytes
11:26:11.557948 ogstun In  IP 10.45.0.3 > dns.google: ICMP echo request, id 4, seq 114, length 64
11:26:11.557962 eth0  Out IP 10.45.0.3 > dns.google: ICMP echo request, id 4, seq 114, length 64
11:26:12.577942 ogstun In  IP 10.45.0.3 > dns.google: ICMP echo request, id 4, seq 115, length 64
11:26:12.577956 eth0  Out IP 10.45.0.3 > dns.google: ICMP echo request, id 4, seq 115, length 64
11:26:13.597964 ogstun In  IP 10.45.0.3 > dns.google: ICMP echo request, id 4, seq 116, length 64
11:26:13.597979 eth0  Out IP 10.45.0.3 > dns.google: ICMP echo request, id 4, seq 116, length 64
11:26:14.637897 ogstun In  IP 10.45.0.3 > dns.google: ICMP echo request, id 4, seq 117, length 64
11:26:14.637912 eth0  Out IP 10.45.0.3 > dns.google: ICMP echo request, id 4, seq 117, length 64
```

dominikheinz, 2024-08-12:
Seems like what was missing was `sudo iptables -t nat -A POSTROUTING -o enp0s31f6 -j MASQUERADE` - now I can ping 8.8.8.8.

However, by default requests to the internet still  go over the ethernet interface.
Do I need to change the default route on the system with the UE connected to it, to ensure normal web browsing also goes over the modem?

pgawlowicz, 2024-08-12:
yes, your default route at UE pc is going over eth interface:
```
default via 192.168.42.1 dev enp0s31f6 proto static metric 100 
```
you need to change it.

dominikheinz, 2024-08-13:
@pgawlowicz Awesome that works! Very happy with that. :partying_face: 

Do you know if I can also persist these changes in the docker-compose file, so that in the future, I don't have to do these steps manually? Mainly, I want to ensure, that running docker compose up, also:
- Creates a default route `sudo ip route add 10.45.0.0/16 via 10.53.1.2 dev br-b2eb147102f3`
- Applies NAT rules `sudo iptables -t nat -A POSTROUTING -o enp0s31f6 -j MASQUERADE`
- Changes the default route `default via 192.168.42.1 dev enp0s31f6 proto static metric 100`

And when stopping the containers these changes are reversed?
Would like the docker compose file to do all these steps, so I don't have to do any manual configuration to get things running.

pgawlowicz, 2024-08-13:
Hi, I think that docker compose does not support running commands on the host machine. Probably, the simplest solution would be to create a bash script to set up the environment.

closed 2024-09-19 by pgawlowicz