srsran/srsRAN_Project #748: srsUE establishes RRC connection  gets released

xixihahahha, 2024-07-31:
## Issue Description ##
The rrc is released immediately after connection

## Setup Details ##
ubuntu 22.04 arm and follow srsRAN_Project tutorial for ZMQ-based setup

## Expected Behavior ##
RRC NR reconfiguration successful.

## Actual Behaviour ##
srsUE
```
Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_zmq.so
Inactive RF plugins: 
Reading configuration file ue_zmq.conf...

Built in Release mode using commit ec29b0c1f on branch master.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://127.0.0.1:2001,rx_port=tcp://127.0.0.1:2000,base_srate=23.04e6
Supported RF device list: UHD zmq file
CHx base_srate=23.04e6
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://127.0.0.1:2000
CH0 tx_port=tcp://127.0.0.1:2001
Current sample rate is 23.04 MHz with a base rate of 23.04 MHz (x1 decimation)
Current sample rate is 23.04 MHz with a base rate of 23.04 MHz (x1 decimation)
Waiting PHY to initialize ... done!
Attaching UE...
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=174
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=334
Random Access Complete.     c-rnti=0x4602, ta=0
RRC Connected
Received RRC Release
```

open5GS
```
pen5gs_5gc       | 07/31 04:49:56.087: [app] INFO: Configuration: 'open5gs-5gc.yml' (../lib/app/ogs-init.c:130)
open5gs_5gc       | 07/31 04:49:56.089: [sbi] INFO: NF Service [namf-comm] (../lib/sbi/context.c:1812)
open5gs_5gc       | 07/31 04:49:56.091: [sbi] INFO: nghttp2_server() [http://127.0.0.5]:7777 (../lib/sbi/nghttp2-server.c:414)
open5gs_5gc       | 07/31 04:49:56.091: [amf] INFO: ngap_server() [10.53.1.2]:38412 (../src/amf/ngap-sctp.c:61)
open5gs_5gc       | 07/31 04:49:56.091: [sctp] INFO: AMF initialize...done (../src/amf/app.c:33)
open5gs_5gc       | 07/31 04:49:56.092: [nrf] INFO: [91b73ea6-4ee7-41ef-9170-ab378469ed59] NF registered [Heartbeat:10s] (../src/nrf/nf-sm.c:190)
open5gs_5gc       | 07/31 04:49:56.092: [sbi] INFO: (NRF-notify) NF registered [91b73ea6-4ee7-41ef-9170-ab378469ed59:1] (../lib/sbi/nnrf-handler.c:924)
open5gs_5gc       | 07/31 04:49:56.092: [sbi] INFO: [AMF] (NRF-notify) NF Profile updated [91b73ea6-4ee7-41ef-9170-ab378469ed59:1] (../lib/sbi/nnrf-handler.c:938)
open5gs_5gc       | 07/31 04:49:56.093: [sbi] INFO: [91b73ea6-4ee7-41ef-9170-ab378469ed59] NF registered [Heartbeat:10s] (../lib/sbi/nf-sm.c:221)
open5gs_5gc       | 07/31 04:49:56.093: [nrf] INFO: [91b7e496-4ee7-41ef-bcb8-0bdd9b9ee35e] Subscription created until 2024-08-01T04:49:56.093461+02:00 [validity_duration:86400] (../src/nrf/nnrf-handler.c:445)
open5gs_5gc       | 07/31 04:49:56.093: [nrf] INFO: [91b7e78e-4ee7-41ef-bcb8-0bdd9b9ee35e] Subscription created until 2024-08-01T04:49:56.093534+02:00 [validity_duration:86400] (../src/nrf/nnrf-handler.c:445)
open5gs_5gc       | 07/31 04:49:56.093: [nrf] INFO: [91b7eb58-4ee7-41ef-bcb8-0bdd9b9ee35e] Subscription created until 2024-08-01T04:49:56.093631+02:00 [validity_duration:86400] (../src/nrf/nnrf-handler.c:445)
open5gs_5gc       | 07/31 04:49:56.093: [nrf] INFO: [91b7ec5c-4ee7-41ef-bcb8-0bdd9b9ee35e] Subscription created until 2024-08-01T04:49:56.093656+02:00 [validity_duration:86400] (../src/nrf/nnrf-handler.c:445)
open5gs_5gc       | 07/31 04:49:56.093: [nrf] INFO: [91b7f080-4ee7-41ef-bcb8-0bdd9b9ee35e] Subscription created until 2024-08-01T04:49:56.093762+02:00 [validity_duration:86400] (../src/nrf/nnrf-handler.c:445)
open5gs_5gc       | 07/31 04:49:56.093: [nrf] INFO: [91b7f7ec-4ee7-41ef-bcb8-0bdd9b9ee35e] Subscription created until 2024-08-01T04:49:56.093952+02:00 [validity_duration:86400] (../src/nrf/nnrf-handler.c:445)
open5gs_5gc       | 07/31 04:49:56.094: [nrf] INFO: [91b7faf8-4ee7-41ef-bcb8-0bdd9b9ee35e] Subscription created until 2024-08-01T04:49:56.094030+02:00 [validity_duration:86400] (../src/nrf/nnrf-handler.c:445)
open5gs_5gc       | 07/31 04:49:56.094: [sbi] INFO: [91b7e496-4ee7-41ef-bcb8-0bdd9b9ee35e] Subscription created until 2024-08-01T04:49:56.093461+02:00 [duration:86400,validity:86399.998787,patch:43199.999393] (../lib/sbi/nnrf-handler.c:708)
open5gs_5gc       | 07/31 04:49:56.094: [sbi] INFO: [91b7e78e-4ee7-41ef-bcb8-0bdd9b9ee35e] Subscription created until 2024-08-01T04:49:56.093534+02:00 [duration:86400,validity:86399.998840,patch:43199.999420] (../lib/sbi/nnrf-handler.c:708)
open5gs_5gc       | 07/31 04:49:56.094: [sbi] INFO: [91b7eb58-4ee7-41ef-bcb8-0bdd9b9ee35e] Subscription created until 2024-08-01T04:49:56.093631+02:00 [duration:86400,validity:86399.998825,patch:43199.999412] (../lib/sbi/nnrf-handler.c:708)
open5gs_5gc       | 07/31 04:49:56.094: [sbi] INFO: [91b7ec5c-4ee7-41ef-bcb8-0bdd9b9ee35e] Subscription created until 2024-08-01T04:49:56.093656+02:00 [duration:86400,validity:86399.998838,patch:43199.999419] (../lib/sbi/nnrf-handler.c:708)
open5gs_5gc       | 07/31 04:49:56.094: [sbi] INFO: [91b7f080-4ee7-41ef-bcb8-0bdd9b9ee35e] Subscription created until 2024-08-01T04:49:56.093762+02:00 [duration:86400,validity:86399.998868,patch:43199.999434] (../lib/sbi/nnrf-handler.c:708)
open5gs_5gc       | 07/31 04:49:56.094: [sbi] INFO: [91b7f7ec-4ee7-41ef-bcb8-0bdd9b9ee35e] Subscription created until 2024-08-01T04:49:56.093952+02:00 [duration:86400,validity:86399.999004,patch:43199.999502] (../lib/sbi/nnrf-handler.c:708)
open5gs_5gc       | 07/31 04:49:56.095: [sbi] INFO: [91b7faf8-4ee7-41ef-bcb8-0bdd9b9ee35e] Subscription created until 2024-08-01T04:49:56.094030+02:00 [duration:86400,validity:86399.999009,patch:43199.999504] (../lib/sbi/nnrf-handler.c:708)
open5gs_5gc       | Open5GS daemon v2.7.0
open5gs_5gc       | 
open5gs_5gc       | 07/31 04:49:56.136: [app] INFO: Configuration: 'open5gs-5gc.yml' (../lib/app/ogs-init.c:130)
open5gs_5gc       | 07/31 04:49:56.136: [sbi] INFO: NF Service [nausf-auth] (../lib/sbi/context.c:1812)
open5gs_5gc       | 07/31 04:49:56.137: [sbi] INFO: nghttp2_server() [http://127.0.0.11]:7777 (../lib/sbi/nghttp2-server.c:414)
open5gs_5gc       | 07/31 04:49:56.137: [app] INFO: AUSF initialize...done (../src/ausf/app.c:31)
open5gs_5gc       | 07/31 04:49:56.137: [nrf] INFO: [91be8526-4ee7-41ef-9721-1dfeb3c929b4] NF registered [Heartbeat:10s] (../src/nrf/nf-sm.c:190)
open5gs_5gc       | 07/31 04:49:56.137: [sbi] INFO: (NRF-notify) NF registered [91be8526-4ee7-41ef-9721-1dfeb3c929b4:1] (../lib/sbi/nnrf-handler.c:924)
open5gs_5gc       | 07/31 04:49:56.137: [sbi] INFO: [AUSF] (NRF-notify) NF Profile updated [91be8526-4ee7-41ef-9721-1dfeb3c929b4:1] (../lib/sbi/nnrf-handler.c:938)
open5gs_5gc       | 07/31 04:49:56.138: [sbi] INFO: [91be8526-4ee7-41ef-9721-1dfeb3c929b4] NF registered [Heartbeat:10s] (../lib/sbi/nf-sm.c:221)
[... 105 lines cut ...]
open5gs_5gc       | 07/31 04:50:11.231: [amf] INFO: [suci-0-001-01-0000-0-0-0123456780] Unknown UE by SUCI (../src/amf/context.c:1835)
open5gs_5gc       | 07/31 04:50:11.232: [amf] INFO: [Added] Number of AMF-UEs is now 1 (../src/amf/context.c:1616)
open5gs_5gc       | 07/31 04:50:11.232: [gmm] INFO: Registration request (../src/amf/gmm-sm.c:1165)
open5gs_5gc       | 07/31 04:50:11.232: [gmm] INFO: [suci-0-001-01-0000-0-0-0123456780]    SUCI (../src/amf/gmm-handler.c:166)
open5gs_5gc       | 07/31 04:50:11.633: [gmm] INFO: [imsi-001010123456780] Registration complete (../src/amf/gmm-sm.c:2146)
open5gs_5gc       | 07/31 04:50:11.633: [amf] INFO: [imsi-001010123456780] Configuration update command (../src/amf/nas-path.c:612)
open5gs_5gc       | 07/31 04:50:11.633: [gmm] INFO:     UTC [2024-07-31T02:50:11] Timezone[0]/DST[0] (../src/amf/gmm-build.c:559)
open5gs_5gc       | 07/31 04:50:11.633: [gmm] INFO:     LOCAL [2024-07-31T04:50:11] Timezone[7200]/DST[1] (../src/amf/gmm-build.c:564)
open5gs_5gc       | 07/31 04:50:11.634: [amf] INFO: [Added] Number of AMF-Sessions is now 1 (../src/amf/context.c:2571)
open5gs_5gc       | 07/31 04:50:11.634: [gmm] INFO: UE SUPI[imsi-001010123456780] DNN[srsapn] S_NSSAI[SST:1 SD:0xffffff] smContextRef [NULL] (../src/amf/gmm-handler.c:1241)
open5gs_5gc       | 07/31 04:50:11.634: [gmm] INFO: No SMF Instance (../src/amf/gmm-handler.c:1278)
open5gs_5gc       | 07/31 04:50:11.638: [scp] ERROR: (NF discover) No NF-Instance [nsmf-pdusession:AMF] (../src/scp/sbi-path.c:786)
open5gs_5gc       | 07/31 04:50:11.639: [amf] ERROR: [1:0] No SmContextCreateError (../src/amf/nsmf-handler.c:126)
open5gs_5gc       | 07/31 04:50:11.639: [amf] WARNING: [suci-0-001-01-0000-0-0-0123456780] DL NAS transport (../src/amf/nas-path.c:1014)
[... cut ...]
open5gs_5gc       | 07/31 04:50:14.561: [amf] INFO: UE Context Release [Action:2] (../src/amf/ngap-handler.c:1698)
open5gs_5gc       | 07/31 04:50:14.561: [amf] INFO:     RAN_UE_NGAP_ID[0] AMF_UE_NGAP_ID[1] (../src/amf/ngap-handler.c:1699)
open5gs_5gc       | 07/31 04:50:14.561: [amf] INFO:     SUCI[suci-0-001-01-0000-0-0-0123456780] (../src/amf/ngap-handler.c:1702)
open5gs_5gc       | 07/31 04:50:14.561: [amf] INFO: [Removed] Number of gNB-UEs is now 0 (../src/amf/context.c:2557)
```

## Steps to reproduce the problem ##
Follow srsRAN_Project tutorial for ZMQ-based setup

andrepuschmann, 2024-07-31:
Seems like the attach fails and the AMF error causes the gNB to also release the UE immediately. Check the reason for `No NF-Instance`.

closed 2024-07-31 by andrepuschmann

xixihahahha, 2024-07-31:
> Seems like the attach fails and the AMF error causes the gNB to also release the UE immediately. Check the reason for `No NF-Instance`.

I follow srsRAN_Project tutorial for ZMQ-based setup step by step. Open5gs is  based on dcoker  and the configuration files are given on the official website  as follows.
I'm new to this and I don't know what went wrong.

[gnb_zmq.txt](https://github.com/user-attachments/files/16439162/gnb_zmq.txt)
[ue_zmq.txt](https://github.com/user-attachments/files/16439163/ue_zmq.txt)

pgawlowicz, 2024-08-07:
could you share log files?