srsran/srsRAN_Project #753: UE on usrp b210 not connecting to gNb on x310

facundopedre, 2024-08-01:
## Issue Description ##
Hi. We are currently using srsRANProject as Core + gNb on an Ubuntu 22.04 PC with an USRP x310 and srsRAN4g as UE with a b210 on another one. We have carefully followed the steps on  [https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html](https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html) for over the air transmission, and have now even connected both USRPs via coaxial cable with -30db attenuators on the Tx ports, so there is no air interference. We have also tried using both two x310 and two b210 configurations, connected via coax, and have had no chance making the UE attach to the gNb. We have also downloaded the latest versions for both srsRANProject and srsRAN4G and the latest FPGAs for the USRPs.
We are attaching our configurations as well as logs.

We don't have a GPSDO for synchronization available, but have tried to connect the UE sdr B210 to the "ref out" exported clock of the X310 sdr, with the same results as using internal clocks on both.

## Setup Details ##
gNb:
Ubuntu 22.04.4 LTS, 16 GB RAM, CPU i7 12700
usrp Ettus X310, fpga=39

gNb:
Ubuntu 22.04.4 LTS, 16 GB RAM, CPU i7 12700
usrp Ettus B210

## Expected Behavior ##
UE attaching to gNb and reachable by core. 

## Actual Behaviour ##
With sdr B210 in UE, we get this error repeatedly: Tx while waiting for EOB, timed out... ... Starting new burst...
Trying other sample rates like in #732 has'nt worked.

Also, the gNb gives this error while starting, after setting up CU-UP and DU succesfully:
Real-time failure in RF: Type=start-of-burst Source=tx Timestamp=4885080

## Steps to reproduce the problem ##
The configuration files used are attached below.

## Additional Information ##
Config files:
[gnb_x310_rf_fdd.yml.txt](https://github.com/user-attachments/files/16462093/gnb_x310_rf_fdd.yml.txt)
[ue_rf.conf.txt](https://github.com/user-attachments/files/16462095/ue_rf.conf.txt)
[open5gs.env.txt](https://github.com/user-attachments/files/16462097/open5gs.env.txt)

Logs:
[logs_ue_1_8.log](https://github.com/user-attachments/files/16462068/logs_ue_1_8.log)
[gnb_1_8.log](https://github.com/user-attachments/files/16462069/gnb_1_8.log)
[gnb_mac.pcap.txt](https://github.com/user-attachments/files/16462079/gnb_mac.pcap.txt)
[gnb_ngap.pcap.txt](https://github.com/user-attachments/files/16462080/gnb_ngap.pcap.txt)

Terminal outputs:
[gnb_terminal_output.txt](https://github.com/user-attachments/files/16462081/gnb_terminal_output.txt)
[ue_terminal_output.txt](https://github.com/user-attachments/files/16462087/ue_terminal_output.txt)

pgawlowicz, 2024-08-06:
Please set the cpu scaling governor to performance mode. 

CFO seems to be very high:
```
2024-08-01T21:11:14.017966 [RRC-NR ] [I] Proc "Cell Selection" - Cell search found ARFCN=0 PCI=1 epre=-1.0 snr=+17.8 cfo=+5490.2 delay=+0.0  sfn=414 ssb_idx=0 hrf=n scs=15 ssb_offset=6 dmrs_typeA_pos=pos2 coreset0=12 ss0=0 barred=n intra_freq_reselection=n spare=0
```

so the ue cannot sync to the network. you might need to use an external clock for the usrps.

facundopedre, 2024-08-06:
Hello, thanks for the input. We had already set the CPU governor to performance, using the command suggested in the documentation. Also, checking CPU usage using htop, the CPU doesn’t seem to be too stressed during the execution of either module.

Using the freq_offset parameter to fix the cfo manually, we did get the UE to attempt to connect, but after several prach attempts it seems to stop trying and hang while disconnected. We have attached the new logs as well as the output on the UE terminal below.

We do not have access to an external clock yet, but will try to get one to conduct further tests. Do you think it is absolutely necessary to have one anyway, or can it be compensated by configuring both ue and gnb appropiately? We have tried using the internal clock of the x310 as a reference for the b210 ue, but that doesn't seem to be recognized as a clock input when running said ue.

[gnb.log](https://github.com/user-attachments/files/16516533/gnb.log)
[gnb_x310_rf_fdd.yml.txt](https://github.com/user-attachments/files/16516536/gnb_x310_rf_fdd.yml.txt)

[ue_rf_06082024.conf.txt](https://github.com/user-attachments/files/16516538/ue_rf_06082024.conf.txt)
[ue_terminal_output_06082024.txt](https://github.com/user-attachments/files/16516539/ue_terminal_output_06082024.txt)
[ue_log_06082024.txt](https://github.com/user-attachments/files/16516537/ue_log_06082024.txt)

pgawlowicz, 2024-08-07:
seems that the gnb never receives the PRACH from srsUE. Could you try to tune srsUE txgain and gnb rx gain?

pgawlowicz, 2024-10-16:
@facundopedre any updates on this issue?

facundopedre, 2024-10-18:
Hello, sorry for the lack of updates. 
We ended up migrating away from this setup to one with two X310 both as gnb and ue to try a different approach, and in that case had to tune the transmission and reception gains very finely but did end up getting a proper connection. We believe the appropiate gain range of the X310 might have been the issue there, since it showed a tendency to saturate even on very low gain values. Also, those x310 were not synchronized with an external clock but each used their internal one and still worked, perhaps because theirs are more precise than the b210's.

We haven't had much success with this mixed setup of using both an X310 and b210 yet. It's not clear from the logs we have if its that same gain issue, or a synchronization issue as you suggested initially. 

If you believe that situation with only x310's clears anything up about this setup with mixed sdrs please let me know, as we'll still have to include the b210 in the future.

Thanks

pgawlowicz, 2024-10-21:
could you share your logs?

xiaoming124, 2024-10-21:
> Hello, sorry for the lack of updates. We ended up migrating away from this setup to one with two X310 both as gnb and ue to try a different approach, and in that case had to tune the transmission [...]

Sorry to bother, could you please share your logs and config files for the double X310 testbed? We met the same trouble in this issue. Thanks!!!

facundopedre, 2024-11-01:
Hello,
After receiving your message we had already changed to a different setup so we tried to remake our original one with the x310 to provide you with the logs from an example test run, but found that one of our x310s had broken down. We apologize for the delay, but after spending the last couple of weeks learning to reset the faulty x310 we have been able to re-run our old setup. We attach the configuration files and resulting logs and pcap captures.
[git_example.zip](https://github.com/user-attachments/files/17604586/git_example.zip)
As we said, most of the issues we had were solved by fine tuning the gain values in both ue and gnb.

closed 2024-11-02 by pgawlowicz

xiaoming124, 2024-11-02:
> Hello, After receiving your message we had already changed to a different setup so we tried to remake our original one with the x310 to provide you with the logs from an example test run, but found [...]

Thank you very much for your reply. Now we can establish connection between x310 UE and x310 gNB. We do observe that the gain values impact a lot for the connection stability. Currently we face another problem that the connection breaks down (RRC release) after a period of connection setup. The stable period is quite random (from several seconds to 1 minute), even using the same configurations for multiple tests. We are still looking for the reasons....

pgawlowicz, 2024-11-02:
@xiaoming124 could you send your logs?

xiaoming124, 2024-11-05:
> @xiaoming124 could you send your logs?

Thanks for your reply! We have tried the wired setup with attenuators and there is no error reported. So I think the problem would be the wireless environment, we will modify the gain values and setup ranges to find the problem. Many thanks!!

FabianJohnThl, 2024-11-19:
Hello,
we determined also an issue with the B210 and COTS UEs. Maybe the same cause. 

If we use a B210 and try to connect some UEs, some work and other not. For example the Galaxy Tab S9 5G connects immediately. The Samsung Galaxy S24 doesn't connect directly. Coming to the lab the next day, the S24 was connected. But after switching into FlightMode and back, again no connection was established.

In the next step, we disconnected B210 and used the B205-mini with exact the same configuration and all devices, the S24 also, connecting immediately.

In the setup, we use PPS external sync (for both setups, with the B210 and B205-mini).

We also checked the spectral output of the SSB and both look very similar and have the same center frequency.

Maybe, anyone have a guess why B210 is not working as expected. We were wondering, why the smaller B205-mini is working.