srsran/srsRAN_Project #757: Dynamic MCS adaptation in srsRAN for 5G simulation with ZMQ

RicGobs, 2024-08-02:
## Issue Description ##
I am using srsRAN ZMQ and I would like to understand how to observe dynamic MCS adaptation and throughput changes based on RSRP/path loss changes.

## Setup Details ##
I am using the tutorial for multiple UEs with GNU Radio integration (https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#zeromq-based-setup) .
I launch the Core, gNB, UEs and GNU-Radio Flowgraph, as told in the tutorial.
I launch iperf server and client for traffic between gNB and UE-1. 
The Iperf command that I am using is the one with UDP 10 Mbit/s (downlink) in the tutorial.

## Expected Behavior ##
I am expecting to have a throughput equal to 10M in the Downlink.
Changing the pathloss using the GUI I am expecting see changes in RSRP with reflection in MCS and throughput.

## Actual Behaviour ##
The throughput is fixed to 35 Mbit/s in the Downlink. 
RSRP is changing, but with no reflection in MCS and throughput.

## Additional Details ##
I tried to change the GNU-Radio Flowgraph, adding a Noise on the signal (adding a new GUI that allows you to change dinamically the noise).
I have seen that adding noise, there are changes in the MCS and throughput, but no changes in the RSRP. 

Why the throughput is 35 Mbit/s if I am only sending 10 Mbit/s?
How can I observe the change in rsrp, mcs and throughput when only pathloss is changed?

pgawlowicz, 2024-08-06:
MCS adaptation is triggered by the signal SNR changes. So you need to add a noise source to impact the signal SNR.
Why should the RSRP change if you are only adding noise? 
You need to control both noise source and Path Gain to control SNR and RPRP, respectively.

closed 2024-08-06 by pgawlowicz

RicGobs, 2024-08-29:
Thank you for the explanation, it makes perfect sense.
I modified the GNU Radio file accordingly, and it worked as expected. 

If it might help others, here is the flow graph I used.
[image_22_gnuradio_flowgraph.pdf](https://github.com/user-attachments/files/16801360/image_22_gnuradio_flowgraph.pdf)

lgs96, 2024-12-20:
Hi, is this work well? I'm trying to adjust snr, but signal attenuation only works when this goes higher than some threshold in my case.