srsran/srsRAN_Project #783: srsran docker repository does not exist?

dominikheinz, 2024-08-21:
## Issue Description ##
For some reason, today I was not able to pull the docker containers for gnb and metrics server.
I am being told the repository does not exist or may require docker login? (Which doesn't seem right, because all the repos are normally public?)

This is my docker file (I only added an additional service, the speedtest, the rest is more or less the same as provided here:

```
services:
  5gc:
    container_name: open5gs_5gc
    build:
      context: open5gs
      target: open5gs
      args:
        OS_VERSION: "22.04"
        OPEN5GS_VERSION: "v2.7.0"
    env_file:
      - ${OPEN_5GS_ENV_FILE:-open5gs/open5gs.env}
    privileged: true
    ports:
      - "9999:9999/tcp"
    command: 5gc -c open5gs-5gc.yml
    healthcheck:
      test: [ "CMD-SHELL", "nc -z 127.0.0.20 7777" ]
      interval: 3s
      timeout: 1s
      retries: 60
    networks:
      ran:
        ipv4_address: ${OPEN5GS_IP:-10.53.1.2}

  gnb:
    container_name: srsran_gnb
    image: srsran/gnb
    build:
      context: ..
      dockerfile: docker/Dockerfile
      args:
        OS_VERSION: "24.04"
    privileged: true
    cap_add:
      - SYS_NICE
      - CAP_SYS_PTRACE
    volumes:
      - /dev/bus/usb/:/dev/bus/usb/
      - /usr/share/uhd/images:/usr/share/uhd/images
      - gnb-storage:/tmp
      - ../configs/gnb_custom.yml:/gnb_config.yml:ro
    configs:
      - gnb_config.yml
    networks:
      ran:
        ipv4_address: ${GNB_IP:-10.53.1.3}
      metrics:
        ipv4_address: 172.19.1.3
    depends_on:
      5gc:
        condition: service_healthy
    command: gnb -c /gnb_config.yml amf --addr ${OPEN5GS_IP:-10.53.1.2} --bind_addr ${GNB_IP:-10.53.1.3}

  metrics-server:
    container_name: metrics_server
    image: srsran/metrics_server
    build:
      context: metrics_server
    environment:
      - PORT=${METRICS_SERVER_PORT}
      - BUCKET=${DOCKER_INFLUXDB_INIT_BUCKET}
      - TESTBED=default
      - URL=http://${DOCKER_INFLUXDB_INIT_HOST}:${DOCKER_INFLUXDB_INIT_PORT}
      - ORG=${DOCKER_INFLUXDB_INIT_ORG}
      - TOKEN=${DOCKER_INFLUXDB_INIT_ADMIN_TOKEN}
    ports:
      - 55555:${METRICS_SERVER_PORT}/udp
    networks:
      metrics:
        ipv4_address: 172.19.1.4

  influxdb:
    container_name: influxdb
    image: influxdb:${DOCKER_INFLUXDB_VERSION}
    volumes:
      - influxdb-storage:/var/lib/influxdb2:rw
    env_file:
      - .env
    restart: on-failure:10
    networks:
      metrics:
        ipv4_address: 172.19.1.5

  grafana:
    container_name: grafana
    image: srsran/grafana
    build:
      context: grafana
    volumes:
      - grafana-storage:/var/lib/grafana:rw
    env_file:
      - .env
    depends_on:
      - influxdb
      - metrics-server
    ports:
      - 3300:${GRAFANA_PORT}
    networks:
      metrics:
        ipv4_address: 172.19.1.6

  speedtest:
    container_name: speedtest
    image: ghcr.io/librespeed/speedtest:latest
    restart: always
    environment:
      MODE: standalone
      TITLE: "5G SA Net - LibreSpeed"
    ports:
      - "80:80" 

configs:
  gnb_config.yml:
    file: ${GNB_CONFIG_PATH:-../configs/gnb_custom.yml}

volumes:
  gnb-storage:
  grafana-storage:
  influxdb-storage:

networks:
  ran:
    ipam:
      driver: default
      config:
        - subnet: 10.53.1.0/24
  metrics:
    ipam:
      driver: default
      config:
        - subnet: 172.19.1.0/24
```

## Setup Details ##
ArchLinux 6.10.4-arch2-1

## Expected Behavior ##
All images can be pulled and the containers start.

## Actual Behaviour ##
Pulling the images for `gnb` and `metrics_server` fails:

```
❯ docker compose up     
[+] Running 2/2
 ! grafana Warning pull access denied for srsran/grafana, repository does not exist or may require 'docker login': denied: requested access to the resource is denie...                  1.3s 
 ! gnb Warning     pull access denied for srsran/gnb, repository does not exist or may require 'docker login': denied: requested access to the resource is denied                        1.3s 
[+] Building 0.1s (3/3) FINISHED                                                                                                                                               docker:default
 => [5gc internal] load build definition from Dockerfile                                                                                                                                 0.0s
 => => transferring dockerfile: 3.11kB                                                                                                                                                   0.0s
 => WARN: FromAsCasing: 'as' and 'FROM' keywords' casing do not match (line 10)                                                                                                          0.0s
 => WARN: FromAsCasing: 'as' and 'FROM' keywords' casing do not match (line 88)                                                                                                          0.0s
 => [grafana internal] load build definition from Dockerfile                                                                                                                             0.0s
 => => transferring dockerfile: 2B                                                                                                                                                       0.0s
 => CANCELED [5gc internal] load metadata for docker.io/library/ubuntu:22.04                                                                                                             0.0s
failed to solve: failed to read dockerfile: open Dockerfile: no such file or directory
```

## Steps to reproduce the problem ##
1. Use my docker compose script.
2. I am on commit 1483bda
3. Run docker compose up and observe the error

## Additional Information ##
The same script worked fine on a different system yesterday..

BurmazDaniel, 2024-08-24:
I think you are looking for

`docker pull softwareradiosystems/srsran-project`
`docker pull softwareradiosystems/grafana`

dominikheinz, 2024-08-24:
> I think you are looking for
> 
> `docker pull softwareradiosystems/srsran-project` `docker pull softwareradiosystems/grafana`

Thanks. I will have a look on monday.
So, did the docker image names change then? I assume in that case the docker-compose file needs to be updated?
Unless I am missing something.

pgawlowicz, 2024-09-19:
here docker-compose builds its own images as there is `build:` parameter in the `grapfana` service.

closed 2024-09-19 by pgawlowicz