srsran/srsRAN_Project #788: E2setup Response issue in OSC RIC

aliasadi2001, 2024-08-26:
Hello everyone,

I’m working on connecting multiple gNBs with multiple UEs (using GNU Radio) to Open5GS and OSC RIC. All components are functioning correctly, but I’m encountering an issue where I do not receive an E2 setup response from the RIC.

I’ve attached a screenshot showing the Wireshark capture for port 36421 messages, as well as terminal outputs from both the OSC RIC and the gNB.

[image]

[image]

[image]

For context, I’m using [this ](https://github.com/srsran/oran-sc-ric/blob/main/e2-agents/srsRAN/gnb_zmq.yaml)configuration file for the gNB with an added gNB-ID to support multiple gNBs.

Could anyone assist me with resolving this issue?

Thank you very much!

aliasadi2001, 2024-08-26:
Yes, I am running both components on the same machine, but they are configured with different IP addresses.

I have attached screenshots of the logs from both components:

[image]

[image]

It appears that the issue is not related to running multiple gNBs, as I am also unable to receive the E2 setup response even when running a single gNB.

pgawlowicz, 2024-08-26:
just checked it with one gnb and it works without any issues. Could you share your configs and logs? Which IPs did you use for e2 in gnbs?

aliasadi2001, 2024-08-26:
Yes, of course. This is the GNB configuration file:

[This configuration file.txt](https://github.com/user-attachments/files/16750389/Thisconfiguration.file.txt).

I rebuilt the OSC RIC and deleted the old images, but now I can't get the E2Setup response in the RIC terminal.

Thank you for your response

pgawlowicz, 2024-08-26:
did you connect srsUE to both gNB? It is requited as the radio device (here zmq-based RF) drives all timers in the gNB stack.

aliasadi2001, 2024-08-26:
I connected 3 UEs to each gNB, and I have 2 gNBs. All UE terminals connected successfully, but the OSC RIC doesn't display any logs. I ran only the OSC RIC and Open5GS with just one gNB, without any UEs or other gNBs, but the issue still exists.

pgawlowicz, 2024-08-26:
ok, could you share gnb logs?

aliasadi2001, 2024-08-26:
The gNB terminal only shows the following logs (according to the documentation, it is working correctly).
[image]

aliasadi2001, 2024-08-26:
https://docs.srsran.com/projects/project/en/latest/tutorials/source/near-rt-ric/source/index.html#:~:text=%2D%2D%3D%3D%20srsRAN,to%20view%20trace

pgawlowicz, 2024-08-27:
could you share `/tmp/gnb.log` file? and also e2 pcap file?

aliasadi2001, 2024-08-27:
yes
the files provided in below
[gnb.log](https://github.com/user-attachments/files/16758517/gnb.log)
[gnb_e2ap.zip](https://github.com/user-attachments/files/16758548/gnb_e2ap.zip)

pgawlowicz, 2024-08-27:
did you connect srsUE over zmq to this gnb?

aliasadi2001, 2024-08-27:
These files are associated with a setup involving just one gNB (gNodeB), Open5GS, and OSC RIC, with no connected UE or additional components.

aliasadi2001, 2024-08-27:
I implemented this setup on a new VM with only one gNB and without any UE.

pgawlowicz, 2024-08-27:
when running with the ZMQ-based RF-device you have to connect the srsUE. Otherwise, the timers inside the srsgnb stack won't be advanced.

aliasadi2001, 2024-08-27:
I changed the all IPs using this command:
git grep -l "10.0.2." | xargs sed -i 's/10\.0\.2\./192\.168\.63\./g'
and it is working correctly now.
thank you

closed 2024-08-27 by pgawlowicz