srsran/srsRAN_Project #792: SINR Measurement and MCS Calculation

MaryamAmini, 2024-08-27:
Hello srsRAN team,

Recently I've been trying to understand the procedure by which the SNR/SINR is measured and how MCS values are then calculated based on that. I've done a few tests on uplink, and the logs on PUSCH give me three SINR values: sinr_ch_est, sinr_eq[sel], and sinr_evm. 

[image]

It seems like "sinr_eq[sel]" is the one that's actually reported as PUSCH SINR. However, looking at the MCS table at "srsRAN_Project/lib/scheduler/support/mcs_calculator.cpp" the calculated MCS is way lower than what it should be based on the value of sinr_eq[sel] (I'm looking at ul_snr_mcs_table). So I have a couple of questions that I hope you can help me with:
1. How do you define these three SINR values?
2. Does OLLA happen before MCS calculation, and that's why there is a mismatch between what I read for SINR, and what I see for MCS?

Your help is greatly appreciated :).
Regards,
Maryam.

dvdgrgrtt, 2024-09-20:
Hi,
the three SINR are estimated at the channel estimator (simply as the ratio between the estimated received signal power and noise power), after the equalizer (that is, when the fading has been removed and after combining contributions from all antenna ports) and as a derived metric from the EVM estimator (this is an approximation that was useful at some point, but doesn't have a rigorous definition).

MaryamAmini, 2024-09-20:
Hello @dvdgrgrtt and @herlesupreeth,
Thank you so much for the response.
I went through the code, and got some answers to my questions.
based on my understanding, the PUSCH SINR that is reported in the trace, is "sinr_eq", whereas the SINR that is used for MCS calculation is "sinr_evm". Now, when I look at the logs for a given speedtest, "sinr_eq" seems to be around 3-4 dB higher than the "sinr_evm", and that's why I was asking about the mismatch of the MCS table and the SINR that I saw on the trace.
Anyways, Thank you guys for the help :)
Maryam.

closed 2024-09-20 by MaryamAmini

HisheUW, 2024-10-09:
> Hi,
> the three SINR are estimated at the channel estimator (simply as the ratio between the estimated received signal power and noise power), after the equalizer (that is, when the fading has been [...]

Can you please let me know where the calculations of these SINRs are done in the code?

dvdgrgrtt, 2024-10-10:
Sure @HisheUW.
For the SINR estimated by the channel estimator, look at the code leading up to line 281 [here](https://github.com/srsran/srsRAN_Project/blob/e73b4618204151e6b95c142bec5798f5bc21b2b2/lib/phy/upper/signal_processors/port_channel_estimator_average_impl.cpp#L281).
For the SINR estimated form the EVM, you can start exploring from [here](https://github.com/srsran/srsRAN_Project/blob/e73b4618204151e6b95c142bec5798f5bc21b2b2/lib/phy/upper/channel_processors/pusch/pusch_demodulator_impl.cpp#L270).
And for the EVM-based heuristic, a good starting point is [this one](https://github.com/srsran/srsRAN_Project/blob/e73b4618204151e6b95c142bec5798f5bc21b2b2/lib/phy/upper/channel_processors/pusch/pusch_processor_notifier_adaptor.h#L161).