srsran/srsRAN_Project #799: FAPI debug logs missing

RafalArciszewski, 2024-09-02:
## Issue Description ##
In the gnb.log there are missing FAPI debug logs after the upgrade to latest commit. There is only information about PDSCH precoding matrix, but other messages like slot.indication or DL_TTI.request UL_TTI.request are gone.

## Setup Details ##
yml configuration:
```
log:
  filename: /tmp/gnb_ra_latest.log     # Path of the log file.
  #all_level: warning                  # Accepted values [debug,info,warning,error]
  # FAPI DEBUG
  lib_level: error
  rrc_level: error
  ngap_level: error
  sec_level: error
  pdcp_level: error
  sdap_level: error
  gtpu_level: error
  cu_level: error
  mac_level: error
  rlc_level: error
  f1ap_level: error
  f1u_level: error
  du_level: error
  phy_level: error
  fapi_level: debug
  ofh_level: error
  radio_level: error
```

## Expected Behavior ##
Logs from commit f3ed07a5a6f04245a57ca667194e1a3bf9684149
```
2024-09-02T12:36:06.957025 [FAPI    ] [D] [  639.19] UL_TTI.request slot=639.19
2024-09-02T12:36:06.957505 [FAPI    ] [D] [   640.0] Slot.indication
2024-09-02T12:36:06.957535 [FAPI    ] [D] [   640.0] DL_TTI.request slot=640.0, is_last_message_in_slot=true
        - SSB pointA=0 L_max=8 pci=1 k_SSB=0
2024-09-02T12:36:06.958006 [FAPI    ] [D] [   640.1] Slot.indication
2024-09-02T12:36:06.958052 [FAPI    ] [D] [   640.1] Omnidirectional PDSCH precoding matrix, nof_layers=1
2024-09-02T12:36:06.958060 [FAPI    ] [D] [   640.1] DL_TTI.request slot=640.1, is_last_message_in_slot=false
        - PDCCH bwp=0:24 symb=0:2 nof_dcis=1
        - PDSCH rnti=0xffff bwp=0:24 symb=2:12 CW: tbs=101 mod=2 rv_idx=0
2024-09-02T12:36:06.958088 [FAPI    ] [D] [   640.1] Tx_Data.request slot=640.1 nof_pdus=1
2024-09-02T12:36:06.958503 [FAPI    ] [D] [   640.2] Slot.indication
2024-09-02T12:36:06.958538 [FAPI    ] [D] [   640.2] DL_TTI.request slot=640.2, is_last_message_in_slot=true
        - NZP-CSI-RS crbs=0:24 row=1 symbL0=4 symbL1=2 scramb_id=1
        - NZP-CSI-RS crbs=0:24 row=1 symbL0=8 symbL1=2 scramb_id=1
2024-09-02T12:36:06.959006 [FAPI    ] [D] [   640.3] Slot.indication
2024-09-02T12:36:06.959039 [FAPI    ] [D] [   640.3] DL_TTI.request slot=640.3, is_last_message_in_slot=true
        - NZP-CSI-RS crbs=0:24 row=1 symbL0=4 symbL1=2 scramb_id=1
        - NZP-CSI-RS crbs=0:24 row=1 symbL0=8 symbL1=2 scramb_id=1
2024-09-02T12:36:06.959506 [FAPI    ] [D] [   640.4] Slot.indication
2024-09-02T12:36:06.959553 [FAPI    ] [D] [   640.4] DL_TTI.request slot=640.4, is_last_message_in_slot=true
        - ZP-CSI-RS crbs=0:24 row=3 symbL0=8 symbL1=2
        - ZP-CSI-RS crbs=0:24 row=3 symbL0=8 symbL1=2
        - NZP-CSI-RS crbs=0:24 row=3 symbL0=4 symbL1=2 scramb_id=1
```

## Actual Behaviour ##
Logs from latest commit: 4ac5300d4927b5199af69e6bc2e55d061fc33652
```
2024-09-02T13:48:31.118022 [FAPI    ] [D] [   880.1] Omnidirectional PDSCH precoding matrix, nof_layers=1
2024-09-02T13:48:31.278029 [FAPI    ] [D] [   896.1] Omnidirectional PDSCH precoding matrix, nof_layers=1
2024-09-02T13:48:31.438019 [FAPI    ] [D] [   912.1] Omnidirectional PDSCH precoding matrix, nof_layers=1
2024-09-02T13:48:31.598025 [FAPI    ] [D] [   928.1] Omnidirectional PDSCH precoding matrix, nof_layers=1
2024-09-02T13:48:31.758022 [FAPI    ] [D] [   944.1] Omnidirectional PDSCH precoding matrix, nof_layers=1
2024-09-02T13:48:31.918023 [FAPI    ] [D] [   960.1] Omnidirectional PDSCH precoding matrix, nof_layers=1
2024-09-02T13:48:32.078020 [FAPI    ] [D] [   976.1] Omnidirectional PDSCH precoding matrix, nof_layers=1
2024-09-02T13:48:32.238024 [FAPI    ] [D] [   992.1] Omnidirectional PDSCH precoding matrix, nof_layers=1
2024-09-02T13:48:32.398028 [FAPI    ] [D] [  1008.1] Omnidirectional PDSCH precoding matrix, nof_layers=1
2024-09-02T13:48:32.558017 [FAPI    ] [D] [     0.1] Omnidirectional PDSCH precoding matrix, nof_layers=1
2024-09-02T13:48:32.718019 [FAPI    ] [D] [    16.1] Omnidirectional PDSCH precoding matrix, nof_layers=1
2024-09-02T13:48:32.878018 [FAPI    ] [D] [    32.1] Omnidirectional PDSCH precoding matrix, nof_layers=1
2024-09-02T13:48:33.038028 [FAPI    ] [D] [    48.1] Omnidirectional PDSCH precoding matrix, nof_layers=1
2024-09-02T13:48:33.198023 [FAPI    ] [D] [    64.1] Omnidirectional PDSCH precoding matrix, nof_layers=1
2024-09-02T13:48:33.358022 [FAPI    ] [D] [    80.1] Omnidirectional PDSCH precoding matrix, nof_layers=1
2024-09-02T13:48:33.518015 [FAPI    ] [D] [    96.1] Omnidirectional PDSCH precoding matrix, nof_layers=1
```

## Steps to reproduce the problem ##
upgrade from commit f3ed07a5a6f04245a57ca667194e1a3bf9684149 to f3ed07a5a6f04245a57ca667194e1a3bf9684149
setup `fapi_level = debug` in yml config.

## Additional Information ##
none

herlesupreeth, 2024-09-02:
Thanks @RafalArciszewski for raising this issue. We are aware of this issue and the fix for this issue will available in the next release

RafalArciszewski, 2024-09-03:
Good to know! Thanks @herlesupreeth

pgawlowicz, 2024-10-16:
@RafalArciszewski could you pull the newest code and check?

RafalArciszewski, 2024-10-24:
@pgawlowicz Problem solved! Thanks!

closed 2024-10-24 by RafalArciszewski