srsran/srsRAN_Project #801: gnb does not support IPv6

PiscesDR, 2024-09-03:
Hi there,

I'm running srsgnb in ZMQ mode with open5gs. When I configure IPv4 address in gnb configuration file, it works fine, but it seems that it  does not support IPv6. Is there a solution?

Thank you!

pgawlowicz, 2024-09-03:
did you enable ipv6 in open5gs config?

PiscesDR, 2024-09-03:
> 你在 open5gs 配置中启用了 ipv6 吗？

Yes, I enable IPv6 in open5gs, and when I configure it as follows:
[yaml.zip](https://github.com/user-attachments/files/16851054/yaml.zip)

I am able to successfully establish the PDU session, and get the following output:

RRC Connected
PDU Session Establishment successful. IP: 10.45.10.2
RRC NR reconfiguration successful.

In the above configuration, the ngap address of AMF, the gtpu address of UPF and SMF, and the gnb address are still IPv4. However, when I change them to IPv6, I get the error.

pgawlowicz, 2024-09-19:
could you provide ngap pcap file?

pgawlowicz, 2024-10-16:
@PiscesDR any update on this issue? most likely srsUE does not support ipv6. Could you check in NGAP pcap the UE request?

closed 2024-10-30 by pgawlowicz