srsran/srsRAN_Project #837: RRC connects but releases after a few seconds

seoyuloh, 2024-09-28:
## Issue Description ##
I am using two x310 USRPs to follow this tutorial. (https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html)

My UE gets connected to the gNB and then gets released shortly after. When the connection is maintained, I have good rsrp and mcs. 

I have also looked into this issue (https://github.com/srsran/srsRAN_Project/issues/533) and added  cu_cp:
  inactivity_timer: 7200

to my config file. It does not help.

I cannot open this https://github.com/srsran/srsRAN_Project/tree/fix_disconnect to check if there's anything more I can do. Could someone please help? 

## Setup Details ##
Two USRP X310s
Ubuntu 22.04
I am using the open5gs docker inside the project as instructed. 
I have attached my configuration files both for UE and gNB. 

## Actual Behaviour ##
I have attached my gnb.log and ue.log files. 

Here is the log of open5gs.
open5gs_5gc  | 09/28 05:27:15.137: [smf] INFO: UE SUPI[imsi-001010123456780] DNN[srsapn] IPv4[10.45.1.2] IPv6[] (../src/smf/npcf-handler.c:539)
open5gs_5gc  | 09/28 05:27:15.137: [upf] INFO: [Added] Number of UPF-Sessions is now 1 (../src/upf/context.c:208)
open5gs_5gc  | 09/28 05:27:15.137: [upf] INFO: UE F-SEID[UP:0xe2a CP:0xd34] APN[srsapn] PDN-Type[1] IPv4[10.45.1.2] IPv6[] (../src/upf/context.c:485)
open5gs_5gc  | 09/28 05:27:15.137: [upf] INFO: UE F-SEID[UP:0xe2a CP:0xd34] APN[srsapn] PDN-Type[1] IPv4[10.45.1.2] IPv6[] (../src/upf/context.c:485)
open5gs_5gc  | 09/28 05:27:15.565: [amf] INFO: [imsi-001010123456780:1:11][0:0:NULL] /nsmf-pdusession/v1/sm-contexts/{smContextRef}/modify (../src/amf/nsmf-handler.c:837)
open5gs_5gc  | 09/28 05:27:15.690: [gmm] INFO: [imsi-001010123456780] No GUTI allocated (../src/amf/gmm-sm.c:1443)
open5gs_5gc  | 09/28 05:28:11.189: [amf] INFO: [imsi-001010123456780:1:13][0:0:NULL] /nsmf-pdusession/v1/sm-contexts/{smContextRef}/modify (../src/amf/nsmf-handler.c:837)
open5gs_5gc  | 09/28 05:28:11.343: [amf] INFO: UE Context Release [Action:2] (../src/amf/ngap-handler.c:1698)
open5gs_5gc  | 09/28 05:28:11.343: [amf] INFO:     RAN_UE_NGAP_ID[0] AMF_UE_NGAP_ID[23] (../src/amf/ngap-handler.c:1699)
open5gs_5gc  | 09/28 05:28:11.343: [amf] INFO:     SUCI[suci-0-001-01-0000-0-0-0123456780] (../src/amf/ngap-handler.c:1702)
open5gs_5gc  | 09/28 05:28:11.343: [amf] INFO: [Removed] Number of gNB-UEs is now 0 (../src/amf/context.c:2557)
[uelog.txt](https://github.com/user-attachments/files/17172148/uelog.txt)
[gnblog.txt](https://github.com/user-attachments/files/17172150/gnblog.txt)
[gnb_uhd_yml.txt](https://github.com/user-attachments/files/17172153/gnb_uhd_yml.txt)
[ue_uhd_conf.txt](https://github.com/user-attachments/files/17172154/ue_uhd_conf.txt)

pgawlowicz, 2024-10-07:
There are lates and underflows in the gnb, do you run both gnb and srsue on the same host pc? did you run this script? https://github.com/srsran/srsRAN_Project/blob/main/scripts/srsran_performance

pgawlowicz, 2024-10-16:
@seoyuloh any update on this issue?

closed 2024-10-30 by pgawlowicz