srsran/srsRAN_Project #846: O-RAN NearRT-RIC and XApp Tutorial: Missing gNB executable

rcbarke, 2024-10-04:
## Issue Description ##
Hello, I am learning how to leverage SRSRAN with NearRT-RIC and xApp software. To that effect, I am building the below tutorial on a virtual machine:
https://docs.srsran.com/projects/project/en/latest/tutorials/source/near-rt-ric/source/index.html

In the **gNB** section, the tutorial notes: 
_"We run gNB directly from the build folder (the config file is also located there) with the following command:
# when running with ORAN SC RIC
sudo ./gnb -c gnb_zmq.yaml e2 --addr="10.0.2.10" --bind_addr="10.0.2.1""_

This does not explicitly specify which build folder to leverage. As seen below in the "additional information" section, I do not see the gNB executable nor the gnb yml file located within either the srsRAN4G or srsRAN Project build directories, so I am unable to build the gNB with ZMQ.

Where are these files located? Is there additional compilation or build needed to generate them?

## Setup Details ##
Please refer to the linked tutorial. I am running all software dependencies described at the link including:
1. Ubuntu 22.04.1 LTS
2. srsRAN Project
3. srsRAN Project Dependencies
4. srsRAN4G/srsRAN UE Dependencies
5. ZeroMQ (built from source)
6. Open5GS 5G Core (already successfully deployed)
7. ORAN SC RIC (already successfully deployed) 

## Expected Behavior ##
As per the tutorial, these files should be present in the ambiguous build directory. I suspect they are intended to be sourced from SRSRAN Project.

## Actual Behavior ##
Earlier in the tutorial, when compiling SRSRAN Project, the tutorial explicitly notes to ensure the below CLI output is present. As seen in the "Additional Information" section, this shell output is present, however, the gNB files are not.
```
...
-- FINDING ZEROMQ.
-- Checking for module 'ZeroMQ'
--   No package 'ZeroMQ' found
-- Found libZEROMQ: /usr/local/include, /usr/local/lib/libzmq.so
...
```

## Steps to reproduce the problem ##
I have not deviated from the tutorial I linked. I opted to deploy both the Open5GS Core and NearRT-RIC with docker without issue. I have not attempted to deploy UEs, IP traffic, or xApps given I have not deployed a gNB yet.

## Additional Information ##
[shelloutput.txt](https://github.com/user-attachments/files/17264101/shelloutput.txt)

pgawlowicz, 2024-10-07:
Hi, Thanks for pointing this issue out -- I will update the tutorial. 
You need to run gnb binary from the `srsRAN_Project/build/apps/gnb/` directory.

rcbarke, 2024-10-07:
Thanks for the file tree, Piotr! I was able to locate the gNB executable file. The config yaml file was missing from this directory, however, I copied it in from the tutorial. However, I am now running into:
`root@OSC-Core:/home/ric/srsRAN_Project/build/apps/gnb# sudo ./gnb -c gnb_zmq.yaml e2 --addr="10.0.2.10" --bind_addr="10.0.2.1"

--== srsRAN gNB (commit ee1d86cd8) ==--

INI was not able to parse amf.++
Run with --help for more information.`

I am building my instance on a Ubuntu 22.04 LTS VirtualBox instance with the default IP address of 10.0.2.15 on the host machine. Open5GS core and OSC RIC are already successfully deployed. I have attempted building the gNB by substituting this address into both CLI parameters. I have not modified the yaml file given the tutorial advises against this.

Also, I note the troubleshooting section which mentions modifying IP addresses in the even the core and RIC are deployed on different VMs than the gNB. I am testing with everything collocated on one machine for now, but do have intentions to scale to more complex network deployments involving multiple VMs and virtual private networks in the future. 

So, three questions:
1. What do I need to modify given my current setup to deploy the gNB on the same machine as the core and RIC?
2. When I scale to multiple VMs, do I need to account for #1 to deploy the gNBs? 
3. When I scale to multiple VMs, am I able to simultaneously deploy multiple gNBs, both collocated with the core machine VM and on separate VMs? Or is running only one gNB per simulated network currently supported?

Thank you for your help!

rcbarke, 2024-10-07:
> Hi, Thanks for pointing this issue out -- I will update the tutorial. You need to run gnb binary from the `srsRAN_Project/build/apps/gnb/` directory.

Thanks for the file tree, Piotr! I was able to locate the gNB executable file. The config yaml file was missing from this directory, however, I copied it in from the tutorial. However, I am now running into:
`root@OSC-Core:/home/ric/srsRAN_Project/build/apps/gnb# sudo ./gnb -c gnb_zmq.yaml e2 --addr="10.0.2.10" --bind_addr="10.0.2.1"

--== srsRAN gNB (commit https://github.com/srsran/srsRAN_Project/commit/ee1d86cd80841f47061bbdbe25786e3702059937) ==--

INI was not able to parse amf.++
Run with --help for more information.`

I am building my instance on a Ubuntu 22.04 LTS VirtualBox instance with the default IP address of 10.0.2.15 on the host machine. Open5GS core and OSC RIC are already successfully deployed. I have attempted building the gNB by substituting this address into both CLI parameters. I have not modified the yaml file given the tutorial advises against this.

Also, I note the troubleshooting section which mentions modifying IP addresses in the even the core and RIC are deployed on different VMs than the gNB. I am testing with everything collocated on one machine for now, but do have intentions to scale to more complex network deployments involving multiple VMs and virtual private networks in the future.

So, three questions:

1. What do I need to modify given my current setup to deploy the gNB on the same machine as the core and RIC?
2. When I scale to multiple VMs, do I need to account for https://github.com/srsran/srsRAN_Project/issues/1 to deploy the gNBs?
3. When I scale to multiple VMs, am I able to simultaneously deploy multiple gNBs, both collocated with the core machine VM and on separate VMs? Or is running only one gNB per simulated network currently supported?

Thank you for your help!

pgawlowicz, 2024-10-07:
please check this answer
https://github.com/srsran/srsRAN_Project/discussions/92#discussioncomment-10868505

rcbarke, 2024-10-08:
I was able to complete the tutorial on one device with the modified gNB configuration file. Thank you, Piotr! 

I do have one further inquiry about scaling the architecture to build UEs across a virtual private network. I will make this known issue resolved and open another issue for that question. There seems to be a conflict with a bridge adapter installed by SRSRAN on the virtual private network preventing connectivity to another VM.

> please check this answer [#92 (reply in thread)](https://github.com/srsran/srsRAN_Project/discussions/92#discussioncomment-10868505)