srsran/srsRAN_Project #853: OSC RIC: E2setup Response Failure

rcbarke, 2024-10-09:
## Issue Description ##
Please see this issue, closed ambiguously last August:
https://github.com/srsran/srsRAN_Project/issues/788

I am experiencing the exact same problem described here, preventing xApp deployment or development.

All other components of my implementation Open5GS 5GC, SRSGNB, SRSUE, and GNURadio are functional. I am able to deploy multiple UEs on the RAN network within a single VM, however, I am currently unable to connect the RAN gNB to the RIC.

## Setup Details ##
Following this sequence of steps:
https://docs.srsran.com/projects/project/en/latest/tutorials/source/near-rt-ric/source/index.html

During this set of steps, the RIC deploys correctly:
`ric_submgr          | RMR is ready now ...`

The gNB deploys correctly:
```
ric@OSC-Core:~/srsRAN_Project/build/apps/gnb$ sudo ./gnb -c gnb_zmq.yaml e2 --addr="10.0.2.10" --bind_addr="10.0.2.1"

--== srsRAN gNB (commit ee1d86cd8) ==--

The PRACH detector will not meet the performance requirements with the configuration {Format 0, ZCZ 0, SCS 1.25kHz, Rx ports 1}.
Lower PHY in executor blocking mode.
Connecting to NearRT-RIC on 10.0.2.10:36421
Available radio types: zmq.
Cell pci=1, bw=10 MHz, 1T1R, dl_arfcn=368500 (n3), dl_freq=1842.5 MHz, dl_ssb_arfcn=368410, ul_freq=1747.5 MHz

N2: Connection to AMF on 10.53.1.2:38412 completed
==== gNB started ===
Type <h> to view help
```

## Expected Behavior ##
At this point, the ric_rtmgr_sim should log:
`ric_rtmgr_sim       | 2024/04/02 11:07:39 POST /ric/v1/handles/associate-ran-to-e2t  body: [{"E2TAddress":"10.0.2.10:38000","ranNamelist":["gnb_001_001_00019b"]}] elapsed: 10.77µs`

## Actual Behaviour ##
**The RIC never acknowledges the gNB's subscription request. However, the gNB logs reflects registering on the shared E2 link.**

## Steps to reproduce the problem ##
The [rtmgr git repo](https://github.com/o-ran-sc/ric-plt-e2mgr/tree/master/tools/RoutingManagerSimulator) has not been updated since 2019 by AT&T/Nokia. 

I note that the docker deployment of this RIC is setting up a bridged network adapter for the E2 link between the RIC and gNB on 10.0.2.1. I believe this may be conflicting with the virtual machine's subnet. The original issue seems to suggest replacing these addresses within the git repo into an unused subnet. Wouldn't this break communication with the gNB if its configuration isn't also updated?

Is there a known issue with the OSC RIC's E2 interface? I note several research efforts (ex: OAIC, NS-O-RAN, etc.) are modifying the E2Term/E2 Interface. I have not made any modifications to the RIC.

rcbarke, 2024-10-09:
Note this log with the e2 docker logs:
`{"ts":1728489330638,"crit":"ERROR","id":"E2Terminator","mdc":{"PID":"1","POD_NAME":"e2term_pod","CONTAINER_NAME":"ric_e2term","SERVICE_NAME":"ric_e2term_service","HOST_NAME":"ric_e2term_host","SYSTEM_NAME":"oran_sc_ric"},"msg":"Error sending E2_TERM_INIT cause : RMR_ERR_NOENDPT - send//call could not find an endpoint based on msg type "}`

Full log files are attached.
[logs.zip](https://github.com/user-attachments/files/17318474/logs.zip)

rcbarke, 2024-10-09:
I am also wondering if the RCA of both of these issues are related. Are the RIC messages being intercepted by the VirtualBox subnet?
https://github.com/srsran/srsRAN_Project/issues/849

rcbarke, 2024-10-09:
Here are runtime logs:
```
open5gs_5gc  | 10/09 20:56:53.866: [amf] INFO: [Added] Number of gNB-UEs is now 1 (../src/amf/context.c:2550)
open5gs_5gc  | 10/09 20:56:53.866: [amf] INFO:     RAN_UE_NGAP_ID[0] AMF_UE_NGAP_ID[1] TAC[7] CellID[0x66c000] (../src/amf/ngap-handler.c:562)
open5gs_5gc  | 10/09 20:56:53.866: [amf] INFO: [suci-0-001-01-0000-0-0-0123456780] Unknown UE by SUCI (../src/amf/context.c:1835)
open5gs_5gc  | 10/09 20:56:53.866: [amf] INFO: [Added] Number of AMF-UEs is now 1 (../src/amf/context.c:1616)
open5gs_5gc  | 10/09 20:56:53.866: [gmm] INFO: Registration request (../src/amf/gmm-sm.c:1165)
open5gs_5gc  | 10/09 20:56:53.866: [gmm] INFO: [suci-0-001-01-0000-0-0-0123456780]    SUCI (../src/amf/gmm-handler.c:166)
open5gs_5gc  | 10/09 20:56:54.388: [gmm] INFO: [imsi-001010123456780] Registration complete (../src/amf/gmm-sm.c:2146)
open5gs_5gc  | 10/09 20:56:54.388: [amf] INFO: [imsi-001010123456780] Configuration update command (../src/amf/nas-path.c:612)
open5gs_5gc  | 10/09 20:56:54.388: [gmm] INFO:     UTC [2024-10-09T20:56:54] Timezone[0]/DST[0] (../src/amf/gmm-build.c:559)
open5gs_5gc  | 10/09 20:56:54.388: [gmm] INFO:     LOCAL [2024-10-09T20:56:54] Timezone[0]/DST[0] (../src/amf/gmm-build.c:564)
open5gs_5gc  | 10/09 20:56:54.388: [amf] INFO: [Added] Number of AMF-Sessions is now 1 (../src/amf/context.c:2571)
open5gs_5gc  | 10/09 20:56:54.388: [gmm] INFO: UE SUPI[imsi-001010123456780] DNN[srsapn] S_NSSAI[SST:1 SD:0xffffff] smContextRef [NULL] (../src/amf/gmm-handler.c:1241)
open5gs_5gc  | 10/09 20:56:54.388: [gmm] INFO: No SMF Instance (../src/amf/gmm-handler.c:1278)
open5gs_5gc  | 10/09 20:56:54.389: [sbi] WARNING: [SMF] (NRF-discover) NF has already been added [87518e96-8680-41ef-af99-09ea17f61dda:1] (../lib/sbi/nnrf-handler.c:1057)
open5gs_5gc  | 10/09 20:56:54.389: [sbi] WARNING: NF EndPoint(addr) updated [127.0.0.4:80] (../lib/sbi/context.c:2174)
open5gs_5gc  | 10/09 20:56:54.389: [sbi] WARNING: NF EndPoint(addr) updated [127.0.0.4:7777] (../lib/sbi/context.c:1917)
open5gs_5gc  | 10/09 20:56:54.389: [sbi] INFO: [SMF] (NF-discover) NF Profile updated [87518e96-8680-41ef-af99-09ea17f61dda:1] (../lib/sbi/nnrf-handler.c:1095)
open5gs_5gc  | 10/09 20:56:54.390: [smf] INFO: [Added] Number of SMF-UEs is now 1 (../src/smf/context.c:1019)
open5gs_5gc  | 10/09 20:56:54.390: [smf] INFO: [Added] Number of SMF-Sessions is now 1 (../src/smf/context.c:3068)
open5gs_5gc  | 10/09 20:56:54.392: [sbi] INFO: [SMF] (SCP-discover) NF registered [87518e96-8680-41ef-af99-09ea17f61dda:1] (../lib/sbi/path.c:211)
open5gs_5gc  | 10/09 20:56:54.394: [smf] INFO: UE SUPI[imsi-001010123456780] DNN[srsapn] IPv4[10.45.1.2] IPv6[] (../src/smf/npcf-handler.c:539)
open5gs_5gc  | 10/09 20:56:54.394: [upf] INFO: [Added] Number of UPF-Sessions is now 1 (../src/upf/context.c:208)
open5gs_5gc  | 10/09 20:56:54.394: [gtp] INFO: gtp_connect() [127.0.0.4]:2152 (../lib/gtp/path.c:60)
open5gs_5gc  | 10/09 20:56:54.394: [upf] INFO: UE F-SEID[UP:0x3ab CP:0x3a1] APN[srsapn] PDN-Type[1] IPv4[10.45.1.2] IPv6[] (../src/upf/context.c:485)
open5gs_5gc  | 10/09 20:56:54.394: [upf] INFO: UE F-SEID[UP:0x3ab CP:0x3a1] APN[srsapn] PDN-Type[1] IPv4[10.45.1.2] IPv6[] (../src/upf/context.c:485)
open5gs_5gc  | 10/09 20:56:54.394: [gtp] INFO: gtp_connect() [10.53.1.2]:2152 (../lib/gtp/path.c:60)
open5gs_5gc  | 10/09 20:56:54.507: [gtp] INFO: gtp_connect() [10.53.1.1]:2152 (../lib/gtp/path.c:60)
open5gs_5gc  | 10/09 20:56:54.508: [amf] INFO: [imsi-001010123456780:1:11][0:0:NULL] /nsmf-pdusession/v1/sm-contexts/{smContextRef}/modify (../src/amf/nsmf-handler.c:837)
open5gs_5gc  | 10/09 20:56:54.628: [gmm] INFO: [imsi-001010123456780] No GUTI allocated (../src/amf/gmm-sm.c:1443)
open5gs_5gc  | 10/09 20:56:54.628: [amf] INFO: InitialUEMessage (../src/amf/ngap-handler.c:401)
open5gs_5gc  | 10/09 20:56:54.628: [amf] INFO: [Added] Number of gNB-UEs is now 2 (../src/amf/context.c:2550)
open5gs_5gc  | 10/09 20:56:54.628: [amf] INFO:     RAN_UE_NGAP_ID[1] AMF_UE_NGAP_ID[2] TAC[7] CellID[0x66c000] (../src/amf/ngap-handler.c:562)
open5gs_5gc  | 10/09 20:56:54.628: [amf] INFO: [suci-0-001-01-0000-0-0-0123456790] Unknown UE by SUCI (../src/amf/context.c:1835)
open5gs_5gc  | 10/09 20:56:54.628: [amf] INFO: [Added] Number of AMF-UEs is now 2 (../src/amf/context.c:1616)
open5gs_5gc  | 10/09 20:56:54.628: [gmm] INFO: Registration request (../src/amf/gmm-sm.c:1165)
open5gs_5gc  | 10/09 20:56:54.628: [gmm] INFO: [suci-0-001-01-0000-0-0-0123456790]    SUCI (../src/amf/gmm-handler.c:166)
open5gs_5gc  | 10/09 20:56:54.628: [amf] INFO: InitialUEMessage (../src/amf/ngap-handler.c:401)
open5gs_5gc  | 10/09 20:56:54.628: [amf] INFO: [Added] Number of gNB-UEs is now 3 (../src/amf/context.c:2550)
open5gs_5gc  | 10/09 20:56:54.628: [amf] INFO:     RAN_UE_NGAP_ID[2] AMF_UE_NGAP_ID[3] TAC[7] CellID[0x66c000] (../src/amf/ngap-handler.c:562)
open5gs_5gc  | 10/09 20:56:54.628: [amf] INFO: [suci-0-001-01-0000-0-0-0123456791] Unknown UE by SUCI (../src/amf/context.c:1835)
open5gs_5gc  | 10/09 20:56:54.628: [amf] INFO: [Added] Number of AMF-UEs is now 3 (../src/amf/context.c:1616)
open5gs_5gc  | 10/09 20:56:54.628: [gmm] INFO: Registration request (../src/amf/gmm-sm.c:1165)
open5gs_5gc  | 10/09 20:56:54.628: [gmm] INFO: [suci-0-001-01-0000-0-0-0123456791]    SUCI (../src/amf/gmm-handler.c:166)
open5gs_5gc  | 10/09 20:56:55.112: [gmm] INFO: [imsi-001010123456790] Registration complete (../src/amf/gmm-sm.c:2146)
[... cut ...]
>>> Warning: vocoder_codec2_encode_sp - option_attributes are for enums only, ignoring
<<< Welcome to GNU Radio Companion 3.10.1.1 >>>

Block paths:
	/usr/share/gnuradio/grc/blocks

Loading: "/home/ric/srsRAN_4G/build/srsue/src/multi_ue_scenario.grc"
>>> Done

Generating: '/home/ric/srsRAN_4G/build/srsue/src/multi_ue_scenario.py'

Executing: /usr/bin/python3 -u /home/ric/srsRAN_4G/build/srsue/src/multi_ue_scenario.py

QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
```

pgawlowicz, 2024-10-10:
In your gnb config you have
```
e2:
  enable_du_e2: true                # Enable DU E2 agent (one for each DU instance)
  e2sm_kpm_enabled: true            # Enable KPM service module
  addr: 127.0.0.1                   # RIC IP address
  #bind_addr: 127.0.0.100           # A local IP that the E2 agent binds to for traffic from the RIC. ONLY required if running the RIC on a separate machine. 
  port: 36421                       # RIC port
```

Do you run dockerized ORAN SC RIC, or a manually installed one?

rcbarke, 2024-10-10:
> In your gnb config you have
>
> ```
> e2:
>   enable_du_e2: true                # Enable DU E2 agent (one for each DU instance)
>   e2sm_kpm_enabled: true            # Enable KPM service module
>   addr: [...]

The dockerized RIC, instantiated with **docker compose up**

pgawlowicz, 2024-10-10:
In the tutorial, we overwrite those parameters when starting with the following command:
```
sudo ./gnb -c gnb_zmq.yaml e2 --addr="10.0.2.10" --bind_addr="10.0.2.1"
```

However, when running dockerized gnb the arguments are not added.
You can add them to this command in docker-compose.yml file: https://github.com/srsran/srsRAN_Project/blob/main/docker/docker-compose.yml#L88

Or simply change the IPs in the gnb config to
```
e2:
  enable_du_e2: true                # Enable DU E2 agent (one for each DU instance)
  e2sm_kpm_enabled: true            # Enable KPM service module
  addr: 10.0.2.10                  # RIC IP address
  bind_addr: 10.0.2.1           # A local IP that the E2 agent binds to for traffic from the RIC. 
  port: 36421                       # RIC port
```

pgawlowicz, 2024-10-10:
ah, you did already

rcbarke, 2024-10-10:
> ah, you did already

One last question: When xApps (ex: KPIMon) are deployed, would they need to be pointed at the modified RIC subnet? I haven't been successful there yet, though the E2 link would need to be fully established for us to begin testing.

pgawlowicz, 2024-10-10:
It should be already handled here : https://github.com/srsran/oran-sc-ric/blob/main/xApps/python/lib/xAppBase.py#L28-L32

rcbarke, 2024-10-11:
Root cause of this issue was the virtualbox subnet overlapping with the RIC's subnet. Changing the RIC to an unused class C IP address range as suggested here created connectivity across the gNB/RIC E2 link as well as restored connectivity across both virtualbox VMs.
https://github.com/srsran/srsRAN_Project/issues/788#issuecomment-2311996007

closed 2024-10-11 by rcbarke

AlejandraOliver, 2025-02-17:
Hi Ryan, please, can you teel me how you change the RIC to an unused class C IP address range as suggested here created connectivity across the gNB/RIC E2 link as well as restored connectivity across both virtualbox VMs???

rcbarke, 2025-02-17:
> Hi Ryan, please, can you teel me how you change the RIC to an unused class C IP address range as suggested here created connectivity across the gNB/RIC E2 link as well as restored connectivity across both virtualbox VMs???
> 
> Thanks!!! I need it for a master project and I´m so lost!!

Certainly. This will depend on which software version you're working with (containerized RIC, built from source, etc.), as well as the subnet your VirtualBox host network is assigned to. By default, VirtualBox uses 10.0.2.0/24. If you review the docker files within OSC's containerized RIC directory, you will note the RIC is leveraging the same subnet. 

To resolve, you either need to change the RIC's subnet or change Virtualbox's subnet to eliminate the overlap. The latter can be more complex, especially in multiple machine scenarios. For the former, a scripted sed command does the job:
```
    # Replace IP addresses within the repository files
    FILES_TO_UPDATE="$(git grep -l "10.0.2.")"
    if [ -n "$FILES_TO_UPDATE" ]; then  
       echo "$FILES_TO_UPDATE" | xargs sed -i "s/10.0.2./${masked_address}/g"
       echo "Modified RIC IPs to $RIC_NETADDR $RIC_SUBNET_MASK..."
    fi;
```

I recommend performing an IPV4 mapping of your entire architecture to ensure no other overlaps exist. Software defined networking frameworks aren't magic, they communicate via hostnames, ports, and IPs on each machine, and within a private network, all IPV4 addressing must be unique. Provided this is the case across your entire architecture, the gNB config correctly points to the RIC IP, you will achieve connectivity across machines.

AlejandraOliver, 2025-02-17:
Hi Ryan, thans you so much for your response. I could change the Virtualbox subnet and now I received the response of the RIC to the connection of the gNB. Later, the srsUE connect as well and get an IP for the PDU session. The problem comes when I try to execute `sudo docker-compose exec python_xapp_runner ./kpm_mon_xapp.py --metrics=DRB.UEThpDl,DRB.UEThpUl --kpm_report_style=5`. I get the following issue:

_______________________
srsran@srsran-VirtualBox:~/oran-sc-ric$ sudo docker-compose exec python_xapp_runner ./kpm_mon_xapp.py --metrics=DRB.UEThpDl,DRB.UEThpUl --kpm_report_style=5

1739753690211 7/RMR [INFO] ric message routing library on SI95 p=4562 mv=3 flg=00 id=a (f447e29 4.9.4 built: Dec 13 2023)

INFO: Subscription for E2SM_KPM Report Service Style 5 requires at least two UE IDs -> add dummy UeID: 1

Subscribe to E2 node ID: gnbd_001_001_00019b_0, RAN func: e2sm_kpm, Report Style: 5, UE_ids: [0, 1], metrics: ['DRB.UEThpDl', 'DRB.UEThpUl']

Exception in thread Thread-2:

Traceback (most recent call last):

  File "/usr/local/lib/python3.8/threading.py", line 932, in _bootstrap_inner

    self.run()

  File "/usr/local/lib/python3.8/threading.py", line 870, in run

    self._target(*self._args, **self._kwargs)

  File "./kpm_mon_xapp.py", line 91, in start

    self.e2sm_kpm.subscribe_report_service_style_5(e2_node_id, report_period, ue_ids, metric_names, granul_period, subscription_callback)

  File "/opt/xApps/lib/e2sm_kpm_module.py", line 53, in subscribe_report_service_style_5

    self.parent.subscribe(e2_node_id, self.ran_func_id, event_trigger_def, action_def, indication_callback, e2sm_types.E2SM_KPM)

  File "/opt/xApps/lib/xAppBase.py", line 109, in subscribe

    data, reason, status  = self.subscriber.Subscribe(subReq)

  File "/opt/ric-plt-xapp-frame-py/ricxappframe/xapp_subscribe.py", line 101, in Subscribe

    response = self.api.request(method="POST", url=self.uri + "/subscriptions", headers=None, body=subs_params.to_dict())

  File "/opt/ric-plt-xapp-frame-py/ricxappframe/subsclient/api_client.py", line 368, in request

    return self.rest_client.POST(url,

  File "/opt/ric-plt-xapp-frame-py/ricxappframe/subsclient/rest.py", line 269, in POST

    return self.request("POST", url,

  File "/opt/ric-plt-xapp-frame-py/ricxappframe/subsclient/rest.py", line 228, in request

    raise ApiException(http_resp=r)

ricxappframe.subsclient.rest.ApiException: (503)

Reason: Service Unavailable

HTTP response headers: HTTPHeaderDict({'Date': 'Mon, 17 Feb 2025 00:54:51 GMT', 'Content-Length': '0', 'Connection': 'close'})

__________________________________

And on the RIC side:
ric_submgr          | {"ts":1739753691324,"crit":"ERROR","id":"submgr","mdc":{"CONTAINER_NAME":"ric_submgr","HOST_NAME":"ric_submgr_host","PID":"1","POD_NAME":"submgr_pod","SERVICE_NAME":"ric_submgr_service","SYSTEM_NAME":"oran_sc_ric","time":"2025-02-17T00:54:51"},"msg":"No E2 connection for ranName gnbd_001_001_00019b_0"}

ric_dbaas           | 1:M 17 Feb 2025 00:59:46.025 * 100 changes in 300 seconds. Saving...

ric_dbaas           | 1:M 17 Feb 2025 00:59:46.026 * Background saving started by pid 15

ric_dbaas           | 15:C 17 Feb 2025 00:59:46.042 * DB saved on disk

ric_dbaas           | 15:C 17 Feb 2025 00:59:46.043 * RDB: 0 MB of memory used by copy-on-write

ric_dbaas           | 1:M 17 Feb 2025 00:59:46.127 * Background saving terminated with success

ric_dbaas           | 1:M 17 Feb 2025 01:04:47.090 * 100 changes in 300 seconds. Saving...

ric_dbaas           | 1:M 17 Feb 2025 01:04:47.091 * Background saving started by pid 16

ric_dbaas           | 16:C 17 Feb 2025 01:04:47.106 * DB saved on disk

ric_dbaas           | 16:C 17 Feb 2025 01:04:47.107 * RDB: 0 MB of memory used by copy-on-write

ric_dbaas           | 1:M 17 Feb 2025 01:04:47.192 * Background saving terminated with success

____________________________
I don`t know what´s going on. I f you could help me I would be so plessed!!!

Thanks!!

pgawlowicz, 2025-02-18:
@AlejandraOliver make sure you also set the following IP correcty:

https://github.com/srsran/oran-sc-ric/blob/main/xApps/python/lib/xAppBase.py#L32

AlejandraOliver, 2025-02-18:
Besides the file you mentioned in the link, where else should it be placed?

pgawlowicz, 2025-02-18:
You can grep for "10.0.2" in the directory and find all places. 
The IPs are hard-coded in 2-3 places. We will make them configurable when time permits.

AlejandraOliver, 2025-02-19:
Thanks!!! I have it know!! The other issue is that I need to create a custom xApp and there is no information about that:(( @pgawlowicz is there any documentation or example for create a custom xApp? It would be so usegul because I dont know how to start it!!

Thanks you!

pgawlowicz, 2025-02-19:
Unfortunately, there is not tutorial on xApp development. You can check the existing xApps here: https://github.com/srsran/oran-sc-ric/tree/main/xApps/python

AlejandraOliver, 2025-02-19:
Hi!! Ok, and do you plan to work with bthe new versions of the files?: 

O-RAN.WG3.E2AP-R003-v03.00

O-RAN.WG3.E2SM-R003-v03.00

O-RAN.WG3.E2SM-KPM-R003-v03.00

O-RAN.WG3.E2SM-RC-R003-v03.00

AlejandraOliver, 2025-02-19:
I am doing a work based on your project and I wonder why you are still keeping E2 release 3 specification 3 and have not moved to the new ones. Do they have less stability or parameters still not supported? I'm just curious so I can document it.