srsran/srsRAN_Project #857: The phone cannot find the network

Guo-arch1017, 2024-10-11:
## Issue Description ##
I successfully started gnb but my phone cannot find the network

## Setup Details ##
Ubuntu 24.04
Usrp X310
RF front-end: Vert 2450

## Expected Behavior ##
The mobile phone can successfully log into the Internet

## Actual Behaviour ##
my phone cannot find the network

My gnb.yaml
[image]

My  Termel output

[image]
[image]

pgawlowicz, 2024-10-11:
Could you share your gnb logs?

Guo-arch1017, 2024-10-12:
This is my gnb.log
[gnb.log](https://github.com/user-attachments/files/17348587/gnb.log)

Guo-arch1017, 2024-10-12:
> Could you share your gnb logs?
This is my gnb.log
[gnb.log](https://github.com/user-attachments/files/17348602/gnb.log)

Guo-arch1017, 2024-10-13:
> Could you share your gnb logs?
This is my log
[gnb.log](https://github.com/user-attachments/files/17352781/gnb.log)

Guo-arch1017, 2024-10-13:
> Could you share your gnb logs?

> Could you share your gnb logs?

I would like to ask if the transmitting power is too low?

pgawlowicz, 2024-10-15:
could you pull the newest code base and try again?

Guo-arch1017, 2024-10-17:
> could you pull the newest code base and try again?

After I reinstalled it's still the same
[image]
[image]

Guo-arch1017, 2024-10-17:
Do I need to use a spectrum analyzer to check the RF side?

Guo-arch1017, 2024-10-17:
I use a spectrum analyzer with 30db attenuation

[image]

Guo-arch1017, 2024-10-18:
I updated UHD to the latest version and reinstalled srsRAN still the same
[image]
[image]

pgawlowicz, 2024-10-21:
Did you follow this tutorial?
https://docs.srsran.com/projects/project/en/latest/tutorials/source/cotsUE/source/index.html

Which SIM card do you use?

Guo-arch1017, 2024-10-23:
> Did you follow this tutorial? https://docs.srsran.com/projects/project/en/latest/tutorials/source/cotsUE/source/index.html
> 
> Which SIM card do you use?
I switched to an ASUS phone and was able to connect to the 5G network, but I could only upload but not download, so I still couldn’t access the Internet.
[image]

pgawlowicz, 2024-10-23:
Did you enable IP forwarding in the host running Open5gs?

sudo sysctl -w net.ipv4.ip_forward=1
sudo iptables -t nat -A POSTROUTING -o <IFNAME> -j MASQUERADE

Where <IFNAME> is the name of the interface connected to the internet.

Guo-arch1017, 2024-10-23:
> Did you enable IP forwarding in the host running Open5gs?
> 
> ```
> sudo sysctl -w net.ipv4.ip_forward=1
> sudo iptables -t nat -A POSTROUTING -o <IFNAME> -j MASQUERADE
> ```
> 
> Where is the name of the interface connected to the internet.

Do I need to type these instructions if I use docker open5gc?

pgawlowicz, 2024-10-23:
no, on the hostmachine

Guo-arch1017, 2024-10-23:
> no, on the hostmachine

Do you mean that I don’t need to type these two instructions when using docker open5gc?

pgawlowicz, 2024-10-23:
you have to run those two commands  on the PC running docker container.

Guo-arch1017, 2024-10-23:
> you have to run those two commands on the PC running docker container.

Thanks, I'll try it

Guo-arch1017, 2024-10-24:
> you have to run those two commands on the PC running docker container.

Still the same, can’t access the Internet
[image]

Guo-arch1017, 2024-10-24:
> you have to run those two commands on the PC running docker container.
sudo ip ro add 10.45.0.0/16 via 10.53.1.2
I typed this command later and I was able to access the Internet.
But you can only browse web pages, you can’t click into web pages, and you can’t use Youtube.

Guo-arch1017, 2024-10-24:
> > you have to run those two commands on the PC running docker container.
> > sudo ip ro add 10.45.0.0/16 via 10.53.1.2
> > I enter the command to access the Internet.
> > I can use Google Chrome to search the web very quickly, but the bottom of the browser will show that you are offline. Please turn on wifi or mobile data, and you cannot click to enter the web page or use Youtube.
> 
> My phone is searching for mobile networks and cannot find any network, but I can use the 5G network and browse web pages, but I cannot click on them.

However, the network currently displays srsRAN
[image]

Guo-arch1017, 2024-10-24:
> could you share your config and logs?

[image]

[gnb.zip](https://github.com/user-attachments/files/17504332/gnb.zip)

Guo-arch1017, 2024-10-24:
> > could you share your config and logs?
> 
> ![螢幕快照 2024-10-24 16-58-16](https://private-user-images.githubusercontent.com/182059645/379696731-91467eca-8d0b-48ca-8da3-d2b316bdc6f7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk3NjA5MjAsIm5iZiI6MTcyOTc2MDYyMCwicGF0aCI6Ii8xODIwNTk2NDUvMzc5Njk2NzMxLTkxNDY3ZWNhLThkMGItNDhjYS04ZGEzLWQyYjMxNmJkYzZmNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAyNFQwOTAzNDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02ZTk1MmVhN2E4ODIzYmI1MTIyMjM0ODJjMDcxY2YzYzMzNjcxZWU0MzU5NjNiNzU5ZWM5ZmZlODBjYjc2ZDNiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.UYSuZo1EEAM0BsAswRhSSglaM080YSrM4oNxRgKBYnY)
> 
> [gnb.zip](https://github.com/user-attachments/files/17504332/gnb.zip)

[image]

pgawlowicz, 2024-10-24:
could you share also ngap pcap?

Guo-arch1017, 2024-10-24:
> could you share also ngap pcap?

[ngappacap.zip](https://github.com/user-attachments/files/17504472/ngappacap.zip)

pgawlowicz, 2024-10-24:
the UE connects correctly, could you check if you can `ping 8.8.8.8` from open5gs docker container?

Guo-arch1017, 2024-10-24:
> the UE connects correctly, could you check if you can `ping 8.8.8.8` from open5gs docker container?

If I set up automatic DNS on my phone, I can use Youtube, Facebook and even send messages. 
But not find mobile phone provider

Guo-arch1017, 2024-10-24:
> the UE connects correctly, could you check if you can `ping 8.8.8.8` from open5gs docker container?
Yes, but the problem is still the same
[image]

pgawlowicz, 2024-10-24:
can you `ping www.google.com` from the open5gs container?

Guo-arch1017, 2024-10-24:
> can you `ping www.google.com` from the open5gs container?

ok
[image]

Guo-arch1017, 2024-10-24:
> 你可以`ping www.google.com`從 open5gs 容器中嗎？

I'm wondering if it's because my phone can't find any telecommunications network

Guo-arch1017, 2024-10-25:
> can you `ping www.google.com` from the open5gs container?

I also need to enable roaming to use the 5G network. Is this one of the reasons?

Guo-arch1017, 2024-10-25:
> UE 連線正確，您可以檢查是否可以`ping 8.8.8.8`從 open5gs docker 容器連接嗎？

After I checked today, I found that it is not possible to browse the web using a browser, but I can use Facebook, IG, and Messenger apps.

pgawlowicz, 2024-10-30:
@Guo-arch1017  Any update on this issue? 
seems that sth is wrong with DNS. Could you sniff packets between the core network and internet?

Guo-arch1017, 2024-11-17:
> @Guo-arch1017 Any update on this issue? seems that sth is wrong with DNS. Could you sniff packets between the core network and internet?

I switched to a 5G router and the situation is still the same. I am wondering if this is the problem?

External Clock
A Leo Bodnar GPS reference clock was used as the USRP clock source for this set-up.

The use of an external clock is not essential, but it mitigates possible errors occurring as result of inaccuracies in the internal clock of the RF front-end being used. The issue comes from the CFOs and timings tolerated by 5G handsets - the onboard clocks within SDRs are not always accurate enough to operate within the limits allowed by the handsets. Using an external clock, which is much more accurate than an SDR onboard clock, reduces the likelihood that a COTS UE will be unable to see/ connect to the network due to synchronization issues.

pgawlowicz, 2024-11-18:
It seems that the radio link already works correctly. There is probably some issue with core network settings (DNS or routing).

Guo-arch1017, 2024-11-18:
> It seems that the radio link already works correctly. There is probably some issue with core network settings (DNS or routing).

How can I check the route and DNS? I've already provided all the relevant data to you, and no issues were encountered.