srsran/srsRAN_Project #858: Running gNB ZMQ config file | INI was not able to parse amf.++ file

womper25, 2024-10-11:
## Issue Description ##
Running the gNB using the gnb_zmq.yaml configuration file from the [documentation](https://docs.srsran.com/projects/project/en/latest/_downloads/a7c34dbfee2b765503a81edd2f02ec22/gnb_zmq.yaml) outputs the error shown below:

[image]

## Setup Details ##
VM: Proxmox
Memory: 16 GB
Cores: 8
Hard Disk: 50 GB
Processor: host
OS: Ubuntu 22.04.5 Desktop
BIOS: Default (SeaBIOS)

I am trying to run the srsRAN gNB with srsUE using ZMQ from the tutorial in the [documentation](https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#overview).

## Expected Behavior ##
After inputting the command `sudo ./gnb -c gnb_zmq.yaml` , it should output the following lines:

[image]

This shows that the gNB is waiting for a core network for it to connect.

## Actual Behaviour ##
For some reason, it always shows the INI cannot parse the amf in the configuration file:

[image]
 

## Steps to reproduce the problem ##
The problem appears when trying to install srsRAN_Project (install from source with ZMQ enabled) on a fresh VM (both in Proxmox and VirtualBox).

## Additional Information ##
- I have not change any of the parameters inside the gnb_zmq.yaml configuration file.
- I made sure that ZeroMQ has been found when doing the cmake../ command. 
- I have copied the gnb binary file from `/srsRAN_Project/build/apps/gnb` to `/usr/bin/gnb`
- Aside from the actual documentation, I also followed this [tutorial](https://github.com/devopsjourney23/my-srsproject-demo)

HaoxinSEU, 2024-10-11:
My two cents:
AMF section is now under cu_cp in the configuration file, you can check examples under the folder `./configs/`

pgawlowicz, 2024-10-11:
indeed, there was a change in the config structure. We are updating the docs.

womper25, 2024-10-11:
> indeed, there was a change in the config structure. We are updating the docs.

Hello thank you for your feedback. With that, should we just wait for the documentation update? Or is there a fix we can do?

womper25, 2024-10-11:
Hello thank you for your feedback. With that, should we just wait for the documentation update? Or is there a fix we can do?

pgawlowicz, 2024-10-11:
just update your config as done here: https://github.com/srsran/srsRAN_Project/blob/main/configs/gnb_rf_b210_fdd_srsUE.yml#L8-L19

womper25, 2024-10-13:
> just update your config as done here: https://github.com/srsran/srsRAN_Project/blob/main/configs/gnb_rf_b210_fdd_srsUE.yml#L8-L19

I tried placing the amf IP addresses within the cu-cp header [and pasted some of the info from the b210 config file] and it works now. Thank you so much!

closed 2024-10-13 by womper25

zhaofeidudu, 2024-10-20:
> > just update your config as done here: https://github.com/srsran/srsRAN_Project/blob/main/configs/gnb_rf_b210_fdd_srsUE.yml#L8-L19
> 
> I tried placing the amf IP addresses within the cu-cp header [and pasted some of the info from the b210 config file] and it works now. Thank you so much!

zhaofeidudu, 2024-10-20:
> > just update your config as done here: https://github.com/srsran/srsRAN_Project/blob/main/configs/gnb_rf_b210_fdd_srsUE.yml#L8-L19
> 
> I tried placing the amf IP addresses within the cu-cp header [and pasted some of the info from the b210 config file] and it works now. Thank you so much!

how to do this，i also have this problem，thank you

pgawlowicz, 2024-10-21:
Please adapt your `amf` config to be now under `cu_cp` as in the example config linked above.

zhaofeidudu, 2024-10-21:
> Please adapt your `amf` config to be now under `cu_cp` as in the example config linked above.

Thank you for your answer. I have changed gnb_zmq.yml in /srsRAN Project/build/apps/gnb,and amf.yml in ~/srsRAN Project/apps/gnb.But the problem still exists, my work is srsRAN gNB for 5G NTN
[image]
[image]
[image]

pgawlowicz, 2024-10-21:
Please pull the newest Project version.

zhaofeidudu, 2024-10-21:
> Please pull the newest Project version.

i already use the latest version

pgawlowicz, 2024-10-21:
the newest commit is `9d5dd74`

zhaofeidudu, 2024-10-21:
> the newest commit is `9d5dd74`

i dont know how to change my commit to 9d5dd74,my commit is 51e44a642. is there a way to change it ?thanks a lot

pgawlowicz, 2024-10-21:
git pull

27-Parth, 2025-06-09:
even after pulling the latest version, it is not working, please help me out.