srsran/srsRAN_Project #864: iperf in srsue not able to connect to open5gs with udp traffic

joelrenjith, 2024-10-15:
## Issue Description ##
### System details ###
  - Ubuntu 20.04 LTS
  - GCP Virtual Machine
  
  
## Setup Details ##
  - I am trying to do the srsue zermq setup given in this link : [srsRAN gNB with srsUE](https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html)
  - srsRAN_4G [Version 23.04] installation done according to [srsran4g Installation Guide](https://docs.srsran.com/projects/4g/en/latest/general/source/1_installation.html)
  - srsRAN_Project [Version 24.04] installation done according to [srsProject Installation Guide](https://docs.srsran.com/projects/project/en/latest/user_manuals/source/installation.html) using **Clone and Build**.
  - ZeroMQ setup using [srsRAN 4G with ZMQ Virtual Radios](https://docs.srsran.com/projects/4g/en/next/app_notes/source/zeromq/source/index.html) and [Build srsRAN_Project 5G RAN with ZeroMQ
](https://github.com/s5uishida/build_srsran_5g_zmq)

## Expected Behavior ##
  - I have a setup that connects a UE to a RAN and the RAN to the Core using zeromq as shown in this link: [ZeroMQ-based Setup](https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#zeromq-based-setup)
  - I am able to connect the srsue to open5gs
[image]

  - Setup testing results for everything other than iperf
[image]

## Actual Behaviour ##
  -For Iperf Testing, I setup a server on the core container:     
[image]

  - When I try running the command for iperf client on the host machine, this comes up
[image]

  - So I changed the IP Address and it works for tcp:
[image]

  - But for UDP this is the error:
[image]

## Steps to reproduce the problem ##
   - Follow the steps mentioned in zeromq setup of [ZeroMQ-based Setup](https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#zeromq-based-setup)
   - i ran  `make ` command again for srsran_4g and srsran_project to include zeromq
   
## Additional Information ##
  - I tried fixing the issue by doing this but it didnt work.
[image]

  
  
  
Any help that's provided would be greatly appreciated. Thanks in advance

pgawlowicz, 2024-10-15:
could your share your gnb and srsue logs? is it possible to ping core from UE?

joelrenjith, 2024-10-15:
Yes ping in the uplink direction works from the UE:
[image]
[image]

Here are the gnb and ue logs:
[gnb.log](https://github.com/user-attachments/files/17375066/gnb.log)
[ue.log](https://github.com/user-attachments/files/17375068/ue.log)

joelrenjith, 2024-10-15:
> could you share your gnb and srsue logs? is it possible to ping core from UE?

Hey @pgawlowicz ,
Yes ping in the uplink direction works from the UE:
[image]
[image]

Here are the gnb and ue logs:
[ue.log](https://github.com/user-attachments/files/17375079/ue.log)
[gnb.log](https://github.com/user-attachments/files/17375080/gnb.log)

joelrenjith, 2024-10-17:
@pgawlowicz  I have added an update to the reply to your question regarding pinging the core. I have edited the previous message but its basically this:
[image]

joelrenjith, 2024-10-27:
ohh ohkay thanks a lot!

closed 2024-10-27 by joelrenjith