srsran/srsRAN_Project #865: Connect more than one sdr

faratell, 2024-10-15:
Hello.
Is it possible to connect two (or more) SDRs to srsRAN_4G (or srsRAN_project)?

pgawlowicz, 2024-10-15:
With srsRAN_project, it is possible to run disaggregated gnb with multiple DUs each connected to own SDR.

faratell, 2024-10-16:
Thanks. I do not have access to 5G SIM cards. Can 4G SIM cards and open5gs (4G or 5G) core integrated with srsRAN_Project?

pgawlowicz, 2024-10-16:
It should be possible, please try and report back.

closed 2024-10-16 by pgawlowicz

faratell, 2024-10-19:
I tested my 4G sims in srsRAN_project with 5GC. The sims can not find the network.

pgawlowicz, 2024-10-21:
please follow this tutorial
https://docs.srsran.com/projects/project/en/latest/tutorials/source/cotsUE/source/index.html

faratell, 2024-10-21:
Perhaps the problem is that my phone does not support 5G.

faratell, 2024-11-09:
Hello. 
I use a phone that support 5g. My sim is Oyeitimes 4G simcard. I run this function in open5gs:
nrf,scp,amf,smf,upf,ausf,udm,pcf,nssf,udr and bsf .

My sim can find open5gs network but can not register on it.
in amf show this error:
[gmm] INFO: [suci-0-001-01-0-0-0-0000000018]    SUCI (../src/amf/gmm-handler.c:157)
11/09 11:09:56.796: [gmm] ERROR: [suci-0-001-01-0-0-0-0000000018] HTTP response error [400] (../src/amf/gmm-sm.c:1555)
11/09 11:09:56.796: [amf] WARNING: [suci-0-001-01-0-0-0-0000000018] Registration reject [95] (../src/amf/nas-path.c:219)
11/09 11:09:56.933: [amf] INFO: UE Context Release [Action:3] (../src/amf/ngap-handler.c:1677)
11/09 11:09:56.933: [amf] INFO:     RAN_UE_NGAP_ID[0] AMF_UE_NGAP_ID[1] (../src/amf/ngap-handler.c:1678)
11/09 11:09:56.933: [amf] INFO:     SUCI[suci-0-001-01-0-0-0-0000000018] (../src/amf/ngap-handler.c:1681)
11/09 11:09:56.933: [amf] INFO: [Removed] Number of gNB-UEs is now 0 (../src/amf/context.c:2530)
11/09 11:09:56.933: [amf] INFO: [Removed] Number of AMF-UEs is now 0 (../src/amf/context.c:1663)

and in nrf show this error:
11/09 11:09:31.686: [nrf] INFO: [c2196076-9e6d-41ef-aa3b-db7872dff51e] NF registered [Heartbeat:10s] (../src/nrf/nf-sm.c:190)
11/09 11:09:56.796: [nrf] ERROR: Invalid API name [nausf-auth] (../src/nrf/nrf-sm.c:243).

faratell, 2024-11-24:
I solved this issue. Now my ue connected to network and show 5G tag. But have not connection to internet and on amf show this log:
11/24 16:15:30.217: [gmm] INFO: Service request (../src/amf/gmm-sm.c:1171)
11/24 16:15:30.217: [gmm] INFO: [suci-0-001-01-0-0-0-0000000023]    5G-S_GUTI[AMF_ID:0x20040,M_TMSI:0xc00001ff] (../src/amf/gmm-handler.c:658)
11/24 16:15:30.365: [amf] INFO: [Added] Number of AMF-Sessions is now 2 (../src/amf/context.c:2544)
11/24 16:15:30.365: [gmm] INFO: UE SUPI[imsi-001010000000023] DNN[internet] S_NSSAI[SST:1 SD:0xffffff] (../src/amf/gmm-handler.c:1247)
11/24 16:15:30.490: [amf] INFO: [imsi-001010000000023:2:11][0:0:NULL] /nsmf-pdusession/v1/sm-contexts/{smContextRef}/modify (../src/amf/nsmf-handler.c:837)
11/24 16:15:39.077: [gmm] WARNING: [imsi-001010000000023] DNN Not Supported OR Not Subscribed in the Slice (../src/amf/gmm-handler.c:1234)
11/24 16:15:55.077: [gmm] WARNING: [imsi-001010000000023] DNN Not Supported OR Not Subscribed in the Slice (../src/amf/gmm-handler.c:1234)
11/24 16:16:11.077: [gmm] WARNING: [imsi-001010000000023] DNN Not Supported OR Not Subscribed in the Slice (../src/amf/gmm-handler.c:1234)
11/24 16:16:19.148: [amf] INFO: [imsi-001010000000023:2:13][0:0:NULL] /nsmf-pdusession/v1/sm-contexts/{smContextRef}/modify (../src/amf/nsmf-handler.c:837)
11/24 16:16:19.285: [amf] INFO: UE Context Release [Action:2] (../src/amf/ngap-handler.c:1677)
11/24 16:16:19.285: [amf] INFO:     RAN_UE_NGAP_ID[2] AMF_UE_NGAP_ID[3] (../src/amf/ngap-handler.c:1678)
11/24 16:16:19.285: [amf] INFO:     SUCI[suci-0-001-01-0-0-0-0000000023] (../src/amf/ngap-handler.c:1681)
11/24 16:16:19.285: [amf] INFO: [Removed] Number of gNB-UEs is now 0 (../src/amf/context.c:2530)

pgawlowicz, 2024-11-25:
could you share ngap.pcap file?

faratell, 2024-12-02:
Sorry. After  a week, now I have access to my system for getting pcap files.
By updating NAT rule as follow, Now my sim connected to 5G and have internet and 5G sign in my phone. 
%%%%%%%
sysctl -w net.ipv4.ip_forward=1
sysctl -w net.ipv6.conf.all.forwarding=1
ip tuntap add name ogstun mode tun
ip addr add 10.45.0.1/16 dev ogstun
ip addr add 2001:db8:cafe::1/48 dev ogstun
ip link set ogstun up
iptables -t nat -A POSTROUTING -s 10.45.0.0/16 ! -o ogstun -j MASQUERADE
ip6tables -t nat -A POSTROUTING -s 2001:db8:cafe::/48 ! -o ogstun -j MASQUERADE
ufw disable
%%%%%%%

but after awhile (approximate 1 min) ue disconnected: the pcap,ngap and log file is attached.
[amf.txt](https://github.com/user-attachments/files/17978068/amf.txt)
[gnb_log.zip](https://github.com/user-attachments/files/17978128/gnb_log.zip)

pgawlowicz, 2024-12-03:
Could you share gnb logs in debug mode?

pgawlowicz, 2024-12-11:
Looks strange, the PUSCH rsrp changes suddenly drops a lot:
```
2024-12-11T07:58:36.436907 [PHY     ] [D] [   545.7] PUSCH: rnti=0x4602 h_id=3 prb=[3, 9) symb=[0, 14) mod=64QAM rv=0 tbs=544 crc=OK iter=2.0 sinr=27.2dB t=110.3us uci_t=0.0us ret_t=0.0us
  rnti=0x4602
  h_id=3
  bwp=[0, 51)
  prb=[3, 9)
  symb=[0, 14)
  oack=0
  ocsi1=0
  part2=entries=[]
  alpha=0.0
  betas=[0.0, 0.0, 0.0]
  mod=64QAM
  tcr=0.92578125
  rv=0
  bg=1
  new_data=true
  n_id=1
  dmrs_mask=00100001000100
  tbs_lbrm=159749bytes
  slot=545.7
  cp=normal
  nof_layers=1
  ports=0
  dc_position=306
  n_scr_id=1
  n_scid=false
  n_cdm_g_wd=2
  dmrs_type=1
  crc=OK
  iter=2.0
  max_iter=2
  min_iter=2
  nof_cb=1
  sinr_ch_est=30.2dB
  sinr_eq[sel]=27.2dB
  sinr_evm=25.7dB
  evm=0.03
  epre=-8.6dB
  rsrp=-8.6dB
  sinr=+27.2dB
  t_align=-0.01us
  cfo=+14.9Hz
```

Next PUSCH:
```
2024-12-11T07:58:36.447318 [PHY     ] [D] [   546.8] PUCCH: rnti=0x4602 format=1 prb1=50 prb2=na symb=[0, 14) cs=0 occ=1 sr=no t=26.8us
  rnti=0x4602
  format=1
  bwp=[0, 51)
  prb1=50
  prb2=na
  symb=[0, 14)
  n_id=1
  cs=0
  occ=1
  slot=546.8
  cp=normal
  ports=0
  status=invalid
  sr=no
  detection_metric=0.3
  sinr_ch_est[sel]=-16.4dB
  epre=-46.3dB
  rsrp=-62.7dB
  sinr=-16.4dB
  t_align=na
  cfo=+1710.2Hz
```

Did you move the UE or sth? Could you measure RSRP on the UE side, for example, with NetMonster app

faratell, 2024-12-14:
Thanks for your attention.
No. My ue fixed and not any move. Sorry I have not access now to measure rsrp. Could this be related to the My SIM card that is 4G sim not 5G?

faratell, 2024-12-14:
Is there any parameter respective to rrc_inactivity_timer in srsenb for srsgnb?

pgawlowicz, 2024-12-16:
what kind of traffic do you run? do you ping any Internet address or just core network IP?

faratell, 2024-12-16:
with my browser I open some addresses. it work correctly but after awhile stopped and not open. in amf also I see the ue is released.

12/02 16:14:45.454: [amf] INFO: gNB-N2 accepted[192.168.75.185]:44573 in ng-path module (../src/amf/ngap-sctp.c:113)
12/02 16:14:45.455: [amf] INFO: gNB-N2 accepted[192.168.75.185] in master_sm module (../src/amf/amf-sm.c:741)
12/02 16:14:45.455: [amf] INFO: [Added] Number of gNBs is now 1 (../src/amf/context.c:1185)
12/02 16:14:45.455: [amf] INFO: gNB-N2[192.168.75.185] max_num_of_ostreams : 30 (../src/amf/amf-sm.c:780)
12/02 16:14:50.307: [amf] INFO: InitialUEMessage (../src/amf/ngap-handler.c:401)
12/02 16:14:50.307: [amf] INFO: [Added] Number of gNB-UEs is now 1 (../src/amf/context.c:2523)
12/02 16:14:50.308: [amf] INFO:     RAN_UE_NGAP_ID[0] AMF_UE_NGAP_ID[1] TAC[1] CellID[0x66c000] (../src/amf/ngap-handler.c:562)
12/02 16:14:50.308: [amf] INFO: Unknown UE by 5G-S_TMSI[AMF_ID:0x20040,M_TMSI:0xc0000305] (../src/amf/context.c:1807)
12/02 16:14:50.308: [amf] INFO: [Added] Number of AMF-UEs is now 1 (../src/amf/context.c:1570)
12/02 16:14:50.309: [gmm] INFO: Registration request (../src/amf/gmm-sm.c:1061)
12/02 16:14:50.309: [gmm] INFO: [Unknown ID]    5G-S_GUTI[AMF_ID:0x20040,M_TMSI:0xc0000305] (../src/amf/gmm-handler.c:170)
12/02 16:14:50.346: [gmm] INFO: Identity response (../src/amf/gmm-sm.c:1234)
12/02 16:14:50.346: [gmm] INFO: [suci-0-001-01-0-0-0-0000000015]    SUCI (../src/amf/gmm-handler.c:927)
12/02 16:14:50.722: [gmm] INFO: [imsi-001010000000015] Registration complete (../src/amf/gmm-sm.c:1993)
12/02 16:14:50.722: [amf] INFO: [imsi-001010000000015] Configuration update command (../src/amf/nas-path.c:612)
12/02 16:14:50.723: [gmm] INFO:     UTC [2024-12-02T12:44:50] Timezone[0]/DST[0] (../src/amf/gmm-build.c:558)
12/02 16:14:50.723: [gmm] INFO:     LOCAL [2024-12-02T16:14:50] Timezone[12600]/DST[0] (../src/amf/gmm-build.c:563)
12/02 16:14:51.487: [amf] INFO: [Added] Number of AMF-Sessions is now 1 (../src/amf/context.c:2544)
12/02 16:14:51.487: [gmm] INFO: UE SUPI[imsi-001010000000015] DNN[internet] S_NSSAI[SST:1 SD:0xffffff] (../src/amf/gmm-handler.c:1247)
12/02 16:14:51.511: [amf] INFO: [Added] Number of AMF-Sessions is now 2 (../src/amf/context.c:2544)
12/02 16:14:51.511: [gmm] WARNING: [imsi-001010000000015] DNN Not Supported OR Not Subscribed in the Slice (../src/amf/gmm-handler.c:1234)
12/02 16:14:51.550: [amf] INFO: [imsi-001010000000015:1:11][0:0:NULL] /nsmf-pdusession/v1/sm-contexts/{smContextRef}/modify (../src/amf/nsmf-handler.c:837)
12/02 16:14:52.132: [amf] INFO: [imsi-001010000000015:1:13][0:0:NULL] /nsmf-pdusession/v1/sm-contexts/{smContextRef}/modify (../src/amf/nsmf-handler.c:837)
12/02 16:14:52.183: [amf] INFO: UE Context Release [Action:2] (../src/amf/ngap-handler.c:1677)
12/02 16:14:52.183: [amf] INFO:     RAN_UE_NGAP_ID[0] AMF_UE_NGAP_ID[1] (../src/amf/ngap-handler.c:1678)
12/02 16:14:52.183: [amf] INFO:     SUCI[suci-0-001-01-0-0-0-0000000015] (../src/amf/ngap-handler.c:1681)
12/02 16:14:52.183: [amf] INFO: [Removed] Number of gNB-UEs is now 0 (../src/amf/context.c:2530)

pgawlowicz, 2024-12-16:
what is your PC performance (cpu?)

faratell, 2024-12-16:
Intel® Core™ i7-7700K CPU @ 4.20GHz × 8
lscpu -e=CPU,MHZ
CPU      MHZ
  0 4200.000
  1 4200.000
  2 4200.000
  3 4200.000
  4 4200.000
  5 4400.197
  6 4426.136
  7 4200.000

pgawlowicz, 2024-12-16:
did you run this script? 
https://github.com/srsran/srsRAN_Project/blob/main/scripts/srsran_performance

faratell, 2024-12-17:
yes. before doing test, I run it.

pgawlowicz, 2024-12-17:
what about this log entry:
```
12/02 16:14:51.511: [gmm] WARNING: [imsi-001010000000015] DNN Not Supported OR Not Subscribed in the Slice (../src/amf/gmm-handler.c:1234)
```
did you configure slices in the core network?

faratell, 2024-12-17:
> what about this log entry:
> 
> ```
> 12/02 16:14:51.511: [gmm] WARNING: [imsi-001010000000015] DNN Not Supported OR Not Subscribed in the Slice (../src/amf/gmm-handler.c:1234)
> ```
> 
> did you configure slices in the core network?
[amf.txt](https://github.com/user-attachments/files/18166614/amf.txt)
yes nsst=1 set in amf.yaml.

pgawlowicz, 2024-12-17:
is it correctly set in the user database? did you configure it also in NSSF config?

faratell, 2024-12-17:
> is it correctly set in the user database? did you configure it also in NSSF config?

[image]
This is my config in open5gs webui

faratell, 2024-12-17:
[image]