srsran/srsRAN_Project #872: How to configure srsRAN for NR-U ?

ohault, 2024-10-16:
In 5G, NR frequency band numbers will set frequency parameters. 
In srsRAN configuration files, we can find a band parameter to set NR band to be used according 3GPP R17. 

How to configure srsRAN to be used in NR-U unlicensed spectrum ?
Does srsRAN implement the 3GPP NR-U specific features like the one to be a good citizen according to other ISM users ?

References:
- https://en.wikipedia.org/wiki/5G_NR_frequency_bands
- https://www.sharetechnote.com/html/5G/5G_NR_Unlicensed.html 
- https://docs.srsran.com/projects/project/en/latest/user_manuals/source/config_ref.html

ohault, 2024-10-17:
> > Does srsRAN implement the 3GPP NR-U specific features like the one to be a good citizen according to other ISM users ?
> 
> This is not supported

Are these features for NR-U support actually already included in the current project development roadmap as srsRAN 5G is aligned to 3GPP release 17

> 3GPP release 16 specifications is developing a brand novel radio access technique known as 5G NR Unlicensed (5G NR-U) which is aimed at expanding 5G NR’s capabilities to bands that are not licensed.  5G NR-U is an evolution of the 4G LTE License Assisted Access –LAA standards, introduced as part of 3GPP release 13.

#https://www.techplayon.com/5g-new-radio-unlicensed-nr-u/

herlesupreeth, 2024-10-17:
> Are these features for NR-U support actually already included in the current project development roadmap as srsRAN 5G is aligned to 3GPP release 17

Its not on our current roadmap. For feature which are on our roadmap, please refer to this link - https://docs.srsran.com/projects/project/en/latest/general/source/2_features_and_roadmap.html

ohault, 2024-10-17:
> Its not on our current roadmap. For feature which are on our roadmap, please refer to this link - https://docs.srsran.com/projects/project/en/latest/general/source/2_features_and_roadmap.html

Thank you @herlesupreeth, does it possible to consider to add "NR-U support" to this roadmap as srsRAN 5G is aligned to 3GPP release 17, and NR-U has been introduced in R16 ?

> NR-Unlicensed (NR-U): In order to facilitate deployment of 5G networks for IIoT and other similar
enterprise scenarios, Rel-16 introduces the features that are necessary for NR to be operated in
unlicensed spectrum. Access to unlicensed spectrum provides an important tool to increase capacity for
both service providers and private networks. For service providers, NR-U enables access to additional
spectrum to improve the cellular network operation by offloading traffic in hotspots, while for private
networks NR-U also enables the operation of stand-alone networks in unlicensed spectrum without any
access to licensed spectrum.
6 White paper
5G Releases 16 and 17 in 3GPP
The initial focus of NR-U in Rel-16 is on enhanced mobile broadband (eMBB) services in the 5 GHz and
6 GHz frequency bands. Rel-16 NR-U uses the same flexible frame and slot structure and fundamental
physical layer design and protocol stack as NR Rel-15, hence limiting the magnitude of changes to the
UEs relative to licensed band operation. NR-U adds channel access procedures to enable fair coexistence
with other systems such as IEEE 802.11 variants or LTE Licensed-Assisted Access (LAA). In Rel-17,
enhancements for IIoT will include enabling URLLC features to be used in unlicensed spectrum to better
facilitate industrial use cases. 

#https://d1p0gxnqcu0lvz.cloudfront.net/documents/Nokia_Bell_Labs_5G_Releases_16_and_17_in_3GPP_White_Paper_1_EN.pdf

herlesupreeth, 2024-10-17:
Please reach out to sales@srs.io regarding your request

closed 2024-10-30 by pgawlowicz