srsran/srsRAN_Project #874: outliers in ru_metrics (gnb)

platyman, 2024-10-16:
## Issue Description ##
When I try to launch my gnb with sudo gnb -c ./srsRAN_Project/configs/gnb_E316.yml, everything seems to launch correctly, but nothing is displayed when I use t and I get strange values when I use ru_metrics

## Setup Details ##
I'm using an ubuntu 22.04 server on a proxmox virtual machine with open5gs up and runing and an ANTSDR E316 radio device

## Expected Behavior ##
ru_metrics should display logical values and t should display something (I suppose)

## Actual Behaviour ##
t doesn't display anything,
ru_metrics return :
| ------------------- TX ------------------ | ------------------- RX ------------------ |
| Avg. power | Peak power | PAPR | Clipping | Avg. power | Peak power | PAPR | Clipping |
|        nan |     -379.3 |  nan |      nan |        nan |     -379.3 |  nan |      nan |

## Additional Information ##
here is my gnb_E316.yml

`cu_cp:
  amf:
    addr: 127.0.0.5
    port: 38412
    bind_addr: 127.0.0.1
    supported_tracking_areas:
      - tac: 7
        plmn_list:
          - plmn: "00101"
            tai_slice_support_list:
              - sst: 1

ru_sdr:
  device_driver: uhd
  device_args: type=ant,num_recv_frames=64,num_send_frames=64,master_clock_rate=30.72e6
  srate: 30.72
  otw_format: sc12
  tx_gain: 80
  rx_gain: 40

cell_cfg:
  dl_arfcn: 667000
  band: 77
  channel_bandwidth_MHz: 20
  common_scs: 30
  plmn: "00101"
  tac: 7
  pci: 1

log:
  filename: /tmp/gnb.log
  all_level: warning

pcap:
  mac_enable: false
  mac_filename: /tmp/gnb_mac.pcap
  ngap_enable: false
  ngap_filename: /tmp/gnb_ngap.pcap`

and here is an execution example

ubuntu@ubuntu:~$ sudo gnb -c ./srsRAN_Project/configs/gnb_E316.yml

--== srsRAN gNB (commit 9d5dd742a) ==--

The PRACH detector will not meet the performance requirements with the configuration {Format B4, ZCZ 0, SCS 30kHz, Rx ports 1}.
Lower PHY in quad executor mode.
Available radio types: uhd.
[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; UHD_4.1.0.0-0-b90c92a8
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Making USRP object with args 'type=ant,num_recv_frames=64,num_send_frames=64,master_clock_rate=30.72e6'
[INFO] [ANT] Detected Device: ANTSDR
[INFO] [ANT] Detecting internal GPSDO....
[INFO] [GPS] Found a generic NMEA GPS device
[INFO] [ANT] Initialize CODEC control...
[INFO] [ANT] Initialize Radio control...
[INFO] [ANT] Performing register loopback test...
[INFO] [ANT] Register loopback test passed
[INFO] [ANT] Performing register loopback test...
[INFO] [ANT] Register loopback test passed
[INFO] [ANT] Asking for clock rate 30.720000 MHz...
[INFO] [ANT] Actually got clock rate 30.720000 MHz.
Cell pci=1, bw=20 MHz, 1T1R, dl_arfcn=667000 (n77), dl_freq=4005.0 MHz, dl_ssb_arfcn=666720, ul_freq=4005.0 MHz

N2: Connection to AMF on 127.0.0.5:38412 completed
t
ru_metrics
| ------------------- TX ------------------ | ------------------- RX ------------------ |
| Avg. power | Peak power | PAPR | Clipping | Avg. power | Peak power | PAPR | Clipping |
|        nan |     -379.3 |  nan |      nan |        nan |     -379.3 |  nan |      nan |

pgawlowicz, 2024-10-16:
you need to connect a UE to get a console trace.

ohault, 2024-10-18:
> you need to connect a UE to get a console trace.

@pgawlowicz do you mean hereunder "nan" strings will be replaced by actuals only once UEs will be connected ?

 | ------------------- TX ------------------ | ------------------- RX ------------------ |
| Avg. power | Peak power | PAPR | Clipping | Avg. power | Peak power | PAPR | Clipping |
| nan | -379.3 | nan | nan | nan | -379.3 | nan | nan |

pgawlowicz, 2024-10-21:
yes, i think so

pgawlowicz, 2024-10-30:
no, running srsgnb is enough as it contains already CU and DU units. 
Please follow this tutorial: https://docs.srsran.com/projects/project/en/latest/tutorials/source/cotsUE/source/index.html

pgawlowicz, 2024-12-05:
@platyman any update on this issue?

platyman, 2024-12-06:
Yes, I managed to launch srsRAN with the antsdr. The problem was that the core was not launched and, for some strange reason (maybe a bug), srsRAN still showed that the connection with AMF had been established. We can now measure a signal with a spectrograph when we launch srsRAN. But it seems to be fluctuating and the GSM (the EU cots we're using) doesn't detect any network (although we have created the APN). We suspect this may be due to the fact that we're using an ethernet cable instead of a USB port (the ANTSDR E316 doesn't have a USB3 port). We'll try again soon with a U220.

pgawlowicz, 2024-12-06:
Could you provide gnb logs?

platyman, 2024-12-11:
Here are my logs:

[gnb.log](https://github.com/user-attachments/files/18092369/gnb.log)

I can't get onto the 10Mhz band (I can only get onto band frequencies that allow me to have a common_scs of 15). On the band frequency I use (1500 Mhz), when I try to reduce the frequency to 5, I get:

“Common SCS 15kHz is not equal to SSB SCS 30kHz. Different SCS for common and SSB is not supported.
srsRAN ERROR: Invalid configuration detected.”

And when I set commons_scs to 15:

“--== srsRAN gNB (commit e5d5b44b9) ==--

Minimum supported Channel BW with SCS common 30kHz is 10MHz.
srsRAN ERROR: Invalid configuration detected.”

I couldn't find a parameter to modify the SSB SCS and I'm not sure what it corresponds to. When I test with a badwidth of 10 Mhz, I get the error

--== srsRAN gNB (commit e5d5b44b9) ==--

“With the given parameters, the number of PRBs for PUCCH exceeds the 50% of the BWP PRBs.
srsRAN ERROR: Invalid configuration detected.”

Another error I'm not sure I understand, but I've already managed to run the program with a bandwidth of 10 Mhz on some lower bandwidths. I've never managed to run it at 5Mhz, however. Either I have to increase the bandwidth because of the SSB SCS, or I have the same PUCCH error as at 10 Mhz. Here is my config (in cas you need it)

[config.txt](https://github.com/user-attachments/files/18092570/config.txt)

pgawlowicz, 2024-12-11:
Which frequency does ANTSDR support?

maybe you can check with Table 5.4.3.3-1 in 3GPP TS 38.104

michalisk13, 2024-12-16:
I also managed to run a 10MHz cell using ANTSDR. Although the cell can be seen from the UE side, I cannot register to it. Does this have to do with synchronisation? With zynq timestamping branch?

pgawlowicz, 2024-12-16:
coudl you share your srsgnb logs? and config

pgawlowicz, 2024-12-17:
could you update to the recent srsRAN project version? also your config is to be repeated twice.

platyman, 2024-12-18:
I'm on the latest version of srs_ran. I realized that the device I was using didn't support the frequency band I was using. I put myself on a frequency band that it can observe and I now have exactly the same problem as @michalisk13. I can see the network but I can't connect to it. I see nothing at all in the logs (neither from gnb nor from open5gc) to suggest that a connection attempt has been received. Also, the signal (observed with a spectrograph) seems rather weak despite the fact I'm using the maximum tx_gain et rx_gain supported by my device.

[image]

pgawlowicz, 2025-01-10:
not sure, maybe @ofontbach can comment?

ofontbach, 2025-01-10:
Hey, @michalisk13, @platyman, unfortunately, the Zynq Timestamping solution is not yet supporting srsRAN Project (i.e., no 5G support yet, only srsRAN 4G code). Hence, use of srsGNB (or the DU) is not yet supported (out-of-the-box) in Zynq-based devices (such as the antsdr), but I believe that there is a good chance that we will add support for that in the future.