srsran/srsRAN_Project #879: Unable to connect open5gs with gnb

mireianogue, 2024-10-18:
## Issue Description ##
The Open5gs and srsRAN are deployed on differents PCs. I am not able to see the connection between them. I think that the issue is coming from the amf bind addr of gnb. I need some help!! I already have followed the tutorials. 

## Setup Details ##
The Open5gs and the srsRAN Project are deployed/installed on differents PCs. Here are the configuration files:

### amf.yaml ###
```
 logger:
  
file:
    path: /var/log/open5gs/amf.log
  #level: info   # fatal|error|warn|info(default)|debug|trace

global:
  max:
    ue: 1024  # The number of UE can be increased depending on memory size.
     #peer: 64

amf:
  sbi:
    server:
      - address: 127.0.0.5
        port: 7777
    client:
#      nrf:
#        - uri: http://127.0.0.10:7777
      scp:
        - uri: http://127.0.0.200:7777
  ngap:
    server:
      - address: 147.83.105.216
  metrics:
    server:
      - address: 127.0.0.5
        port: 9090
  guami:
    - plmn_id:
        mcc: 214
        mnc: 91
      amf_id:
        region: 2
        set: 1
  tai:
    - plmn_id:
        mcc: 214
        mnc: 91
      tac: 1
  plmn_support:
    - plmn_id:
        mcc: 214
        mnc: 91
      s_nssai:
        - sst: 1
  security:
    integrity_order : [ NIA2, NIA1, NIA0 ]
    ciphering_order : [ NEA0, NEA1, NEA2 ]
  network_name:
    full: Open5GS
    short: Next
  amf_name: open5gs-amf0
  time:
#    t3502:
#      value: 720   # 12 minutes * 60 = 720 seconds
    t3512:
      value: 540    # 9 minutes * 60 = 540 seconds

```
### upf.yaml ###

```
logger:
  file:
    path: /var/log/open5gs/upf.log
#  level: info   # fatal|error|warn|info(default)|debug|trace

global:
  max:
    ue: 1024  # The number of UE can be increased depending on memory size.
#    peer: 64

upf:
  pfcp:
    server:
      - address: 127.0.0.7
    client:
#      smf:     #  UPF PFCP Client try to associate SMF PFCP Server
#        - address: 127.0.0.4
  gtpu:
    server:
      - address: 147.83.105.216
  session:
    - subnet: 10.45.0.0/16
      gateway: 10.45.0.1
    - subnet: 2001:db8:cafe::/48
      gateway: 2001:db8:cafe::1
  metrics:
    server:
      - address: 127.0.0.7
        port: 9090
```
### nrf.yaml ###
```

logger:
  file:
    path: /var/log/open5gs/nrf.log
#  level: info   # fatal|error|warn|info(default)|debug|trace

global:
  max:
    ue: 1024  # The number of UE can be increased depending on memory size.
#    peer: 64

nrf:
  serving:  # 5G roaming requires PLMN in NRF
    - plmn_id:
        mcc: 214
        mnc: 91
  sbi:
    server:
      - address: 127.0.0.10
        port: 7777

```
### gnb_rf_b200_tdd_n78_20mhz.yml ###

```
amf:
    addr: 147.83.105.216                                        
    bind_addr: 147.83.105.216   
                                 
ru_sdr:
  device_driver: uhd                                              
  device_args: type=b200,num_recv_frames=64,num_send_frames=64    
  srate: 23.04                                                    
  otw_format: sc12  
  tx_gain: 50                                                     
  rx_gain: 60                                                     

cell_cfg:
  dl_arfcn: 632628                                           
  band: 78                                                       
  channel_bandwidth_MHz: 20                                      
  common_scs: 30                                                 
  plmn: "21491"                                                  
  tac: 1                                                         
  pci: 1                                                         

log:
  filename: /tmp/gnb.log                                         
  all_level: warning                                             

pcap:
  mac_enable: false                                        
  mac_filename: /tmp/gnb_mac.pcap                          
  ngap_enable: false                                       
  ngap_filename: /tmp/gnb_ngap.pcap             
```           

## Expected Behavior ##
I know that this is an example, but i except to get something like this in the gnb:
[image]
And in the core (amf.log):
[image]

## Actual Behaviour ##
Actually, i am obtaining on the open5gs:

[image]

And on the srsRAN_Project (gnb.log):
[gnb.txt.log](https://github.com/user-attachments/files/17432709/gnb.txt.log)

## Steps to reproduce the problem ##
1.How to configure the open5gs addr
2.How to configure the srsRAN_Project config file addr

Anyone has the same problem?

mireianogue, 2024-10-18:
I have solved the problem before. 

But now I'm troubling with this:
[image]
 
The gnb log file is:

[gnb.txt.log](https://github.com/user-attachments/files/17434997/gnb.txt.log)

Anyone knows how to solve "The PRACH detectro will not meet the performance requirments..."

pgawlowicz, 2024-10-21:
Make sure the b210 usrp uses USB 3 connection. Probably you need to restart it.

mireianogue, 2024-10-24:
> Make sure the b210 usrp uses USB 3 connection. Probably you need to restart it.

Thank you! The problem it was the usb 2 connection. The message: The PRACH will not meet the performance ... still appears. Do you know if its going to be a problem when i connect an UE?

pgawlowicz, 2024-10-25:
See [here](https://github.com/srsran/srsRAN_Project/discussions/414)

closed 2024-10-25 by pgawlowicz