srsran/srsRAN_Project #882: F1 Setup failed. Cause: CU-CP responded with F1 Setup Failure

smartsaski, 2024-10-21:
## Issue Description ##
I ran srsCU on a different host machine than the srsRAN DU, When I try to run the DU It shows an error mesage

[image]

## Additional Information ##
# Configuration:
* CU:
`gnb_id: 441
gnb_id_bit_length: 32
ran_node_name: cu_cp_01

amf:
  addr: 127.0.0.5
  bind_addr: 127.0.0.2

cu_cp:
  inactivity_timer: 7200
  plmns: "00101"
  tacs: "1"

  f1ap:
    bind_addr: 192.168.1.40

cu_up:
  nru:
    bind_addr: 192.168.1.40
    udp_max_rx_msgs: 256`

* DU:
`gnb_id: 411
gnb_id_bit_length: 32
#gnb_du_id: 0

f1ap:
  cu_cp_addr:  192.168.1.40
  bind_addr: 192.168.1.41

nru:
  bind_addr: 192.168.1.41

ru_sdr:
  device_driver: uhd
  device_args: type=x300,num_recv_frames=64,num_send_frames=64
  srate: 92.16
  tx_gain: 30
  rx_gain: 15

cell_cfg:
  dl_arfcn: 663264
  band: 77
  channel_bandwidth_MHz: 80
  common_scs: 30
  plmn: "00101"
  tac: 1
  pci: 1
  

log:
  filename: /tmp/du.log
  all_level: warning

pcap:
  mac_enable: true
  mac_filename: /tmp/du_mac.pcap
  f1ap_enable: true
  f1ap_filename: /tmp/du_f1ap.pcap
  f1u_enable: true
  f1u_filename: /tmp/du_f1u.pcap`

## Logs ##
[GTPU    ] [I] GTP-U demux. warn_on_drop=true``

pgawlowicz, 2024-12-05:
@smartsaski any update on this issue?

pgawlowicz, 2024-12-16:
Close due to a lack of activity

closed 2024-12-16 by pgawlowicz