srsran/srsRAN_Project #884: ProblemGEO in build NTN channel Emulator

zhaofeidudu, 2024-10-21:
when i do this :python3 ./geo ntn channel emulator.py --channel-delay-us=119720
i has problem:vmcircbuf _prefs::get :info: /root/.gnuradio/prefs/vmcircbuf default factory failed to open: bad true, fail true, eof true
[image]
[image]

pgawlowicz, 2024-10-21:
does the scritpt exit?

zhaofeidudu, 2024-10-21:
> does the scritpt exit?

it only have this information,and i can return my order column, i dont think the scritpt exit.

closed 2024-12-05 by pgawlowicz