srsran/srsRAN_Project #885: unstable cellular connection

Nikolaj-1, 2024-10-21:
Cellular connection constantly bounces. I use PLuto+ SDR and Raspberry 4.
SRSRAN build ## 23.04
Oct 21 14:46:32 SRS epc.sh[365762]: Initial UE message: LIBLTE_MME_MSG_TYPE_TRACKING_AREA_UPDATE_REQUEST
Oct 21 14:46:32 SRS epc.sh[365762]: Received Initial UE message -- Tracking Area Update Request
Oct 21 14:46:32 SRS epc.sh[365762]: Tracking Area Update Request -- S-TMSI 0x4b37c29f
Oct 21 14:46:32 SRS epc.sh[365762]: Tracking Area Update Request -- eNB UE S1AP Id 3765
Oct 21 14:46:32 SRS epc.sh[365762]: Warning: Tracking area update requests are not handled yet.
Oct 21 14:46:38 SRS epc.sh[365762]: Received UE Context Release Request. MME-UE S1AP Id 4
Oct 21 14:46:38 SRS epc.sh[365762]: There are active E-RABs, send release access bearers request
Oct 21 14:46:38 SRS epc.sh[365762]: Received GTP-C PDU. Message type: GTPC_MSG_TYPE_RELEASE_ACCESS_BEARERS_REQUEST
Oct 21 14:46:38 SRS epc.sh[365762]: Received UE Context Release Complete. MME-UE S1AP Id 4
Oct 21 14:46:38 SRS epc.sh[365762]: UE Context Release Completed.
Oct 21 14:46:42 SRS epc.sh[365762]: Initial UE message: LIBLTE_MME_MSG_TYPE_ATTACH_REQUEST
Oct 21 14:46:42 SRS epc.sh[365762]: Received Initial UE message -- Attach Request
Oct 21 14:46:42 SRS epc.sh[365762]: Attach request -- M-TMSI: 0x4b37c29f
Oct 21 14:46:42 SRS epc.sh[365762]: Attach request -- eNB-UE S1AP Id: 3766
Oct 21 14:46:42 SRS epc.sh[365762]: Attach request -- Attach type: 2
Oct 21 14:46:42 SRS epc.sh[365762]: Attach Request -- UE Network Capabilities EEA: 11110000
Oct 21 14:46:42 SRS epc.sh[365762]: Attach Request -- UE Network Capabilities EIA: 01110000
Oct 21 14:46:42 SRS epc.sh[365762]: Attach Request -- MS Network Capabilities Present: true
Oct 21 14:46:42 SRS epc.sh[365762]: PDN Connectivity Request -- EPS Bearer Identity requested: 0
Oct 21 14:46:42 SRS epc.sh[365762]: PDN Connectivity Request -- Procedure Transaction Id: 37
Oct 21 14:46:42 SRS epc.sh[365762]: PDN Connectivity Request -- ESM Information Transfer requested: true
Oct 21 14:46:43 SRS epc.sh[365762]: UL NAS: Received Identity Response
Oct 21 14:46:43 SRS epc.sh[365762]: ID Response -- IMSI: 999990000000004
Oct 21 14:46:43 SRS epc.sh[365762]: Downlink NAS: Sent Authentication Request
Oct 21 14:46:43 SRS epc.sh[365762]: UL NAS: Received Authentication Response
Oct 21 14:46:43 SRS epc.sh[365762]: Authentication Response -- IMSI 999990000000004
Oct 21 14:46:43 SRS epc.sh[365762]: UE Authentication Accepted.
Oct 21 14:46:43 SRS epc.sh[365762]: Generating KeNB with UL NAS COUNT: 0
Oct 21 14:46:43 SRS epc.sh[365762]: Downlink NAS: Sending NAS Security Mode Command.
Oct 21 14:46:43 SRS epc.sh[365762]: UL NAS: Received Security Mode Complete
Oct 21 14:46:43 SRS epc.sh[365762]: Security Mode Command Complete -- IMSI: 999990000000004
Oct 21 14:46:43 SRS epc.sh[365762]: Sending ESM information request
Oct 21 14:46:43 SRS epc.sh[365762]: UL NAS: Received ESM Information Response
Oct 21 14:46:43 SRS epc.sh[365762]: ESM Info: APN mtelecom
Oct 21 14:46:43 SRS epc.sh[365762]: Getting subscription information -- QCI 7
Oct 21 14:46:43 SRS epc.sh[365762]: Sending Create Session Request.
Oct 21 14:46:43 SRS epc.sh[365762]: Creating Session Response -- IMSI: 999990000000004
Oct 21 14:46:43 SRS epc.sh[365762]: Creating Session Response -- MME control TEID: 2
Oct 21 14:46:43 SRS epc.sh[365762]: Received GTP-C PDU. Message type: GTPC_MSG_TYPE_CREATE_SESSION_REQUEST
Oct 21 14:46:43 SRS epc.sh[365762]: SPGW: Allocated Ctrl TEID 2
Oct 21 14:46:43 SRS epc.sh[365762]: SPGW: Allocated User TEID 2
Oct 21 14:46:43 SRS epc.sh[365762]: SPGW: Allocate UE IP 172.16.0.3
Oct 21 14:46:43 SRS epc.sh[365762]: Received Create Session Response
Oct 21 14:46:43 SRS epc.sh[365762]: Create Session Response -- SPGW control TEID 2
Oct 21 14:46:43 SRS epc.sh[365762]: Create Session Response -- SPGW S1-U Address: 127.0.1.100
Oct 21 14:46:43 SRS epc.sh[365762]: SPGW Allocated IP 172.16.0.3 to IMSI 999990000000004
Oct 21 14:46:43 SRS epc.sh[365762]: Adding attach accept to Initial Context Setup Request
Oct 21 14:46:43 SRS epc.sh[365762]: Sent Initial Context Setup Request. E-RAB id 5
Oct 21 14:46:43 SRS epc.sh[365762]: Received Initial Context Setup Response
Oct 21 14:46:43 SRS epc.sh[365762]: E-RAB Context Setup. E-RAB id 5
Oct 21 14:46:43 SRS epc.sh[365762]: E-RAB Context -- eNB TEID 0xdf0; eNB GTP-U Address 127.0.1.1
Oct 21 14:46:43 SRS epc.sh[365762]: UL NAS: Received Attach Complete
Oct 21 14:46:43 SRS epc.sh[365762]: Unpacked Attached Complete Message. IMSI 999990000000004
Oct 21 14:46:43 SRS epc.sh[365762]: Unpacked Activate Default EPS Bearer message. EPS Bearer id 5
Oct 21 14:46:43 SRS epc.sh[365762]: Received GTP-C PDU. Message type: GTPC_MSG_TYPE_MODIFY_BEARER_REQUEST
Oct 21 14:46:43 SRS epc.sh[365762]: Sending EMM Information
Oct 21 14:46:45 SRS epc.sh[365762]: UL NAS: PDN Connectivity Request
Oct 21 14:46:45 SRS epc.sh[365762]: DL NAS: Sending PDN Connectivity Reject
Oct 21 14:46:50 SRS epc.sh[365762]: Initial UE message: NAS Message Type Unknown
Oct 21 14:46:50 SRS epc.sh[365762]: Received Initial UE message -- Service Request
Oct 21 14:46:50 SRS epc.sh[365762]: Service request -- S-TMSI 0x493626be
Oct 21 14:46:50 SRS epc.sh[365762]: Service request -- eNB UE S1AP Id 3767
Oct 21 14:46:50 SRS epc.sh[365762]: Service Request -- Short MAC valid
Oct 21 14:46:50 SRS epc.sh[365762]: Service Request -- User is ECM DISCONNECTED
Oct 21 14:46:50 SRS epc.sh[365762]: UE previously assigned IP: 172.16.0.2
Oct 21 14:46:50 SRS epc.sh[365762]: Generating KeNB with UL NAS COUNT: 5
Oct 21 14:46:50 SRS epc.sh[365762]: UE Ctr TEID 0
Oct 21 14:46:50 SRS epc.sh[365762]: Sent Initial Context Setup Request. E-RAB id 5
[... 17 lines cut ...]
Oct 21 14:47:09 SRS epc.sh[365762]: UL NAS: PDN Connectivity Request
Oct 21 14:47:09 SRS epc.sh[365762]: DL NAS: Sending PDN Connectivity Reject
Oct 21 14:47:13 SRS epc.sh[365762]: Received UE Context Release Request. MME-UE S1AP Id 6
Oct 21 14:47:13 SRS epc.sh[365762]: There are active E-RABs, send release access bearers request
Oct 21 14:47:13 SRS epc.sh[365762]: Received GTP-C PDU. Message type: GTPC_MSG_TYPE_RELEASE_ACCESS_BEARERS_REQUEST
Oct 21 14:47:13 SRS epc.sh[365762]: Received UE Context Release Complete. MME-UE S1AP Id 6
Oct 21 14:47:13 SRS epc.sh[365762]: UE Context Release Completed.
Oct 21 14:47:17 SRS epc.sh[365762]: Initial UE message: NAS Message Type Unknown
Oct 21 14:47:17 SRS epc.sh[365762]: Received Initial UE message -- Service Request
Oct 21 14:47:17 SRS epc.sh[365762]: Service request -- S-TMSI 0x493626bf
Oct 21 14:47:17 SRS epc.sh[365762]: Service request -- eNB UE S1AP Id 3768
Oct 21 14:47:17 SRS epc.sh[365762]: Service Request -- Short MAC valid
Oct 21 14:47:17 SRS epc.sh[365762]: Service Request -- User is ECM DISCONNECTED
Oct 21 14:47:17 SRS epc.sh[365762]: UE previously assigned IP: 172.16.0.3
Oct 21 14:47:17 SRS epc.sh[365762]: Generating KeNB with UL NAS COUNT: 7
Oct 21 14:47:17 SRS epc.sh[365762]: UE Ctr TEID 0
Oct 21 14:47:17 SRS epc.sh[365762]: Sent Initial Context Setup Request. E-RAB id 5
Oct 21 14:47:17 SRS epc.sh[365762]: Received Initial Context Setup Response
Oct 21 14:47:17 SRS epc.sh[365762]: E-RAB Context Setup. E-RAB id 5
Oct 21 14:47:17 SRS epc.sh[365762]: E-RAB Context -- eNB TEID 0xdf2; eNB GTP-U Address 127.0.1.1
Oct 21 14:47:17 SRS epc.sh[365762]: Initial Context Setup Response triggered from Service Request.
Oct 21 14:47:17 SRS epc.sh[365762]: Sending Modify Bearer Request.
Oct 21 14:47:17 SRS epc.sh[365762]: Received GTP-C PDU. Message type: GTPC_MSG_TYPE_MODIFY_BEARER_REQUEST
Oct 21 14:47:17 SRS epc.sh[365762]: UL NAS: PDN Connectivity Request
Oct 21 14:47:17 SRS epc.sh[365762]: DL NAS: Sending PDN Connectivity Reject

pgawlowicz, 2024-10-21:
Which core do you use?

Nikolaj-1, 2024-10-21:
root@SRS:/srsRAN_4G# cat /proc/version
Linux version 5.15.0-1055-raspi (buildd@bos02-arm64-026) (gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #58-Ubuntu SMP PREEMPT Sat May 4 03:52:40 UTC 2024

pgawlowicz, 2024-10-21:
do you want to setup 4g or 5g network?

Nikolaj-1, 2024-10-21:
I setup 4G network.  Below you can find some additional  EPC logs

Oct 21 15:49:14 SRS epc.sh[365762]: Received GTP-C PDU. Message type: GTPC_MSG_TYPE_DELETE_SESSION_REQUEST
Oct 21 15:49:14 SRS epc.sh[365762]: Downlink NAS: Sent Authentication Request
Oct 21 15:49:14 SRS epc.sh[365762]: UL NAS: Received Authentication Response
Oct 21 15:49:14 SRS epc.sh[365762]: Authentication Response -- IMSI 999990000000004
Oct 21 15:49:14 SRS epc.sh[365762]: UE Authentication Accepted.
Oct 21 15:49:14 SRS epc.sh[365762]: Generating KeNB with UL NAS COUNT: 0
Oct 21 15:49:14 SRS epc.sh[365762]: Downlink NAS: Sending NAS Security Mode Command.
Oct 21 15:49:14 SRS epc.sh[365762]: UL NAS: Received Security Mode Complete
Oct 21 15:49:14 SRS epc.sh[365762]: Security Mode Command Complete -- IMSI: 999990000000004
Oct 21 15:49:14 SRS epc.sh[365762]: Sending ESM information request
Oct 21 15:49:34 SRS epc.sh[365762]: Received UE Context Release Request. MME-UE S1AP Id 87
Oct 21 15:49:34 SRS epc.sh[365762]: No UE context to release found. MME-UE S1AP Id: 87
Oct 21 15:49:37 SRS epc.sh[365762]: Received UE Context Release Request. MME-UE S1AP Id 88
Oct 21 15:49:37 SRS epc.sh[365762]: There are active E-RABs, send release access bearers request
Oct 21 15:49:37 SRS epc.sh[365762]: Received GTP-C PDU. Message type: GTPC_MSG_TYPE_RELEASE_ACCESS_BEARERS_REQUEST
Oct 21 15:49:37 SRS epc.sh[365762]: Received UE Context Release Complete. MME-UE S1AP Id 88
Oct 21 15:49:37 SRS epc.sh[365762]: UE Context Release Completed.
Oct 21 15:49:38 SRS epc.sh[365762]: Received UE Context Release Request. MME-UE S1AP Id 89
Oct 21 15:49:38 SRS epc.sh[365762]: There are active E-RABs, send release access bearers request
Oct 21 15:49:38 SRS epc.sh[365762]: Received GTP-C PDU. Message type: GTPC_MSG_TYPE_RELEASE_ACCESS_BEARERS_REQUEST
Oct 21 15:49:38 SRS epc.sh[365762]: Received UE Context Release Complete. MME-UE S1AP Id 89
Oct 21 15:49:38 SRS epc.sh[365762]: UE Context Release Completed.
Oct 21 15:49:39 SRS epc.sh[365762]: Initial UE message: LIBLTE_MME_MSG_TYPE_ATTACH_REQUEST
Oct 21 15:49:39 SRS epc.sh[365762]: Received Initial UE message -- Attach Request
Oct 21 15:49:39 SRS epc.sh[365762]: Attach request -- M-TMSI: 0x493626c0
Oct 21 15:49:39 SRS epc.sh[365762]: Attach request -- eNB-UE S1AP Id: 3850
Oct 21 15:49:39 SRS epc.sh[365762]: Attach request -- Attach type: 2
Oct 21 15:49:39 SRS epc.sh[365762]: Attach Request -- UE Network Capabilities EEA: 11110000
Oct 21 15:49:39 SRS epc.sh[365762]: Attach Request -- UE Network Capabilities EIA: 01110000
Oct 21 15:49:39 SRS epc.sh[365762]: Attach Request -- MS Network Capabilities Present: true
Oct 21 15:49:39 SRS epc.sh[365762]: PDN Connectivity Request -- EPS Bearer Identity requested: 0
Oct 21 15:49:39 SRS epc.sh[365762]: PDN Connectivity Request -- Procedure Transaction Id: 43
Oct 21 15:49:39 SRS epc.sh[365762]: PDN Connectivity Request -- ESM Information Transfer requested: true
Oct 21 15:49:39 SRS epc.sh[365762]: Attach Request -- Found previously attach UE.
Oct 21 15:49:39 SRS epc.sh[365762]: Found UE context. IMSI: 999990000000004, old eNB UE S1ap Id 3849, old MME UE S1AP Id 91
Oct 21 15:49:39 SRS epc.sh[365762]: GUTI Attach -- NAS Integrity OK. UL count 1, DL count 1
Oct 21 15:49:39 SRS epc.sh[365762]: Generating KeNB with UL NAS COUNT: 1
Oct 21 15:49:39 SRS epc.sh[365762]: Secure ESM information transfer requested.
Oct 21 15:49:39 SRS epc.sh[365762]: UL NAS: Received ESM Information Response
Oct 21 15:49:39 SRS epc.sh[365762]: ESM Info: APN mtelecom
Oct 21 15:49:39 SRS epc.sh[365762]: Getting subscription information -- QCI 7
Oct 21 15:49:39 SRS epc.sh[365762]: Sending Create Session Request.
Oct 21 15:49:39 SRS epc.sh[365762]: Creating Session Response -- IMSI: 999990000000004
Oct 21 15:49:39 SRS epc.sh[365762]: Creating Session Response -- MME control TEID: 5
Oct 21 15:49:39 SRS epc.sh[365762]: Received GTP-C PDU. Message type: GTPC_MSG_TYPE_CREATE_SESSION_REQUEST
Oct 21 15:49:39 SRS epc.sh[365762]: SPGW: Allocated Ctrl TEID 5
Oct 21 15:49:39 SRS epc.sh[365762]: SPGW: Allocated User TEID 5
Oct 21 15:49:39 SRS epc.sh[365762]: SPGW: Allocate UE IP 172.16.0.6
Oct 21 15:49:39 SRS epc.sh[365762]: Received Create Session Response
Oct 21 15:49:39 SRS epc.sh[365762]: Create Session Response -- SPGW control TEID 5
Oct 21 15:49:39 SRS epc.sh[365762]: Create Session Response -- SPGW S1-U Address: 127.0.1.100
Oct 21 15:49:39 SRS epc.sh[365762]: SPGW Allocated IP 172.16.0.6 to IMSI 999990000000004
Oct 21 15:49:39 SRS epc.sh[365762]: Adding attach accept to Initial Context Setup Request
Oct 21 15:49:39 SRS epc.sh[365762]: Sent Initial Context Setup Request. E-RAB id 5
Oct 21 15:49:39 SRS epc.sh[365762]: Received Initial Context Setup Response
Oct 21 15:49:39 SRS epc.sh[365762]: E-RAB Context Setup. E-RAB id 5
Oct 21 15:49:39 SRS epc.sh[365762]: E-RAB Context -- eNB TEID 0xe40; eNB GTP-U Address 127.0.1.1
Oct 21 15:49:39 SRS epc.sh[365762]: UL NAS: Received Attach Complete
Oct 21 15:49:39 SRS epc.sh[365762]: Unpacked Attached Complete Message. IMSI 999990000000004
Oct 21 15:49:39 SRS epc.sh[365762]: Unpacked Activate Default EPS Bearer message. EPS Bearer id 5
Oct 21 15:49:39 SRS epc.sh[365762]: Received GTP-C PDU. Message type: GTPC_MSG_TYPE_MODIFY_BEARER_REQUEST
Oct 21 15:49:39 SRS epc.sh[365762]: Sending EMM Information
Oct 21 15:49:42 SRS epc.sh[365762]: UL NAS: PDN Connectivity Request
Oct 21 15:49:42 SRS epc.sh[365762]: DL NAS: Sending PDN Connectivity Reject
Oct 21 15:49:44 SRS epc.sh[365762]: Received UE Context Release Request. MME-UE S1AP Id 91
Oct 21 15:49:44 SRS epc.sh[365762]: There are active E-RABs, send release access bearers request
Oct 21 15:49:44 SRS epc.sh[365762]: Received GTP-C PDU. Message type: GTPC_MSG_TYPE_RELEASE_ACCESS_BEARERS_REQUEST
Oct 21 15:49:44 SRS epc.sh[365762]: Received UE Context Release Complete. MME-UE S1AP Id 92
Oct 21 15:49:44 SRS epc.sh[365762]: UE Context Release Completed.

pgawlowicz, 2024-10-21:
This repository is dedicated to the 5G srsRAN-Project, please move this discussion to the 4G project [here](https://github.com/srsran/srsRAN_4G)