srsran/srsRAN_Project #886: Trying to replicate the srsRAN gNB with COTS UEs Tutorial

panitsasi, 2024-10-21:
## Issue Description ##
Hi all, I am trying to replicate the srsRAN gNB with COTS UEs Tutorial but I am having several issues. First of all, these are the specs of the PC that I use to replicate the tutorial: 24-Core 4.20 GHz AMD Threadripper Pro, 128 GB  RAM(2 x 64 GB 4800 MHz DDR5 ECC/REG), Ubuntu 22.04 OS. Regarding the gNB configuration, I used exactly the same parameters as you specified, using USRP B210 and OpenPlus 8T TMO.
[image]. 

For the SIM programming, i use the Gemalto IDBridge CT30 USB Smart Card Reader Writer and as programmable SIM cards I use SysmoISIM-SJA2 . For, instance i have one SIM card with the following details: 
[image]

To program the SIM card, I used the following instruction: 
./pySim-prog.py -p0 -s <ICCID> --mcc=<MCC> --mnc=<MNC> -a <ADM-KEY> --imsi=<IMSI> -k <KI> --opc=<OPC>
./pySim-prog.py -p0 -s 8949440000001288206 --mcc=001 --mnc=01 -a 27044531 --imsi=999700000128820 -k 22096A80F5E0A80387EB18C905450A07 --opc=771E373D290E3DFC9018978771BAB20E
[image]

Finally, when i re-read the SIM card, I did not see the services 125, 126. (read_SIM.txt attached)
[read_SIM.txt](https://github.com/user-attachments/files/17465834/read_SIM.txt)

Is there any mistake here ? I did not configure SUCI, because the services were not enabled. 

Also, I added the UE information in open5GS subscriber database. 

[image]
[image]

I used as subscriber key :  KI, opc: OPC, AMF: 8000 (i am not sure about this), DNN/APN*: srsran, Type*: ipv4
Is everything looks okay with the previous configuration ? 

I am not sure, because when I try to create a APN in my OnePLUS 8T TMO, I see that the MCC = 999 and MNC = 70 which it does not make sense. I am trying to change these values to 001 and 01 respectively but the APN cannot be created. 
[image]

Also, I am facing issues when I am trying to enable 5G for the UE. 
[image]
[image]
Does anyone had the same issues? 

I tried to replace the OnePlus 8T with a Google Pixel 7 but i had the same issues except the 5G connectivity. 
[image]
[image]

For the core network, I used these conf_files for amf and nrf. 
[amf_configuration.txt](https://github.com/user-attachments/files/17465896/amf_configuration.txt)
[nrf_configuration.txt](https://github.com/user-attachments/files/17465899/nrf_configuration.txt)

Everything is running locally, so i did not change the upf.yaml file. 

The core network is running properly, and the gnb can connect with the core. 
[image]
[image]

Unfortunately, the UE cannot connect to the network, probably due to SIM/APN misconfiguration. Any hints would be much appreciated. 
Thank you.

fllay, 2024-10-22:
You configuration file has plmn = 90170 but your SIM card is 99970

999700000128820

panitsasi, 2024-10-22:
Yes, that's true. I modified it but it did not work. I am trying to replicate the roaming scenario that's why the UE PLMN is different from the core PLMN. 

gNB configuration file
[image]

AMF configuration file
[image]

NRF configuration file 
[image]

gNB is connecting to Core Network
[image]
[image]

SIM configuration after reprogramming it 
[image]
[image]

Updated the Core database with the SIM information
[image]
[image]

Enable the SIM, roaming and create APN
[image]
[image]

When i tried to search for the network, the UE  did not find it. 
Does anyone faced similar issue ? 
Thanks.

antonioo420, 2024-11-07:
Hi,

The PLMN configured in the SIM is 00101, but the PLMN on the gNB conf file, AMF and NRF is 90170. Try changing the PLMN of the SIM to 90170 and register it in the suscriber data of open5gs. Also try using the force 5G NR app that is shown in the tutorial.

Sonam-22, 2024-11-15:
Following, I am also facing similar issues. Any help will be appreciated.

Sonam-22, 2024-11-18:
After changing the gNB plmn and UE plmn to 00101, everything worked fine.

pgawlowicz, 2024-11-19:
@panitsasi please check this page: https://docs.srsran.com/projects/project/en/latest/knowledge_base/source/cots_ues/source/index.html

There are some notes for OnePlus 8t and Pixel.