srsran/srsRAN_Project #891: COTS UE gets disconnected in a few seconds

Waseem0808, 2024-10-22:
## Issue Description ##
My issue is similar as mentioned in discussion: https://github.com/srsran/srsRAN_Project/issues/533
The phone connects and has network access but disconnects within a few seconds:

Line 2838041: 2024-10-21T19:58:36.508816 [MAC     ] [W] [   418.6] ue=0: RLF detected. Cause: 100 consecutive HARQ-ACK KOs.
Line 2838042: 2024-10-21T19:58:36.508855 [DU-MNG  ] [I] ue=0: RLF detected with cause="MAC max KOs reached". Timer of 4000 msec to release UE started...
Line 2856888: 2024-10-21T19:58:36.743743 [MAC     ] [W] [   442.1] ue=0: RLF detected. Cause: 100 consecutive HARQ-ACK KOs.
Line 2874228: 2024-10-21T19:58:36.958716 [MAC     ] [W] [   463.6] ue=0: RLF detected. Cause: 100 consecutive HARQ-ACK KOs.
Line 2889691: 2024-10-21T19:58:37.215719 [MAC     ] [W] [   489.4] ue=0: RLF detected. Cause: 100 consecutive HARQ-ACK KOs.
Line 2899998: 2024-10-21T19:58:37.685729 [MAC     ] [W] [   536.4] ue=0: RLF detected. Cause: 100 consecutive HARQ-ACK KOs.
Line 2904592: 2024-10-21T19:58:38.332749 [MAC     ] [W] [   601.1] ue=0: RLF detected. Cause: 100 consecutive undecoded CSIs
Line 2909620: 2024-10-21T19:58:40.507679 [DU-MNG  ] [I] ue=0: RLF timer expired with cause="RLC max ReTxs reached". Requesting a UE release..
Logs:
[gnb.zip](https://github.com/user-attachments/files/17482892/gnb.zip)

## Setup Details ##
A linux machine where gNB runs thru UHD B200 SDR with 10 MHZ geobondar clock in FDD mode(band 3).
gNB config:
[gNB_config.txt](https://github.com/user-attachments/files/17482917/gNB_config.txt)

COTS UE: Xiaomi 12
My SRSRAN project is based on this commit: 4cf7513e9fe988449aadedce3278063de761a4ac 

## Additional Information ##
I run the performance script as mentioned in other discussions.

pgawlowicz, 2024-10-23:
Which cone network do you use? Open5gs? could you share its logs and also ngap.pcap file?

Waseem0808, 2024-10-23:
@pgawlowicz yes, I am using open5gs.
I enabled ngap pcap in the new run, please find attached.
[gnb_ngap.zip](https://github.com/user-attachments/files/17488086/gnb_ngap.zip)

andrepuschmann, 2024-10-24:
I've seen the same with a Xiaomi 12 here. The issue is that the phone disconnects if VoLTE/VoNR is enabled but no IMS is running. Easiest is to turn VoLTE off. Just adding a PDN with name `ims` isn't enough. The phone will try to connect through the PDU session.

closed 2024-10-24 by andrepuschmann

Waseem0808, 2024-10-24:
I rebased my repo to latest master but the issue still persists.
However, after disabling voLTE/VoNR the connection is stable.
Thanks @pgawlowicz @andrepuschmann