srsran/srsRAN_Project #896: UE not registered to srsran gnb

hayyanali-cmd, 2024-10-24:
### Discussed in https://github.com/srsran/srsRAN_Project/discussions/876

<div type='discussions-op-text'>

<sup>Originally posted by **hayyanali-cmd** October 17, 2024</sup>
Hello srsran_project team

This is first time for me with srsran_project, i worked before alot on srsran_4G
I installed srsran_project from source last week, followed the instructions in tutorials, I am using pixel 6a (SIM plmnid is 00101) and usrp b210. as you can see:
*************
sudo gnb -c gnb_rf_b200_tdd_n78_20mhz.yml 

--== srsRAN gNB (commit e73b46182) ==--

The PRACH detector will not meet the performance requirements with the configuration {Format B4, ZCZ 0, SCS 30kHz, Rx ports 1}.
Lower PHY in quad executor mode.
Available radio types: uhd.
[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; UHD_4.7.0.0-0ubuntu1~jammy1
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Making USRP object with args 'type=b200,num_recv_frames=64,num_send_frames=64'
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Detecting internal GPSDO.... 
[INFO] [GPS] Found an internal GPSDO: GPSTCXO, Firmware Rev 0.929b
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
[INFO] [MULTI_USRP] Setting master clock rate selection to 'manual'.
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
Cell pci=1, bw=20 MHz, 1T1R, dl_arfcn=627340 (n78), dl_freq=3410.1 MHz, dl_ssb_arfcn=626976, ul_freq=3410.1 MHz

N2: Connection to AMF on 127.0.1.100:38412 completed
==== gNB started ===
Type <h> to view help
Late: 0; Underflow: 1; Overflow: 0;
Late: 0; Underflow: 1; Overflow: 0;
****************************

here is the AMF config file:
[image]
and here is the UPF config:
[image]
here is gnb file config:
[image]

# Expected behavior: 
the UE is attached properly 

# Actual behavior: 
I can see my network but cant connect, when checking the gnb_mac.pcap using wireshark: the registration reject cause is :  "5GMM cause: Semantically incorrect message (95)". 
[image]

here is the amf log:
 ***********
 10:18:09.384: [amf] INFO: [Added] Number of gNB-UEs is now 1 (../src/amf/context.c:2694)
10/17 10:18:09.384: [amf] INFO:     RAN_UE_NGAP_ID[0] AMF_UE_NGAP_ID[1] TAC[7] CellID[0x66c000] (../src/amf/ngap-handler.c:596)
10/17 10:18:09.384: [amf] INFO: [suci-0-001-01-0-0-0-0123456789] Unknown UE by SUCI (../src/amf/context.c:1861)
10/17 10:18:09.384: [amf] INFO: [Added] Number of AMF-UEs is now 1 (../src/amf/context.c:1638)
10/17 10:18:09.384: [gmm] INFO: Registration request (../src/amf/gmm-sm.c:1220)
10/17 10:18:09.384: [gmm] INFO: [suci-0-001-01-0-0-0-0123456789]    SUCI (../src/amf/gmm-handler.c:174)
10/17 10:18:09.386: [gmm] ERROR: [suci-0-001-01-0-0-0-0123456789] HTTP response error [400] (../src/amf/gmm-sm.c:1800)
10/17 10:18:09.386: [amf] WARNING: [suci-0-001-01-0-0-0-0123456789] Registration reject [95] (../src/amf/nas-path.c:211)
10/17 10:18:09.523: [amf] INFO: UE Context Release [Action:3] (../src/amf/ngap-handler.c:1726)
10/17 10:18:09.523: [amf] INFO:     RAN_UE_NGAP_ID[0] AMF_UE_NGAP_ID[1] (../src/amf/ngap-handler.c:1727)
10/17 10:18:09.523: [amf] INFO:     SUCI[suci-0-001-01-0-0-0-0123456789] (../src/amf/ngap-handler.c:1731)
10/17 10:18:09.523: [amf] INFO: [Removed] Number of gNB-UEs is now 0 (../src/amf/context.c:2701)
10/17 10:18:09.523: [amf] INFO: [Removed] Number of AMF-UEs is now 0 (../src/amf/context.c:1735)
10/17 10:19:26.651: [amf] INFO: gNB-N2[127.0.0.1] connection refused!!! (../src/amf/amf-sm.c:855)
10/17 10:19:26.651: [amf] INFO: [Removed] Number of gNBs is now 0 (../src/amf/context.c:1265)

**************************************

It is strange, because i am pretty sure that configs of the UE SIM (programmable osmocom one) are the same of configs of UE in open5gs webui. 
[image]
[image]

I ask if you can suggest thankfully a solution to this problem :), all files are attached here:
[registration_problem.zip](https://github.com/user-attachments/files/17410447/registration_problem.zip)

</div>

hudouseko, 2024-10-30:
The rejection code is interesting :). One point I have noticed is that you use PLMN id 00101 on SIM and in subscriber configuration, but AMF as well as UPF config shows PLMN id 90170. This should be matching. But it is most likely not source of the rejection, I would expect different message (+it is strange that UE tries to attach at all, given that PLMN id's don't match)

hayyanali-cmd, 2024-10-31:
> The rejection code is interesting :). One point I have noticed is that you use PLMN id 00101 on SIM and in subscriber configuration, but AMF as well as UPF config shows PLMN id 90170. This should be [...]

Hi hudouseko, actually the roaming is activated on UE that is why it is trying to attach. Even if in tutorial of srsran_project, the roaming choice was preferred.

Alan1321, 2025-04-02:
@hayyanali-cmd Did you get it connected? What type of sim are you using?

hayyanali-cmd, 2025-12-04:
> Hi [@hayyanali-cmd](https://github.com/hayyanali-cmd) , Did you manage to fix this issue? I am also seeing similar error. Thanks for any pointers.

Hi, yes, it worked by disabled the 4G components like mme, pcrf,  etc. And keeping only the 5g ones, also for my UE, it accepted only the 00101, didn't accept 90170, or roaming or any commercial plmn I'd.