srsran/srsRAN_Project #898: QPSK Modulation with SRSRAN + GNU Radio

rcbarke, 2024-10-24:
## Issue Description ##
I am attempting to simulate network slicing with Open5GS 5GC, OSC RIC, SRSRAN Project gNB, and SRSRAN4G UEs. 

In initial attempts to vary the sd and sst values from the RIC to the gNB, I've identified a way to set the slice parameters on the gNB via a controller xApp on the RIC, however, I found that the channel conditions of the Zero MQ RF signaling from the simulated RU were not changing in response to altered values. I also noted that the UEs did not throttle their performance based on their assigned UE categories in their configuration file (emBB, URLLC, mmTC). When I measured traffic from each UE namespace to the core via iperf, I noted that high bitrate traffic (ex: -b 10G) was not being throttled based on the channel conditions. The bitrate between the UEs and gNB seemed to be infinite, so ZMQ was leveraging the entire pipe for every UE I simulated traffic on. 

## Setup Details ##
To resolve, I have implemented a gnuradio waveform provided below that is performing OFDM QPSK modulation. I've created formulas able to leverage PRB assignments per UE that are output by my xApp (we are not sending these via E2 as there are separate issues with the SRSRAN E2 Node I won't raise here; I have an alternate JSON solution that's already been tested in the script). 

## Expected Behavior ##
I expect the below script to allow 3 UEs to complete RRC with the gNB, then change the PRBs assigned to each UE as values are received in the JSON file. The bitrate of iperf pings should reflect the actual PRB allocation to each UE. For instance, if ue1 has 2 PRBs assigned and makes a 7 gigabit request to the core, its bit rate should reflect its low PRB allocation and not leverage the entire channel.

## Actual Behavior ##
See:
[simulation.zip](https://github.com/user-attachments/files/17514004/simulation.zip)

This script executes without error for 3 UEs without the network running. When Open5GS, OSC RIC, SRSRAN Project srsGNB, and SRS4G srsues are all running, and this script is executed, it presents the error:

```
UE Modulation started. Press Enter to quit: crc32_bb :critical: Missing a required length tag on port 0 at item #0
crc32_bb :critical: Missing a required length tag on port 0 at item #0
crc32_bb :critical: Missing a required length tag on port 0 at item #0
crc32_bb :critical: Missing a required length tag on port 0 at item #0
thread_body_wrapper :error: ERROR thread[thread-per-block[2]: <block crc32_bb(10)>]: Missing length tag.
thread_body_wrapper :error: ERROR thread[thread-per-block[15]: <block crc32_bb(21)>]: Missing length tag.
thread_body_wrapper :error: ERROR thread[thread-per-block[28]: <block crc32_bb(32)>]: Missing length tag.
thread_body_wrapper :error: ERROR thread[thread-per-block[73]: <block crc32_bb(43)>]: Missing length tag.
```
I also note that because of this, the UEs are unable to register with the gNB.

## Additional Information ##
I don't see any documentation on the ZeroMQ signalling sent between srsGNB or srsUE online (if this exists, I would welcome it). A wireshark capture over the srsRAN bridged network adapter shows SCTP packets being sent by the UEs that are 106 bytes or 848 bits long. I'm unsure why gnuradio is not able to pick this up over the ZMQ input stream. I've already verified that my ports align across my config files. I was previously able to connect multiple UEs with gnuradio in the loop prior to adding in the OFDM modulation.

rcbarke, 2024-10-25:
> Hi,
> 
> 
> 
> Adjusting PRBs allocated to a UE by a gNB within an external GNU Radio script may lead to unintended issues.
> 
> 
> 
> For simulating changing channel conditions, please refer to this [thread](https://github.com/srsran/srsRAN_Project/issues/757), which may provide helpful guidance.
> 
> 
> 
> For non-standard deployment scenarios, we encourage using the Discussions section, as the Issues section is primarily reserved for reporting project bugs. Thank you!
> 
> 
> 
> Best Regards,
> 
> Piotr

Thanks Piotr, if I want to modify the PRBs allocated by the gNB to every UE within a simulation, does this need to be changed on both the gNB (SD/sst values in radio_sdr) and UE (nofPRBs in rat.nr)? 

In a real network, configuring the gNB should update the UEs, however in SRSRAN simulations, I don't see the UEs changing in response to gNB configuration updates. I'm not sure the ru_sdr class is impacting ZMQ in any way since we aren't using any SDRs and rather are working with the radio emulator in ZMQ instead.

Also, if the answer to the initial question is yes, are there ways to update the network slices / PRBs on the gNB and each UE in real time after the simulation starts? The network will build with whatever is set in each conf file, but I haven't identified a way to easily change SRSRAN parameterization during a simulation. Editing the conf files, tearing down the entire network, and restarting it seems to apply changes, but this is not a practical or realistic way to update optimization parameters like PRBs, power, etc.

This is why we were trying to use gnuradio to emulate PRB changes.

rcbarke, 2024-10-25:
> Hi,
>
>
>
> Adjusting PRBs allocated to a UE by a gNB within an external GNU Radio script may lead to unintended issues.
>
>
>
> For simulating changing channel conditions, please refer to this [...]

Asked another way, does SRSRAN contain a functional control plane for ZeroMQ, or is it limited to a data plane and PHY layer protocol stack? Thanks!

pgawlowicz, 2024-10-25:
We use ZeroMQ as a pipe transferring signal samples between gNB and UE. So instead of going over-the-air, the samples go over zmq sockets. 

 I am not sure what do you mean by a `functional control plane for ZeroMQ`.

rcbarke, 2024-10-25:
> We use ZeroMQ as a pipe transferring signal samples between gNB and UE. So instead of going over-the-air, the samples go over zmq sockets. 
> 
> 
> 
>  I am not sure what do you mean by a `functional control plane for ZeroMQ`.

No worries, let me try one more time with more detail.

I have Open5GS, OSC RIC, srsGNB, srsUE, and gnuradio all connected to simulate an E2E network.

My goal is to run a network slicing xApp on the RIC, which then sends optimization commands to modify the network slicing parameterization to the gNB over the E2 link via E2AP. This should be possible with ASN.1 encoded messages that modify the gNB configuration in real time after the simulation starts. The commands will be generated from a Deep Reinforcement Learning algorithm within the xApp.

From here, the simulation should adapt. If I change UE1 from 52 PRBs (entire channel) to 2 PRBs, I should see the bitrate throttle when simulating traffic with tools like iperf.

I do not currently observe this. If I send a large file (Ex: -b 10GHz) on the UE1 UL channel after changing the PRBs allocated to UE1 to 2, I observe that all 52 PRBs on the channel are leveraged and the bitrate returns around 7-8 Gbps. I would anticipate seeing a throughput a few orders of magnitude lower, something like 7-8 Mbps. 

In attempting this, I have run into two issues:

1. The SRSGNB C++ source code for E2AP protocol contains ASN1 handlers, but they do not support receiving values for sd/sst from the RIC. Therefore, we're not actually able to send the network slicing optimization commands over E2 without further developing the library.

2. If we modify these parameters directly within the gNB file and restart srsGNB, the config change is successful, but we don't see any changes in the behavior from the UEs or the RAN. iperf isn't throttling when we limit these values, nor do I see anywhere which confirms the numofPRBs on the modified UE is changing.

It's crucial that the configuration changes we are making are applied to the simulated network in real time, as the deep reinforcement learning xApp on the RIC learns from the performance of the network across time to make optimal decisions.

I asked the question about SRSRAN's control plane because I am trying to determine why this isn't working correctly. Something seems off.

Based on the 5G and ORAN standards, the RIC would generate the optimization message within an E2SM for NSSAI or RAN Control, post it as ASN1 encoded to the GNB via the E2 link, and the GNB would send PRB changes to each UE via the CU-CP (centralized unit control plane). All of the changes downstream from the RIC messaging should occur automatically within the architecture in real time; the optimization should occur across the entire end-to-end architecture as soon as the RIC posts to the E2 link.

How can network slicing be implemented with OSC RIC and SRSRAN? Are there any known limitations here? Are there any logs we can leverage to verify the issues within the messaging flow?

Let me know if you have any questions. Thank you for your help!

rcbarke, 2024-10-25:
> We use ZeroMQ as a pipe transferring signal samples between gNB and UE. So instead of going over-the-air, the samples go over zmq sockets. 
> 
> 
> 
>  I am not sure what do you mean by a `functional control plane for ZeroMQ`.

For additional context, the sst and sd values sent by the RIC are the slice service type (eMBB, URLLC, and mmTC) and slice differentiator (unique id per network slice). The RIC sends commands to assign each UE to a given sd. From here, the gNB should be using the assigned network slice to dynamically determine QoS parameters for the UE, including the assigned PRBs.

If I assign a UE to a mmTC network slice, and I send a large file from the uplink channel of the assigned UE, it should see a throttled throughout value due to a decreased number of PRBs. mmTC prioritizes many low bandwidth connections.

I do not believe there is currently an E2SM implemented within srsGNB to convert sd/sst changes from OSC RIC to QoS parameter changes, such as PRB assignment changes.

closed 2024-10-25 by rcbarke

adithya-anand-999, 2024-11-04:
Hi rcbarke 

I am working with the same system (Docker Open5GS, srsRAN gNB, GNU Radio for 3 srsue's) and was looking into Network slicing. I am having some issues finding documentation of what the correct way to add slices for the Core, gNB, or srsue's. Was wondering if you would be able to share your config files from your system for all 3 components. 

Thanks in advance.

rcbarke, 2024-11-04:
> Hi rcbarke
>
>
>
> I am working with the same system (Docker Open5GS, srsRAN gNB, GNU Radio for 3 srsue's) and was looking into Network slicing. I am having some issues finding documentation of what the [...]

Another member of my lab has these on his server, however, send me your email and I'll connect you with him. Thanks!

adithya-anand-999, 2024-11-04:
> Another member of my lab has these on his server, however, send me your email and I'll connect you with him. Thanks!

I have sent an email, its from adithyaanand@umass.edu

Thanks for the help.

lmendiboure, 2024-11-07:
Hi @rcbarke, @pgawlowicz,

Do we agree that the conclusion of your exchanges is that, for the time being, with virtual SRSRAN UEs, it is not possible to manage resource blocks allocation via the RC Xapp? Or is the behaviour (no impact on throughput) I'm seeing when deploying such a system due to errors on my part? 

Thanks in advance for your help.