srsran/srsRAN_Project #907: AMF connection established, but NGAP failed (NGSetupFailure)

meong3000, 2024-10-31:
## Issue Description ##
AMF connection established, but gnb not started & the "t" command for tracing not working

## Setup Details ##
Core: Open5GS
RAN: srsRAN
Core & RAN on different PC (both Ubuntu 22.04) connected via ethernet
USRP b205i mini
Driver: UHD 4.7

## Expected Behavior ##
...
==== gNodeB started ===
Type <t> to view trace

## Actual Behaviour ##
[image]

Log file
[image]

There are no "gnb started"

## Steps to reproduce the problem ##
Config file:
```
u_cp:
  amf:
    addr: 192.168.1.1
    port: 38412
    bind_addr: 192.168.1.4 #ip gnb
    sctp_rto_initial: 120                 # Optional INT. Sets the initial retransmission timeout when creating the SCTP connection.
    sctp_rto_min: 120                     # Optional INT. Sets the minimum retransmission timeout for the SCTP connection.
    sctp_rto_max: 500                     # Optional INT. Sets the maximum retransmission timeout for the SCTP connection.
   # sctp_initial_max_attempts: 3          # Optional INT. Sets the maximum retransmission attempts for the initial SCTP connection.
    sctp_max_init_timeo: 500              # Optional INT. Sets the maximum retransmission timeout for the initial SCTP connection.
    sctp_nodelay: false 
#   n2_bind_addr: 192.168.1.2
    supported_tracking_areas:
      - tac: 7
        plmn_list:
          - plmn: "00101"
            tai_slice_support_list:
              - sst: 1

ru_sdr:
  device_driver: uhd
  device_args: type=b200
  clock: default
  srate: 11.52 #23.04
  tx_gain: 75 #80
  rx_gain: 75 #40

cell_cfg:
  dl_arfcn: 368500
  band: 3
  channel_bandwidth_MHz: 10
  common_scs: 15
  plmn: "00101"
  tac: 7
  pdcch:
    dedicated:
      ss2_type: common
      dci_format_0_1_and_1_1: false
    #common:
     # ss0_index: 0
      #coreset0_index: 12
  prach:
    prach_config_index: 1

log:
  filename: /tmp/gnb.log
  all_level: info

pcap:
  mac_enable: enable
  mac_filename: /tmp/gnb_mac.pcap
  ngap_enable: enable
  ngap_filename: /tmp/gnb_ngap.pcap
```

## Additional Information ##
I'm using the conf file for the srsUE (from the srsRAN documentation), i only want to test the gnB, so i have not used the srsUE in any way

pgawlowicz, 2024-10-31:
The console trace is displayed only if you have at least one UE connected.

isaahmd, 2024-10-31:
kelas mas meong

pgawlowicz, 2024-10-31:
@meong3000 could you share your Open5GS config and logs and ngap pcap file?

pgawlowicz, 2024-10-31:
could you try to connect gnb to amf? and share the logs and ngap pcap

meong3000, 2024-11-08:
> could you try to connect gnb to amf? and share the logs and ngap pcap

Hi, i tried it again and it's still giving the same output, the ngap pcap is also blank

meong3000, 2024-11-09:
> could you try to connect gnb to amf? and share the logs and ngap pcap

update: the problem is now resolved, there was a mismatch on the amf & gnb configuration file about the service differentiator parameter

closed 2024-11-09 by meong3000