srsran/srsRAN_Project #912: Underflow issues when using Rpi 4

faratell, 2024-11-05:
Hello.
I have installed srsRAN_Project in RPi 4 with 32G SD and ubuntu 22 as OS. I use usrp b210 and gps antenna to syncronization.
I set GOVERNOR="performance" for cpu and cpu work as maximum frequecy of pi (1.8 GHz). 
My gnb config file is attached:
[gnb.txt](https://github.com/user-attachments/files/17627821/gnb.txt)

when gnb is run it show underflow. I tried it for both BW=20 and 10.
sudo ./gnb -c gnb1.yml

--== srsRAN gNB (commit ee1d86cd8) ==--

The PRACH detector will not meet the performance requirements with the configuration {Format B4, ZCZ 0, SCS 30kHz, Rx ports 1}.
Lower PHY in dual executor mode.
Available radio types: uhd.
[INFO] [UHD] linux; GNU C++ version 11.2.0; Boost_107400; UHD_4.1.0.5-3
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Making USRP object with args 'type=b200,num_recv_frames=64,num_send_frames=64'
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Detecting internal GPSDO.... 
[INFO] [GPS] Found an internal GPSDO: GPSTCXO, Firmware Rev 0.929b
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
[INFO] [MULTI_USRP] Setting master clock rate selection to 'manual'.
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
Cell pci=1, bw=10 MHz, 1T1R, dl_arfcn=632628 (n78), dl_freq=3489.42 MHz, dl_ssb_arfcn=632640, ul_freq=3489.42 MHz

N2: Connection to AMF on 192.168.75.181:38412 completed
==== gNB started ===
Type <h> to view help
Late: 13; Underflow: 69; Overflow: 0;
tLate: 0; Underflow: 108; Overflow: 0;

Late: 21; Underflow: 75; Overflow: 0;
Late: 9; Underflow: 66; Overflow: 0;
Late: 4; Underflow: 129; Overflow: 0;
Late: 13; Underflow: 73; Overflow: 0;

pgawlowicz, 2024-11-05:
You can try with BW=5MHz, but I am not sure if it works. We run 5MHz cell with RPi5.
Also see the comment here:
https://github.com/srsran/srsRAN_Project/discussions/349#discussioncomment-7818355 and https://github.com/srsran/srsRAN_Project/discussions/339#discussioncomment-7690124

faratell, 2024-11-05:
Thanks for your reply. I also set this config but also have underflow. 
cell_cfg:
  dl_arfcn: 361000
  band: 3
  channel_bandwidth_MHz: 5
  common_scs: 15
  plmn: "00101"
  tac: 1
  pci: 1

pgawlowicz, 2024-12-05:
@faratell any update on this issue?

faratell, 2024-12-07:
> @faratell any update on this issue?
Hi. 
No any update. I am waiting for Rpi5 to be prepared and tested on it.