srsran/srsRAN_Project #915: Failed to load RF plugin libsrsran_radio_uhd.so

antonioo420, 2024-11-06:
## Issue Description ##
I am trying to run the gNB, but it seems that some plugin is not loading and i haven't find any solution on the issues page.

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
OS: Ubuntu 22.04
RF: USRP B200
UHD: 4.7.0.0
C5G: Open5GS
srsRAN commit: 9d5dd742a

## Expected Behavior ##
```
--== srsRAN gNB (commit fbe73a49c) ==--

Connecting to AMF on 127.0.1.100:38412
[INFO] [UHD] linux; GNU C++ version 9.3.0; Boost_107100; UHD_4.0.0.0-666-g676c3a37
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Making USRP object with args 'type=b200,num_recv_frames=64,num_send_frames=64'
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test...
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz...
[INFO] [B200] Actually got clock rate 16.000000 MHz.
[INFO] [MULTI_USRP] Setting master clock rate selection to 'manual'.
[INFO] [B200] Asking for clock rate 23.040000 MHz...
[INFO] [B200] Actually got clock rate 23.040000 MHz.
Cell pci=1, bw=20 MHz, dl_arfcn=627340 (n78), dl_freq=3410.1 MHz, dl_ssb_arfcn=627264, ul_freq=3410.1 MHz

==== gNodeB started ===
Type <t> to view trace
```

## Actual Behaviour ##
[What happens instead e.g. error message]
```
gitaca@gitaca2:~/srsRAN_Project/build/apps/gnb$ sudo ./gnb -c gnb_rf_b200_tdd_n78_20mhz.yml 

--== srsRAN gNB (commit 9d5dd742a) ==--

The PRACH detector will not meet the performance requirements with the configuration {Format B4, ZCZ 0, SCS 30kHz, Rx ports 1}.
Lower PHY in quad executor mode.
Failed to load RF plugin libsrsran_radio_uhd.so: libsrsran_radio_uhd.so: cannot open shared object file: No such file or directory
Factory for radio type uhd not found. Make sure to select a valid type.
srsRAN ERROR: Unable to create radio session.
```

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]
I followed the steps shown in [https://docs.srsran.com/projects/project/en/latest/tutorials/source/cotsUE/source/index.html](url).
I downloaded srsRAN from source, then downloaded UHD with these commands:

```
$ sudo add-apt-repository ppa:ettusresearch/uhd
$ sudo apt update
$ sudo apt install libuhd-dev uhd-host
$ sudo /usr/lib/uhd/utils/uhd_images_downloader.py
```
I also installed open5gs following the quickstart tutorial [https://open5gs.org/open5gs/docs/guide/01-quickstart/](url), but that doesn't seem to be the problem.

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]
I attach the configuration files of the gnb and amf.

[gnb_rf_b200_tdd_n78_20mhz.txt](https://github.com/user-attachments/files/17647077/gnb_rf_b200_tdd_n78_20mhz.txt)

[amf.txt](https://github.com/user-attachments/files/17647093/amf.txt)

pgawlowicz, 2024-11-06:
Sth is wrong with UHD. You say you have 4.7.0.0, but the gnb uses 4.0.0.0:
```
[INFO] [UHD] linux; GNU C++ version 9.3.0; Boost_107100; UHD_4.0.0.0-666-g676c3a37
```

Did you rebuild srsRAN project after installing UHD?

antonioo420, 2024-11-07:
No, I did not. Rebuilding solved the problem.

Thank you so much.

Cheers!

closed 2024-11-07 by antonioo420