srsran/srsRAN_Project #917: [WARN][SMF][PduSess] PDU Session Resource Setup Unsuccessful by RadioNetwork

huahuaLover, 2024-11-07:
## Describe the bug
A clear and concise description of what the bug is.
amf       | 2024-11-07T14:07:03.527512672Z [INFO][AMF][Ngap][ran_addr:192.168.64.133:38738] Handle PDUSessionResourceSetupResponse
amf       | 2024-11-07T14:07:03.527903680Z [INFO][AMF][Ngap][amf_ue_ngap_id:RU:0,AU:1(3GPP)][ran_addr:192.168.64.133:38738] Handle PDUSessionResourceSetupResponse (RAN UE NGAP ID: 0)
smf       | 2024-11-07T14:07:03.529510576Z [INFO][SMF][PduSess] Receive Update SM Context Request
smf       | 2024-11-07T14:07:03.530184872Z [WARN][SMF][PduSess] PDU Session Resource Setup Unsuccessful by RadioNetwork[0]
it has passed the authentication ,why PDU Session Resource Setup Unsuccessful by RadioNetwork[0]?
## To Reproduce
Steps to reproduce the behavior:
free5gc + srsRAN_Ue +srsRAN_Project

## Expected behavior
PDU Session Resource Setup successful
## Screenshots

[image]

## Environment (please complete the following information):
 - free5GC Version: [e.g. v3.0.100]
 - OS: [Ubuntu 22.04 Server]
   srsRAN_UE[23.11]
   srsRAN_Project[24.10]
## Trace File
### Configuration File
/***********************************************gnb*******************************************/
cu_cp:
  amf:
    addr: 10.100.200.9               # The address or hostname of the AMF.
    port: 38412
    bind_addr: 192.168.64.133            # A local IP that the gNB binds to for traffic from the AMF.
    supported_tracking_areas:
      - tac: 1
        plmn_list:
          - plmn: "20893"
            tai_slice_support_list:
              - sst: 1
  inactivity_timer: 7200            # Sets the UE/PDU Session/DRB inactivity timer to 7200 seconds. Supported: [1 - 7200].

ru_sdr:
  device_driver: zmq                # The RF driver name.
  device_args: tx_port=tcp://127.0.0.1:2000,rx_port=tcp://127.0.0.1:2001,base_srate=23.04e6 # Optionally pass arguments to the selected RF driver.
  srate: 23.04                      # RF sample rate might need to be adjusted according to selected bandwidth.
  tx_gain: 75                       # Transmit gain of the RF might need to adjusted to the given situation.
  rx_gain: 75                       # Receive gain of the RF might need to adjusted to the given situation.

cell_cfg:
  dl_arfcn: 368500                  # ARFCN of the downlink carrier (center frequency).
  band: 3                           # The NR band.
  channel_bandwidth_MHz: 20         # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 15                    # Subcarrier spacing in kHz used for data.
  plmn: "20893"                     # PLMN broadcasted by the gNB.
  tac: 1                            # Tracking area code (needs to match the core configuration).
  pdcch:
    common:
      ss0_index: 0                  # Set search space zero index to match srsUE capabilities
      coreset0_index: 12            # Set search CORESET Zero index to match srsUE capabilities
    dedicated:
      ss2_type: common              # Search Space type, has to be set to common
      dci_format_0_1_and_1_1: false # Set correct DCI format (fallback)
  prach:
    prach_config_index: 1           # Sets PRACH config to match what is expected by srsUE
  slicing:                              # Optional TEXT. Configure network slicing options. This can only be set via the configuration file.
    -                                     # Configure Slice 1
      sst: 1                                  # Optional UINT (1). Sets the Slice Service Type. Supported: [0 - 255].
      sd: 66051 
log:
  filename: /tmp/gnb.log            # Path of the log file.
  all_level: info                   # Logging level applied to all layers.
  hex_max_size: 0

pcap:
  mac_enable: false                 # Set to true to enable MAC-layer PCAPs.
  mac_filename: /tmp/gnb_mac.pcap   # Path where the MAC PCAP is stored.
  ngap_enable: false                # Set to true to enable NGAP PCAPs.
  ngap_filename: /tmp/gnb_ngap.pcap # Path where the NGAP PCAP is stored.

/***********************ue*************************************/
[rf]
freq_offset = 0
tx_gain = 50
rx_gain = 40
srate = 23.04e6
nof_antennas = 1

device_name = zmq
device_args = tx_port=tcp://127.0.0.1:2001,rx_port=tcp://127.0.0.1:2000,base_srate=23.04e6

[rat.eutra]
dl_earfcn = 2850
nof_carriers = 0

[rat.nr]
bands = 3
nof_carriers = 1
max_nof_prb = 106
nof_prb = 106

[pcap]
enable = none
mac_filename = /tmp/ue_mac.pcap
mac_nr_filename = /tmp/ue_mac_nr.pcap
nas_filename = /tmp/ue_nas.pcap

[log]
all_level = info
phy_lib_level = none
all_hex_limit = 32
filename = /tmp/ue.log
file_max_size = -1

[usim]
mode = soft
algo = milenage
opc  = 8e27b6af0e692e750f32667a3b14605d
k    = 8baf473f2f8fd09487cccbd7097c6862
imsi = 208930000000001
imei = 353490069873319

[rrc]
release = 15
ue_category = 4
[slicing]
enable = true
nssai-sst = 1
nssai-sd = 66051
[nas]
apn = internet
apn_protocol = ipv4

[gw]
netns = ue1
ip_devname = tun_srsue
ip_netmask = 255.255.255.0

[gui]
enable = false

pgawlowicz, 2024-11-11:
could you share gnb and srsUE logs? could you share ngap pcap file?

LuxeZN, 2024-11-12:
> could you share gnb and srsUE logs? could you share ngap pcap file?

I think we had this issue with free5gc, not sure if it was a free5gc issue or something on the gNB/UE end. Will investigate what we went through and post if I have any updates.

huahuaLover, 2024-11-13:
Thank you for your response. I have found the issue. The default 5QI in free5gc is 8, but srsRAN_Project does not support it. I set the 5QI in free5gc to 1, and that resolved the problem successfully.

closed 2024-11-13 by huahuaLover