srsran/srsRAN_Project #920: SRS gNB with SRS UE for different dl_arfcn and band

Subangkar, 2024-11-08:
## Issue Description ##
Hi, I tried the tutorial at the following link to connect SRS gNB and SRS UE for band 3 as mentioned in the config file
https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html
The used configs
```
cell_cfg:
  dl_arfcn: 368500
  band: 3
  channel_bandwidth_MHz: 20
  common_scs: 15
  plmn: "00101"
```
The setup works successfully. However, if I change the `dl_arfcn` value, the UE does not connect even after I keep all other configs same. How do I resolve this? My actual target is too connect the SRS UE with SRS gNB for **band 78** with `dl_arfcn` ~ 367960. In this case, what should be the modified value of `dl_earfcn` in SRS UE config file?

## Setup Details ##

- gNB config file contents:

```
cu_cp:
  amf:
    addr: 10.53.1.2
    port: 38412
    bind_addr: 10.53.1.1
    supported_tracking_areas:
      - tac: 7
        plmn_list:
          - plmn: "00101"
            tai_slice_support_list:
              - sst: 1

ru_sdr:
  device_driver: uhd
  device_args: type=b200
  clock: external
  srate: 23.04
  tx_gain: 80
  rx_gain: 40

cell_cfg:
  dl_arfcn: 369500
  band: 3
  channel_bandwidth_MHz: 20
  common_scs: 15
  plmn: "00101"
  tac: 7
  pdcch:
    dedicated:
      ss2_type: common
      dci_format_0_1_and_1_1: false
    common:
      ss0_index: 0
      coreset0_index: 12
  prach:
    prach_config_index: 1

log:
  filename: /tmp/gnb.log
  all_level: info

pcap:
  mac_enable: enable
  mac_filename: /tmp/gnb_mac.pcap
  ngap_enable: enable
  ngap_filename: /tmp/gnb_ngap.pcap
```

- UE config file contents:

```
[rf]
freq_offset = 0
tx_gain = 50
rx_gain = 40
srate = 23.04e6
nof_antennas = 1

device_name = uhd
device_args = clock=external
time_adv_nsamples = 300

[rat.eutra]
dl_earfcn = 2850
nof_carriers = 0

[rat.nr]
bands = 3
nof_carriers = 1
max_nof_prb = 106
nof_prb = 106

[pcap]
enable = none
mac_filename = /tmp/ue_mac.pcap
mac_nr_filename = /tmp/ue_mac_nr.pcap
nas_filename = /tmp/ue_nas.pcap

[log]
all_level = info
phy_lib_level = none
all_hex_limit = 32
filename = /tmp/ue.log
file_max_size = -1

[usim]
mode = soft
algo = milenage
opc  = 63BFA50EE6523365FF14C1F45F88737D
k    = 00112233445566778899aabbccddeeff
imsi = 001010123456780
imei = 353490069873319

[rrc]
release = 15
ue_category = 4

[nas]
apn = srsapn
apn_protocol = ipv4

[gw]
#netns = ue1
#ip_devname = tun_srsue
#ip_netmask = 255.255.255.0

[gui]
enable = false
```

## Expected Behavior ##
UE conects with the gNB.

## Actual Behaviour ##
The UE does not connect to the gNB but I do not see any errors.

pgawlowicz, 2024-11-08:
You have to correctly set the following two parameters:

```
[rat.nr]
dl_nr_arfcn = ...
ssb_nr_arfcn = ...
```

pgawlowicz, 2024-11-08:
Band 78 is for TDD.  srsUE supports only FDD

closed 2024-11-09 by Subangkar

ohault, 2024-11-18:
> Band 78 is for TDD.  srsUE supports only FDD

@pgawlowicz like for band 28?

Is this band 28 supported by both srsUE and srsRAN?

How to use a dedicated band for both srsUE and srsRAN ?

Does is possible to avoid setting ARFCN and setting all the parameters manually like(FDD/TDF, freq, ...)?

pgawlowicz, 2024-11-18:
Band 28 should work out of the box. 

Setting the parameters manually is not exposed, but it should be quite easy to add if really needed.

ohault, 2024-11-18:
> Band 28 should work out of the box.
> 
> Setting the parameters manually is not exposed, but it should be quite easy to add if really needed.

@pgawlowicz , thank you. does it possible to have an example of how to set manually these parameters ?

ohault, 2024-11-18:
@

> It requires adding a code that will overwrite the DL and UL frequency derived from the ARFCN.

May I create a new ticket for this as a feature request, or do you prefer to do it ?

pgawlowicz, 2024-11-18:
We don’t plan to support this feature in the mainline.

ohault, 2024-11-18:
> It requires adding a code that will overwrite the DL and UL frequency derived from the ARFCN.

Should it be done only of the files located in this folder https://github.com/srsran/srsRAN_Project/tree/main/lib/ran ?