srsran/srsRAN_Project #921: srsRAN gNB with COTS UE for band 2

Subangkar, 2024-11-09:
## Issue Description ##
Hi, I tried the tutorial at the following link to connect SRS gNB and SRS UE for band  2 and 3 as mentioned in the config file
https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html

The setup works successfully i.e. for band 2 the srsUE connects to srs gNB. Now I want to connect to real COTS UE: "OnePlus Nord N30 5G" ()
From the specs, I have found the device supports band 2 (https://www.oneplus.com/us/n30-5g/specs). So I first write a sim with pysim using the following command
```
./pySim-prog.py -p0 -s 8949440000001313202 --mcc=999 --mnc=70 -a XXXXXXXX --imsi=999700000131320  -k 2C86B67D41DC65AC2FD8F05FEDD5A90B --opc=CC26852F46DC869BFE4EFF538751B733
```
So first I used the same UE configs (imsi, opc, k) with srsUE to connect to SRS gNB for band 2. **The connection was successful**. But when I tried with the real and isim. after configuring the apn. The phone doesn't connect to gNB and the carrier is not even shown in the list of carriers. Nothing is shown on the Open5GS terminal log either. What might be the reason?

## Setup Details ##

- gNB config file contents:

```
cu_cp:
  amf:
    addr: 10.53.1.2
    port: 38412
    bind_addr: 10.53.1.1
    supported_tracking_areas:
      - tac: 7
        plmn_list:
          - plmn: "99970"
            tai_slice_support_list:
              - sst: 1

ru_sdr:
  device_driver: uhd
  device_args: type=b200
  clock: external
  srate: 23.04
  tx_gain: 85
  rx_gain: 75

cell_cfg:
  dl_arfcn: 386000                  # ARFCN of the downlink carrier (center frequency).
  band: 2                           # The NR band.
  channel_bandwidth_MHz: 20         # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 15                    # Subcarrier spacing in kHz used for data.
  plmn: "99970"                     # PLMN broadcasted by the gNB.
  tac: 7                            # Tracking area code (needs to match the core configuration).
  pdcch:
    common:
      ss0_index: 0                  # Set search space zero index to match srsUE capabilities
      coreset0_index: 12            # Set search CORESET Zero index to match srsUE capabilities
    dedicated:
      ss2_type: common              # Search Space type, has to be set to common
      dci_format_0_1_and_1_1: false # Set correct DCI format (fallback)
  prach:
    prach_config_index: 1           # Sets PRACH config to match what is expected by srsUE

log:
  filename: /tmp/gnb.log
  all_level: info

pcap:
  mac_enable: enable
  mac_filename: /tmp/gnb_mac.pcap
  ngap_enable: enable
  ngap_filename: /tmp/gnb_ngap.pcap
```

- srsUE config file contents (It works):

```
[rf]
freq_offset = 0
tx_gain = 50
rx_gain = 40
srate = 23.04e6
nof_antennas = 1

device_name = uhd
device_args = clock=external
time_adv_nsamples = 300

[rat.eutra]
# dl_freq = 1839800000
dl_earfcn = 2850
nof_carriers = 0

[rat.nr]
bands = 2
dl_nr_arfcn = 386000
ssb_nr_arfcn = 385970
# scs = 30
# ssb_scs = 30
nof_carriers = 1
max_nof_prb = 106
nof_prb = 106

[pcap]
enable = none
mac_filename = /tmp/ue_mac.pcap
mac_nr_filename = /tmp/ue_mac_nr.pcap
nas_filename = /tmp/ue_nas.pcap

[log]
all_level = info
phy_lib_level = none
all_hex_limit = 32
filename = /tmp/ue.log
file_max_size = -1

[usim]
mode = soft
algo = milenage
opc  = CC26852F46DC869BFE4EFF538751B733
k    = 2C86B67D41DC65AC2FD8F05FEDD5A90B
imsi = 999700000131320
imei = 860970060845933

[rrc]
release = 15
ue_category = 4

[nas]
apn = srsapn
apn_protocol = ipv4

[gw]
#netns = ue1
#ip_devname = tun_srsue
#ip_netmask = 255.255.255.0

[gui]
enable = false
```

- COTS UE (It doesn't work):
```
OnePlus Nord N30 5G 
Connectivity
LTE: 4×4 MIMO
Band:
  GSM: 850/900/1800/1900MHz
  WCDMA: B1/2/4/5/8
  FDD-LTE: B1/2/3/4/5/7/8/12/17/18/20/25/26/28/66/13/71
  TD-LTE: B38/39/40/41/48
  5G SA: n2/5/12/25/66/71/41/77/78
  5G NSA: n2/5/25/66/71/41/77/78
*Network connectivity may vary depending on carrier’s network and related service deployment.
```

## Expected Behavior ##
COTS UE connects with the gNB.

## Actual Behavior ##
The UE does not connect to the gNB but I do not see any errors.

pgawlowicz, 2024-11-11:
Please note that additional steps may be required to support test PLMNs. You can view our list of supported COTS UEs [here](https://docs.srsran.com/projects/project/en/latest/knowledge_base/source/cots_ues/source/index.html).

To connect your OnePlus phone, please try to use the configurations used for other OnePlus devices.

pgawlowicz, 2024-12-05:
@Subangkar any update?

Subangkar, 2024-12-05:
Could not make it work on this particular device yet. However,  a bit older but another device "Oneplus Nord 5G" worked on band 78. So moved on to proceed with that one.

pgawlowicz, 2024-12-05:
Great! thanks for feedback!

closed 2024-12-05 by pgawlowicz