srsran/srsRAN_Project #923: Can not ping from the core network to UE with B210

Zhang-xzhe, 2024-11-10:
## Issue Description ##
I successfully set up wireless connection with two B210. UE gets its IP:10.45.1.2 and can ping 10.45.1.1. But core network cannot ping 10.45.1.2. I have uninstall the Open5gs.

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

Open 5gs in docker
```
open5gs_5gc  | 11/10 15:20:52.639: [upf] INFO: [Added] Number of UPF-Sessions is now 1 (../src/upf/context.c:208)
open5gs_5gc  | 11/10 15:20:52.639: [upf] INFO: UE F-SEID[UP:0x563 CP:0x8f8] APN[srsapn] PDN-Type[1] IPv4[10.45.1.2] IPv6[] (../src/upf/context.c:485)
open5gs_5gc  | 11/10 15:20:52.639: [upf] INFO: UE F-SEID[UP:0x563 CP:0x8f8] APN[srsapn] PDN-Type[1] IPv4[10.45.1.2] IPv6[] (../src/upf/context.c:485)
open5gs_5gc  | 11/10 15:20:52.670: [amf] INFO: [imsi-001010123456780:1:11][0:0:NULL] /nsmf-pdusession/v1/sm-contexts/{smContextRef}/modify (../src/amf/nsmf-handler.c:837)
open5gs_5gc  | 11/10 15:20:52.795: [gmm] INFO: [imsi-001010123456780] No GUTI allocated (../src/amf/gmm-sm.c:1443)

```
the ip table
```
g@g-System-Product-Name:~$ route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         192.168.79.254  0.0.0.0         UG    100    0        0 enp4s0
10.45.0.0       10.53.1.2       255.255.0.0     UG    0      0        0 br-8219ce3e2932
10.53.1.0       0.0.0.0         255.255.255.0   U     0      0        0 br-8219ce3e2932
169.254.0.0     0.0.0.0         255.255.0.0     U     1000   0        0 br-ff0381c7b524
172.17.0.0      0.0.0.0         255.255.0.0     U     0      0        0 docker0
172.19.1.0      0.0.0.0         255.255.255.0   U     0      0        0 br-ff0381c7b524
192.168.79.0    0.0.0.0         255.255.255.0   U     100    0        0 enp4s0
```

## Expected Behavior ##
[What you expect to happen]
core network can ping ue

## Actual Behaviour ##
[What happens instead e.g. error message]
```
g@g-System-Product-Name:~$ ping 10.45.1.2
PING 10.45.1.2 (10.45.1.2) 56(84) bytes of data.
^C
--- 10.45.1.2 ping statistics ---
15 packets transmitted, 0 received, 100% packet loss, time 14375ms

```

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

pgawlowicz, 2024-11-11:
could you check the routing table in the UE PC?

pgawlowicz, 2024-11-11:
could you remove the followin entry from srsUE PC
```
10.53.1.0       0.0.0.0         255.255.255.0   U     0      0        0 br-6c82738652e0
```

Zhang-xzhe, 2024-11-11:
Yes, I do that
```
g@g-NUC12WSHi5:~$  sudo ip ro del 10.53.1.0/24
g@g-NUC12WSHi5:~$ route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         10.45.1.1       0.0.0.0         UG    0      0        0 tun_srsue
0.0.0.0         10.13.63.254    0.0.0.0         UG    600    0        0 wlo1
10.13.0.0       0.0.0.0         255.255.0.0     U     600    0        0 wlo1
10.45.1.0       0.0.0.0         255.255.255.0   U     0      0        0 tun_srsue
169.254.0.0     0.0.0.0         255.255.0.0     U     1000   0        0 wlo1
172.17.0.0      0.0.0.0         255.255.0.0     U     0      0        0 docker0
172.19.1.0      0.0.0.0         255.255.255.0   U     0      0        0 br-b557ce824e62
```

I can ping core network from UE PC
```
g@g-NUC12WSHi5:~$ ping 10.45.1.1
PING 10.45.1.1 (10.45.1.1) 56(84) bytes of data.
64 bytes from 10.45.1.1: icmp_seq=1 ttl=64 time=37.3 ms
64 bytes from 10.45.1.1: icmp_seq=2 ttl=64 time=36.0 ms
64 bytes from 10.45.1.1: icmp_seq=3 ttl=64 time=35.0 ms
64 bytes from 10.45.1.1: icmp_seq=4 ttl=64 time=34.1 ms
64 bytes from 10.45.1.1: icmp_seq=5 ttl=64 time=33.9 ms
64 bytes from 10.45.1.1: icmp_seq=6 ttl=64 time=32.9 ms
64 bytes from 10.45.1.1: icmp_seq=7 ttl=64 time=31.9 ms
```
but still can not ping UE from core network.

Zhang-xzhe, 2024-11-11:
I found that I can ping UE from the docker in core network PC
```
g@g-System-Product-Name:~$ sudo docker exec -it open5gs_5gc bash 
root@ff0ec1367127:/open5gs# ping 10.45.1.2
PING 10.45.1.2 (10.45.1.2) 56(84) bytes of data.
64 bytes from 10.45.1.2: icmp_seq=1 ttl=64 time=46.6 ms
64 bytes from 10.45.1.2: icmp_seq=2 ttl=64 time=45.0 ms
64 bytes from 10.45.1.2: icmp_seq=3 ttl=64 time=44.4 ms
64 bytes from 10.45.1.2: icmp_seq=4 ttl=64 time=43.1 ms
64 bytes from 10.45.1.2: icmp_seq=5 ttl=64 time=42.4 ms
64 bytes from 10.45.1.2: icmp_seq=6 ttl=64 time=41.4 ms
64 bytes from 10.45.1.2: icmp_seq=7 ttl=64 time=40.2 ms
64 bytes from 10.45.1.2: icmp_seq=8 ttl=64 time=39.4 ms
```

pgawlowicz, 2024-11-12:
@Zhang-xzhe any update?
It seems there is some routing issue between the host and Open5gs docker container.

pgawlowicz, 2024-12-05:
@Zhang-xzhe any update?