srsran/srsRAN_Project #924: Finding SD values for connected UE's in the srsRAN Project

adithya-anand-999, 2024-11-10:
## Issue Description ##
[Describe the issue in detail]
As the title says. I need to find a way to collect the SD values of the UE's that connect to the gNB. From my understanding this is something only known by the core network, and is then forwarded to the gNB after the “Registration Accept” message between the core and gNB. This is then forwarded to the DU over the F1-AP interface with the initiatingMessage. 

The reason why I am doing this is because I want to collect KPM values as an average for each slice and not the current per UE basis. As such I am modifying code on the e2sm service module, specifically the meas provider files. From my understaning the ue's are of type asn1::e2sm::ue_id_c. Looking at that struct I dont see a way I can give it information about sd values. 

I am really stuck on how to more forward from here. Any guidance would be greatly appreciated.

I have attached some of the files of interest. 
[e2sm_kpm_du_meas_provider_impl.zip](https://github.com/user-attachments/files/17693471/e2sm_kpm_du_meas_provider_impl.zip)

pgawlowicz, 2024-11-11:
e2sm-kpm defines an optional label, that will be used to filter UEs before returning their individual or aggregated metric.
For example, the [DRB.UEThpDl](https://github.com/srsran/srsRAN_Project/blob/main/lib/e2/e2sm/e2sm_kpm/e2sm_kpm_metric_defs.h#L231) allows subscribing for the measurements with `FIVE_QI_LABEL, SLICE_ID_LABEL, PLMN_ID_LABEL, NO_LABEL`. 
We are working on label support, but the release date has not yet been set.

adithya-anand-999, 2024-11-11:
Yes I saw those functions as todo's. But the team I am on now is looking to get this done now as part of the research project I am working on. Do you know if any workflow exists for being able to access SD values for each UE.

risa-002, 2024-11-21:
Hello adithya did you able to find anything about finding sd values for connected UEs ? I am also looking for the similar thing.