srsran/srsRAN_Project #926: Failed to run python version KPM monitor xAPP

lingyan98, 2024-11-12:
## Issue Description ##
Failed to run python version KPM monitor xAPP 

## Setup Details ##
`~/project/oran/flexric/build/examples/xApp/python3$ python3 ./monitor/xapp_oran_moni.py -c ./xapp_mon_e2sm_kpm.conf
Setting the config -c file to ./xapp_mon_e2sm_kpm.conf
[LibConf]: loading service models from SM_DIR: /usr/local/lib/flexric/
[LibConf]: reading configuration for xApp
[LibConf]: NearRT_RIC IP: 127.0.0.1
[LibConf]: E42_Port Port: 36422
[LibConf]: Sub_ORAN_SM Name: KPM, Time: 1000
[LibConf]: format 1, RAN type ngran_gNB_DU, actions = DRB.UEThpDl DRB.UEThpUl 
[xAap]: Initializing ... 
[xApp]: nearRT-RIC IP Address = 127.0.0.1, PORT = 36422
[E2-AGENT]: Opening plugin from path = /usr/local/lib/flexric/libmac_sm.so 
[E2-AGENT]: Opening plugin from path = /usr/local/lib/flexric/librc_sm.so 
[E2-AGENT]: Opening plugin from path = /usr/local/lib/flexric/libkpm_sm.so 
[E2-AGENT]: Opening plugin from path = /usr/local/lib/flexric/libtc_sm.so 
[E2-AGENT]: Opening plugin from path = /usr/local/lib/flexric/librlc_sm.so 
[E2-AGENT]: Opening plugin from path = /usr/local/lib/flexric/libgtp_sm.so 
[E2-AGENT]: Opening plugin from path = /usr/local/lib/flexric/libslice_sm.so 
[E2-AGENT]: Opening plugin from path = /usr/local/lib/flexric/libpdcp_sm.so 
[NEAR-RIC]: Loading SM ID = 142 with def = MAC_STATS_V0 
[NEAR-RIC]: Loading SM ID = 3 with def = ORAN-E2SM-RC 
[NEAR-RIC]: Loading SM ID = 2 with def = ORAN-E2SM-KPM 
[NEAR-RIC]: Loading SM ID = 146 with def = TC_STATS_V0 
[NEAR-RIC]: Loading SM ID = 143 with def = RLC_STATS_V0 
[NEAR-RIC]: Loading SM ID = 148 with def = GTP_STATS_V0 
[NEAR-RIC]: Loading SM ID = 145 with def = SLICE_STATS_V0 
[NEAR-RIC]: Loading SM ID = 144 with def = PDCP_STATS_V0 
[xApp]: E42 SETUP-REQUEST tx
[xApp]: E42 SETUP-RESPONSE rx 
[xApp]: xApp ID = 8 
[xApp]: Registered E2 Nodes = 1 
Connected E2 nodes = 1
Global E2 Node [0]: PLMN MCC = 1
Global E2 Node [0]: PLMN MNC = 1
[xApp]: generating action definition from the list of required measurement data:
DRB.UEThpDl
DRB.UEThpUl
[xApp]: reporting period = 1000 [ms]
python3: /home/ics/project/oran/flexric/src/lib/e2ap/v2_03/e2ap_types/common/e2ap_global_node_id.c:49: eq_global_e2_node_id: Assertion `m0->cu_du_id != NULL && m1->cu_du_id != NULL' failed.
Aborted (core dumped)`

## Additional Information ##
The c version KPM monitor xApp can be run successfully: ./flexric/build/examples/xApp/c/monitor/xapp_oran_moni -c ./xapp_mon_e2sm_kpm.conf

yagoda, 2024-11-12:
It looks like there is a mismatch in the DU/CU_ID used in the xApp "Assertion m0->cu_du_id != NULL && m1->cu_du_id != NULL'" please check these values. Also it is always helpful to post a PCAP of the e2 interface.

lingyan98, 2024-11-13:
Hello, thank you for your reply. 
### The gnb_zmq configure is shown as follow

> cu_cp:
  amf:
    addr: 10.53.1.2                 # The address or hostname of the AMF.
    port: 38412
    bind_addr: 10.53.1.1            # A local IP that the gNB binds to for traffic from the AMF.
    supported_tracking_areas:
      - tac: 7
        plmn_list:
          - plmn: "00101"
            tai_slice_support_list:
              - sst: 1
  inactivity_timer: 7200            # Sets the UE/PDU Session/DRB inactivity timer to 7200 seconds. Supported: [1 - 7200].

ru_sdr:
  device_driver: zmq                # The RF driver name.
  device_args: tx_port=tcp://127.0.0.1:2000,rx_port=tcp://127.0.0.1:2001,base_srate=11.52e6 # Optionally pass arguments to the selected RF driver.
  srate: 11.52                      # RF sample rate might need to be adjusted according to selected bandwidth.
  tx_gain: 75                       # Transmit gain of the RF might need to adjusted to the given situation.
  rx_gain: 75                       # Receive gain of the RF might need to adjusted to the given situation.

cell_cfg:
  dl_arfcn: 368500                  # ARFCN of the downlink carrier (center frequency).
  band: 3                           # The NR band.
  channel_bandwidth_MHz: 10         # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 15                    # Subcarrier spacing in kHz used for data.
  plmn: "00101"                     # PLMN broadcasted by the gNB.
  tac: 7                            # Tracking area code (needs to match the core configuration).
  pdcch:
    common:
      ss0_index: 0                  # Set search space zero index to match srsUE capabilities
      coreset0_index: 6             # Set search CORESET Zero index to match srsUE capabilities
    dedicated:
      ss2_type: common              # Search Space type, has to be set to common
      dci_format_0_1_and_1_1: false # Set correct DCI format (fallback)
  prach:
    prach_config_index: 1           # Sets PRACH config to match what is expected by srsUE

log:
  filename: /tmp/gnb.log            # Path of the log file.
  all_level: info                   # Logging level applied to all layers.
  hex_max_size: 0

pcap:
  mac_enable: false                 # Set to true to enable MAC-layer PCAPs.
  mac_filename: /tmp/gnb_mac.pcap   # Path where the MAC PCAP is stored.
  ngap_enable: false                # Set to true to enable NGAP PCAPs.
  ngap_filename: /tmp/gnb_ngap.pcap # Path where the NGAP PCAP is stored.
  e2ap_enable: true                 # Set to true to enable E2AP PCAPs.
  e2ap_filename: /tmp/gnb_e2ap.pcap # Path where the E2AP PCAP is stored.

e2:
  enable_du_e2: true                # Enable DU E2 agent (one for each DU instance)
  e2sm_kpm_enabled: true            # Enable KPM service module
  e2sm_rc_enabled: true             # Enable RC service module
  addr: 127.0.0.1                   # RIC IP address
  #bind_addr: 127.0.0.100           # A local IP that the E2 agent binds to for traffic from the RIC. ONLY required if running the RIC on a separate machine. 
  port: 36421                       # RIC port

metrics: 
  rlc_json_enable: 1                # Enable RLC metrics reporting (need to deliver measurements to E2SM_KPM service model)
  rlc_report_period: 1000           # Set reporting period to 1s

### xapp_mon_e2sm_kpm 
`SM_DIR = "/usr/local/lib/flexric/"

# supported name = xApp
Name = "xApp"
[... 129 lines cut ...]
......E2 Node Me
asurement for mu
ltiple UEs.....@
CQI..DRB.AirIfDe
layUl.`DRB.Packe
tSuccessRateUlgN
BUu..DRB.RlcDela
yUl..DRB.RlcPack
etDropRateDl..DR
B.RlcSduDelayDl.
.DRB.RlcSduTrans
mittedVolumeDL..
DRB.RlcSduTransm
ittedVolumeUL.@D
RB.UEThpDl.@DRB.
UEThpUl.`RACH.Pr
eambleDedCell..R
RU.PrbAvailDl..R
RU.PrbAvailUl.`R
RU.PrbTotDl.`RRU
.PrbTotUl..RRU.P
rbUsedDl..RRU.Pr
bUsedUl.`RSRP.`R
SRQ.....`
[image]

yagoda, 2024-11-13:
Can you please share /tmp/gnb_e2ap.pcap

lingyan98, 2024-11-13:
> Can you please share /tmp/gnb_e2ap.pcap

Thank you for your reply. I have solved this issue according to suggestion in this [link](https://github.com/srsran/srsRAN_Project/discussions/822). Set the gnb_du_id large than 0. But I don't know why this happen?

pgawlowicz, 2024-11-13:
did you compile flexric with those flags?
```
cmake -DKPM_VERSION=KPM_V3_00 -DXAPP_DB=NONE_XAPP ../
```

Coul you try using