srsran/srsRAN_Project #930: Can't detect 5G network

platyman, 2024-11-13:
## Issue Description ##
Everything seems to be starting up, but I'm seeing a series of strange messages in the logs + I can't detect my network on the UE

## Setup Details ##
I work with an ansdr E316 from microphase with an external GPS. Here is my configuration file (from GnodeB):

```

cu_cp:
  amf:
    addr: 127.0.0.5
    port: 38412
    bind_addr: 127.0.0.1
    supported_tracking_areas:
      - tac: 7
        plmn_list:
          - plmn: "20690"
            tai_slice_support_list:
              - sst: 1

ru_sdr:
  device_driver: uhd
  device_args: type=ant,num_recv_frames=64,num_send_frames=64,master_clock_rate=30.72e6
  srate: 30.72
  otw_format: sc12
  tx_gain: 60
  rx_gain: 30
  clock: default
  sync: default

cell_cfg:
  dl_arfcn: 667000
  band: 77
  channel_bandwidth_MHz: 10
  nof_antennas_dl: 2
  nof_antennas_ul: 2
  common_scs: 30
  plmn: "20690"
  tac: 7
  pci: 1
  prach:
    prach_config_index: 0

log:
  filename: /tmp/gnb.log
  all_level: debug

pcap:
  mac_enable: false
  mac_filename: /tmp/gnb_mac.pcap
  ngap_enable: false
  ngap_filename: /tmp/gnb_ngap.pcap

```

## Expected Behavior ##
I should see my network in my UE

## Actual Behaviour ##
When trying to use my gps (by setting clock and sync to gpsdo), I've got the error:

Could not lock reference clock source.
srsRAN ERROR: Unable to create radio session.

I've got in my terminal some [DEBUG] [AD936X] baseband bandwidth too large for current sample rate. Setting bandwidth to: 5e+07 but it's impossible to manually set the bandwidth to 50Mhz (srate to weak) which is a strange behavior. I can't increase my srate neither because it seems like it needs to be equal to the clockrate and my clockrate is limited to 30.72 Mhz (I tried to overcome this problem by using the GPS but as stated before it isn't working). My ideal configuration would be to use a frequency of 4005 Mhz with a 20 Mhz bandwidth and I don't really care (for now) about the srate.

## Additional Information ##
Here is the "complete" trace of execution of gnb

ubuntu@ubuntu:~/srsRAN_Project$ sudo ./build/apps/gnb/gnb -c configs/gnb_E316.yml

--== srsRAN gNB (commit 9d5dd742a) ==--

The PRACH detector will not meet the performance requirements with the configuration {Format 0, ZCZ 0, SCS 1.25kHz, Rx ports 1}.
Lower PHY in quad executor mode.
Available radio types: uhd.
[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; UHD_4.1.0.0-0-b90c92a8
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Making USRP object with args 'type=ant,num_recv_frames=64,num_send_frames=64,master_clock_rate=30.72e6'
[INFO] [ANT] Detected Device: ANTSDR
[INFO] [ANT] Detecting internal GPSDO....
[INFO] [GPS] Found a generic NMEA GPS device
[INFO] [ANT] Initialize CODEC control...
[INFO] [ANT] Initialize Radio control...
[DEBUG] [AD936X] baseband bandwidth too large for current sample rate. Setting bandwidth to: 5e+07
[DEBUG] [AD936X] baseband bandwidth too large for current sample rate. Setting bandwidth to: 5e+07
[DEBUG] [AD936X] baseband bandwidth too large for current sample rate. Setting bandwidth to: 5e+07
[DEBUG] [AD936X] baseband bandwidth too large for current sample rate. Setting bandwidth to: 5e+07
[DEBUG] [AD936X] baseband bandwidth too large for current sample rate. Setting bandwidth to: 5e+07
[DEBUG] [AD936X] baseband bandwidth too large for current sample rate. Setting bandwidth to: 5e+07
[DEBUG] [AD936X] baseband bandwidth too large for current sample rate. Setting bandwidth to: 5e+07
[DEBUG] [AD936X] baseband bandwidth too large for current sample rate. Setting bandwidth to: 5e+07
[INFO] [ANT] Performing register loopback test...
[INFO] [ANT] Register loopback test passed
[INFO] [ANT] Performing register loopback test...
[INFO] [ANT] Register loopback test passed
[DEBUG] [AD936X] Performing CODEC loopback test...
[DEBUG] [AD936X] CODEC loopback test passed.
[DEBUG] [AD936X] Performing CODEC loopback test...
[DEBUG] [AD936X] CODEC loopback test passed.
[INFO] [ANT] Asking for clock rate 30.720000 MHz...
[INFO] [ANT] Actually got clock rate 30.720000 MHz.
[DEBUG] [CORES] Performing timer loopback test...
[DEBUG] [CORES] Timer loopback test passed.
[DEBUG] [CORES] Performing timer loopback test...
[DEBUG] [CORES] Timer loopback test passed.
[INFO] [MULTI_USRP]     1) catch time transition at pps edge
[INFO] [MULTI_USRP]     2) set times next pps (synchronously)
[DEBUG] [CONVERT] get_converter: For converter ID: conversion ID
  Input format:  fc32
  Num inputs:    1
  Output format: sc12_item32_le
  Num outputs:   1
 Using prio: 0
[DEBUG] [CONVERT] get_converter: For converter ID: conversion ID
  Input format:  fc32
  Num inputs:    1
  Output format: sc12_item32_le
  Num outputs:   1
 Using prio: 0
[DEBUG] [CONVERT] get_converter: For converter ID: conversion ID
  Input format:  sc12_item32_le
  Num inputs:    1
  Output format: fc32
  Num outputs:   1
 Using prio: 0
[DEBUG] [CONVERT] get_converter: For converter ID: conversion ID
  Input format:  sc12_item32_le
  Num inputs:    1
  Output format: fc32
  Num outputs:   1
 Using prio: 0
Cell pci=1, bw=10 MHz, 2T2R, dl_arfcn=667000 (n77), dl_freq=4005.0 MHz, dl_ssb_arfcn=667008, ul_freq=4005.0 MHz

N2: Connection to AMF on 127.0.0.5:38412 completed
^CCould not stop application after 5 seconds. Forcing exit.
Killed

pgawlowicz, 2024-11-13:
Please be aware that the ANTSDR E316 is not officially supported. 
Therefore, we're unable to address issues related to this device. We kindly request that you move your post to the Discussion section, where our community members may be able to assist you further.

closed 2024-11-13 by pgawlowicz

platyman, 2024-11-13:
Great idea! Feel free to voncert this issue in a discussion (it don't seems like we have the permissions to do so). I would point out that we have managed to run a 4G network (with the srsran 4g project and uhd) on the E310 without any problems (we connect with a Nokia TA-1362). If we try to observe the frequency band that should be occupied, we don't observe any activity on it (in 5G). However, everything in the logs seems to indicate that the gnb has started up correctly.

pgawlowicz, 2024-11-13:
Please open a new thread in Discussion section.