srsran/srsRAN_Project #933: srsRAN gNB with srsUE over usrp b210 frontends

thaxir, 2024-11-13:
## Issue Description ##
[Describe the issue in detail]
I'm following the recent tutorial 'srsRAN gNB with srsUE' from here https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html. However, the UE (srsue) never obtains an IP from srsgnb staying at this :

```RRC Connected
RF status: O=3, U=0, L=0
Scheduling request failed: releasing RRC connection...
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=1934
```

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

2 USRP b210 as RF frontend
2 NUCs
Open5gs
srsRAN_Project for gnb
srsRAN_4G for the UE
Internal clock/no GPSDO

## Expected Behavior ##
[What you expect to happen]
Cannot use srsue since it is not properly connected to the gnb
## Actual Behaviour ##
[What happens instead e.g. error message]
Seems like  UE attempted a scheduling request but didn't receive a response from the gNB. Not sure though
[image]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]
[gnb.log.txt](https://github.com/user-attachments/files/17734391/gnb.log.txt)
[ue.log.txt](https://github.com/user-attachments/files/17734395/ue.log.txt)

[ue.conf.txt](https://github.com/user-attachments/files/17734375/ue.conf.txt)

[gnb-config.yml.txt](https://github.com/user-attachments/files/17734371/gnb-config.yml.txt)

pgawlowicz, 2024-11-13:
UL signal is too strong at the gnb. Please try to tune down srsUE tx_gain and gnb rx_gain.

thaxir, 2024-11-19:
any suggestions on this topic ??

pgawlowicz, 2024-11-19:
could you try to apply the patch from this comment to srsUE, recompile and try again? 
https://github.com/srsran/srsRAN_Project/issues/826#issuecomment-2370588502

thaxir, 2024-11-21:
@pgawlowicz  Piotr  we have done the changes you suggested .
 After restarting serveral times  srsue and gnb the srsue   logs are changed as you see in the snapshot below
[image]

Gnb reports this:
[image]

and amf in open5gs which is in the same machine with gnb reports this :
[image]

which seems that amf rejects the srsue . Do you know why? (

And another question is why do i need to restart several times both srsue and gnb in order to have this connection open and not the **Scheduling request failed : releasing RRC connection**?
I attach also the logs  for ue and gnb.
[ue.log.3.txt](https://github.com/user-attachments/files/17845645/ue.log.3.txt)
[gnb.log.3.txt](https://github.com/user-attachments/files/17845726/gnb.log.3.txt)

[gnb.log.3.txt](https://github.com/user-attachments/files/17845691/gnb.log.3.txt)

thaxir, 2024-11-26:
Any suggestions  on this?

pgawlowicz, 2024-11-26:
it seems the problem is on the core network side, did you add the UE to the subscriber database?

thaxir, 2024-12-02:
Hi @pgawlowicz  again . So far i have tried to experiment with  the position of the usrps  if the radio link and the line of site was the error but i am getting on the srsue
[image]

On the gnb 
[image]

I was able to get some movement over the amf

[image]

 i started cross checking the database i see the same  values in the srsue and open5gs (deleting and re-adding the subscriber)
 
[image] 
[image]

Searching for others with this issue i came across this https://github.com/srsran/srsRAN_Project/issues/451 which seems to have a lot of similarities with my case.  However, on the newer version of srsran gnb  i don't  know where to put the ``` inactivity_timer: 7200```

Does this inactivity timer can be added somewhere in the gnb ?

pgawlowicz, 2024-12-02:
could you share gnb logs in debug mode?

thaxir, 2024-12-02:
@pgawlowicz  yes here 
i don t  know if i have captured all because there are a lot of msgs.
[gnb-debug.txt](https://github.com/user-attachments/files/17977674/gnb-debug.txt)

thaxir, 2024-12-02:
Apologies, 
here
[gnb-debug.txt](https://github.com/user-attachments/files/17977854/gnb-debug.txt)

pgawlowicz, 2024-12-02:
Could you tell me which code version (commit id) do you use? it is missing in the logs.

thaxir, 2024-12-02:
How to i see this?

pgawlowicz, 2024-12-02:
git log -n1

thaxir, 2024-12-02:
[image]
Merge: eea87b1d8 f8978a697

thaxir, 2024-12-02:
Should i use  binaries from  Package Manager on ubuntu or build it from the source?

pgawlowicz, 2024-12-02:
both ways  work

thaxir, 2024-12-02:
@pgawlowicz i just installed  srsran-project  (apt-get install)  from the package manager. 
nuc6@nuc6:/usr/share/srsran$ gnb --version
srsRAN 5G gNB version 23.10.1 ()

Should i proceed to testing ?

thaxir, 2024-12-02:
Ok done 
[image]

No i get  this
[image]

It seems like a config problem 
Here is the config of gnb 
[image]

pgawlowicz, 2024-12-02:
You have to install uhd driver if you are building from source. Please follow our installation guide https://docs.srsran.com/projects/project/en/latest/user_manuals/source/installation.html

thaxir, 2024-12-02:
I have already installed the uhd from the previous run 
[image]

Do i need to redo this installation?

pgawlowicz, 2024-12-02:
could you check if cmake detects the UHD correctly?

thaxir, 2024-12-02:
No it is not detected so i build 3.15 again . I will send you the update tomorrow. Thank you very much this far for the help

thaxir, 2024-12-03:
Hi @pgawlowicz again, so the update . I have installed uhd 3.15 .
Gnb : 
[image]
Here the logs 
[gnb-debug(2).txt](https://github.com/user-attachments/files/17992430/gnb-debug.2.txt)

The srsue 
[image]

pgawlowicz, 2024-12-03:
and the srsUE config and logs?

thaxir, 2024-12-03:
Ue logs:
[ue.txt.zip](https://github.com/user-attachments/files/17992751/ue.txt.zip)
Ue conf:
[ue-conf.txt](https://github.com/user-attachments/files/17992777/ue-conf.txt)

pgawlowicz, 2024-12-03:
PUSCH SNR is very low, probably caused by a very high CFO. Do you have an external clock to connect to b210s?

thaxir, 2024-12-03:
No i don t have ext clock. I could order  one   but it will take time . I don t know if i could  connect the two usrps through  coaxial cable for using on usrp as master timer and one as slave . 
Is there any configuration i can do now without a gpsdo?

pgawlowicz, 2024-12-03:
you might try to tune TX an RX gains for both gnb and srsUE, also time_adv_nsamples in srsUE config. Some users are able to run this setup without an external clocks. Did you check if it works correctly over zmq?

thaxir, 2024-12-03:
Yes in the past i did use zmq before shifting to ueransim. For the time being i need to have radio link between ue and gnb. Do you have any reference values for start playing with the gnb /ue /time_adv ?
If i replace srsue with a cots ue would this still be a problem?

pgawlowicz, 2024-12-04:
Usually, `tx_gain: 60` and `rx_gain:30-40` and 'time_adv = auto or 300'. Did you apply the srsUE patch that I shared in one of the previous posts?
I think b210 should work correctly with COTS. You might give it a try.