srsran/srsRAN_Project #939: Unable to connect UE to Open5GS core network, AMF sends Registration Reject

PsArzzz, 2024-11-17:
## Issue Description ##
I am using Open5GS as the core network and srsRAN to set up the gNodeB and UE, with ZeroMQ for communication between the gNodeB and UE. The UE successfully connects to the gNodeB but fails to register with the core network. After sending a Registration Request to the AMF, the request is rejected with a Registration Reject message. Eventually, the UE logs show "**Received RRC Release**." How can I resolve this and successfully connect the UE to the core network?

## Setup Details ##
The setup consists of four virtual machines, each running Ubuntu 22.04. The details are as follows:

1、Open5GS Control Plane (5GC C-Plane): IP Address: 192.168.83.131
2、Open5GS User Plane (5GC U-Plane): IP Address: 192.168.83.130
3、srsRAN Project ZMQ RAN (gNodeB): IP Address: 192.168.83.136
4、srsRAN 4G ZMQ UE (NR-UE): IP Address: 192.168.83.137

## Expected Behavior ##
The UE should be able to register successfully with the core network, with the AMF accepting the Registration Request.

## Actual Behaviour ##

1、The following output appears on the srsUE terminal:
a@a:~/srsRAN_4G/build/srsue$ sudo ./src/srsue ue_zmq.conf
```
Reading configuration file ue_zmq.conf...

Built in Release mode using commit ec29b0c1f on branch master.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://192.168.83.137:2001,rx_port=tcp://192.168.83.136:2000,base_srate=23.04e6
Supported RF device list: zmq file
CHx base_srate=23.04e6
Current sample rate is 1.92 MHz with a base rate of 23.04 MHz (x12 decimation)
CH0 rx_port=tcp://192.168.83.136:2000
CH0 tx_port=tcp://192.168.83.137:2001
Current sample rate is 23.04 MHz with a base rate of 23.04 MHz (x1 decimation)
Current sample rate is 23.04 MHz with a base rate of 23.04 MHz (x1 decimation)
Waiting PHY to initialize ... done!
Attaching UE...
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=334
Random Access Complete.     c-rnti=0x4601, ta=0
RRC Connected
Received RRC Release
```
2、Logs from UE:
[ue.log](https://github.com/user-attachments/files/17790642/ue.log)

3、Logs from gNB:
[gnb.log](https://github.com/user-attachments/files/17790636/gnb.log)

4、Logs from AMF:
```
11/17 20:24:46.731: [amf] DEBUG: amf_state_operational(): OGS_EVENT_NAME_SBI_CLIENT (../src/amf/amf-sm.c:81)
11/17 20:24:46.731: [sbi] DEBUG: ogs_sbi_nf_state_registered(): OGS_EVENT_NAME_SBI_CLIENT (../lib/sbi/nf-sm.c:198)
11/17 20:24:56.731: [amf] DEBUG: amf_state_operational(): OGS_EVENT_NAME_SBI_TIMER (../src/amf/amf-sm.c:81)
11/17 20:24:56.731: [sbi] DEBUG: ogs_sbi_nf_state_registered(): OGS_EVENT_NAME_SBI_TIMER (../lib/sbi/nf-sm.c:198)
[{"op":"replace","path":"/nfStatus","value":"REGISTERED"},{"op":"replace","path":"/load","value":0}]11/17 20:24:56.731: [sbi] DEBUG: [PATCH] http://127.0.0.200:7777/nnrf-nfm/v1/nf-instances/edad8896-a4dc-41ef-8084-0bf0484790fe (../lib/sbi/client.c:728)
11/17 20:24:56.731: [sbi] DEBUG: SENDING...[100] (../lib/sbi/client.c:497)
11/17 20:24:56.731: [sbi] DEBUG: [{"op":"replace","path":"/nfStatus","value":"REGISTERED"},{"op":"replace","path":"/load","value":0}] (../lib/sbi/client.c:499)
11/17 20:24:56.736: [sbi] DEBUG: [204:PATCH] http://127.0.0.200:7777/nnrf-nfm/v1/nf-instances/edad8896-a4dc-41ef-8084-0bf0484790fe (../lib/sbi/client.c:675)
11/17 20:24:56.736: [sbi] DEBUG: RECEIVED[0] (../lib/sbi/client.c:686)
11/17 20:24:56.736: [amf] DEBUG: amf_state_operational(): OGS_EVENT_NAME_SBI_CLIENT (../src/amf/amf-sm.c:81)
11/17 20:24:56.736: [sbi] DEBUG: ogs_sbi_nf_state_registered(): OGS_EVENT_NAME_SBI_CLIENT (../lib/sbi/nf-sm.c:198)
11/17 20:25:00.230: [amf] INFO: gNB-N2 accepted[192.168.83.136]:34425 in ng-path module (../src/amf/ngap-sctp.c:113)
11/17 20:25:00.230: [amf] DEBUG: amf_state_operational(): AMF_EVENT_NGAP_LO_ACCEPT (../src/amf/amf-sm.c:81)
11/17 20:25:00.230: [amf] INFO: gNB-N2 accepted[192.168.83.136] in master_sm module (../src/amf/amf-sm.c:796)
11/17 20:25:00.244: [amf] DEBUG: ngap_state_initial(): ENTRY (../src/amf/ngap-sm.c:28)
11/17 20:25:00.244: [amf] DEBUG: ngap_state_operational(): ENTRY (../src/amf/ngap-sm.c:55)
11/17 20:25:00.244: [amf] INFO: [Added] Number of gNBs is now 1 (../src/amf/context.c:1238)
11/17 20:25:00.244: [amf] DEBUG: SCTP_ASSOC_CHANGE:[T:32769, F:0x0, S:0, I/O:10/30] (../src/amf/ngap-sctp.c:152)
11/17 20:25:00.244: [amf] DEBUG: SCTP_COMM_UP (../src/amf/ngap-sctp.c:161)
11/17 20:25:00.244: [amf] DEBUG: amf_state_operational(): AMF_EVENT_NGAP_LO_SCTP_COMM_UP (../src/amf/amf-sm.c:81)
11/17 20:25:00.244: [amf] INFO: gNB-N2[192.168.83.136] max_num_of_ostreams : 30 (../src/amf/amf-sm.c:842)
11/17 20:25:00.244: [amf] DEBUG: amf_state_operational(): AMF_EVENT_NGAP_MESSAGE (../src/amf/amf-sm.c:81)
11/17 20:25:00.245: [amf] DEBUG: ngap_state_operational(): AMF_EVENT_NGAP_MESSAGE (../src/amf/ngap-sm.c:55)
11/17 20:25:00.245: [amf] DEBUG: NGSetupRequest (../src/amf/ngap-handler.c:144)
[... 87 lines cut ...]
5、amf.yaml(On Open5GS 5GC C-Plane)
[amf_yaml.txt](https://github.com/user-attachments/files/17790665/amf_yaml.txt)

6、nrf.yaml(On Open5GS 5GC C-Plane)
[nrf_yaml.txt](https://github.com/user-attachments/files/17790668/nrf_yaml.txt)

7、smf.yaml(On Open5GS 5GC C-Plane)
[smf_yaml.txt](https://github.com/user-attachments/files/17790670/smf_yaml.txt)

8、upf.yaml(On Open5GS 5GC U-Plane)
[upf_yaml.txt](https://github.com/user-attachments/files/17790672/upf_yaml.txt)

9、gnb_zmq.yaml
[gnb_zmq_yaml.txt](https://github.com/user-attachments/files/17790674/gnb_zmq_yaml.txt)

10、ue_zmq.conf
[ue_zmq_conf.txt](https://github.com/user-attachments/files/17790681/ue_zmq_conf.txt)

## Steps to reproduce the problem ##
1、Start the Open5GS core network on VMs with IP addresses 192.168.83.131 and 192.168.83.130.
2、Start the srsRAN gNodeB on VM3 and ensure it connects to the AMF.
3、Start the srsRAN UE on VM4 and attempt to register with the network.

## Additional Information ##
BTW, how can I use Wireshark to capture RRC messages?

PsArzzz, 2024-11-17:
**The gNB and UE times do not match the system time of the AMF, but they occur simultaneously.**
This is the message captured on the AMF.
registration request:
```
Plain NAS 5GS Message
    Extended protocol discriminator: 5G mobility management messages (126)
    0000 .... = Spare Half Octet: 0
    .... 0000 = Security header type: Plain NAS message, not security protected (0)
    Message type: Registration request (0x41)
    5GS registration type
        .... 1... = Follow-On Request bit (FOR): Follow-on request pending
        .... .001 = 5GS registration type: initial registration (1)
    NAS key set identifier
        0... .... = Type of security context flag (TSC): Native security context (for KSIAMF)
        .111 .... = NAS key set identifier: 7
    5GS mobile identity
        Length: 13
        0... .... = Spare: 0
        .000 .... = SUPI format: IMSI (0)
        .... 0... = Spare: 0
        .... .001 = Type of identity: SUCI (1)
        Mobile Country Code (MCC): Unknown (1)
        Mobile Network Code (MNC): Unknown (01)
        Routing indicator: 0000
        .... 0000 = Protection scheme Id: NULL scheme (0)
        Home network public key identifier: 0
        MSIN: 0000000001
    UE security capability
        Element ID: 0x2e
        Length: 2
        1... .... = 5G-EA0: Supported
        .1.. .... = 128-5G-EA1: Supported
        ..1. .... = 128-5G-EA2: Supported
        ...1 .... = 128-5G-EA3: Supported
        .... 0... = 5G-EA4: Not supported
        .... .0.. = 5G-EA5: Not supported
        .... ..0. = 5G-EA6: Not supported
        .... ...0 = 5G-EA7: Not supported
        0... .... = 5G-IA0: Not supported
        .1.. .... = 128-5G-IA1: Supported
        ..1. .... = 128-5G-IA2: Supported
        ...1 .... = 128-5G-IA3: Supported
        .... 0... = 5G-IA4: Not supported
        .... .0.. = 5G-IA5: Not supported
        .... ..0. = 5G-IA6: Not supported
        .... ...0 = 5G-IA7: Not supported
    NSSAI - Requested NSSAI
        Element ID: 0x2f
        Length: 5
        S-NSSAI 1
            Length: 4
            Slice/service type (SST): eMBB (1)
            Slice differentiator (SD): 1
```
registration reject:
```
Item 2: id-NAS-PDU
    ProtocolIE-Field
        id: id-NAS-PDU (38)
        criticality: reject (0)
        value
            NAS-PDU: 7e00445f
                Non-Access-Stratum 5GS (NAS)PDU
                    Plain NAS 5GS Message
                        Extended protocol discriminator: 5G mobility management messages (126)
                        0000 .... = Spare Half Octet: 0
                        .... 0000 = Security header type: Plain NAS message, not security protected (0)
                        Message type: Registration reject (0x44)
                        5GMM cause
                            5GMM cause: Semantically incorrect message (95)
```

pgawlowicz, 2024-11-18:
please try with this patch:
https://github.com/srsran/srsRAN_4G/pull/1214

PsArzzz, 2024-11-18:
Thank you very much for your response. After removing all the slicing configurations, the UE successfully connected to the core network.

closed 2024-11-18 by PsArzzz