srsran/srsRAN_Project #941: Problem integrating E2 interface with external nearRT-RIC

EspiMCF, 2024-11-19:
Hello, 

I am facing issues while I am integrating the srsRAN solution with a nearRT-RIC running on a docker compose.

My current setup is the srsRAN ( commit e73b46182 ) on a baremetal server, with Intel E810 and a Broadcom Quad 25Gb Ethernet. The solution is able to connect to the AMF correctly, a full Core Simulator and also to a O-RU FHI simulator. The nearRT-RIC I am trying to integrate is a comercial solution. But when I am adding the E2 interface setup configuration section to the same yaml file I was using previously, it seems it can not reach the nearRT-RIC somehow. But from the nearRT-RIC I can ping the srsRAN server E2 interface and from the server I can ping the nearRT-RIC E2 address. The core is connected to the Intel card, and the RU and nearRT-RIC to the Broadcom card. As far as I know, my nearRT-RIC is running on docker and it is exposing the port 30000:
```
dric-service-e2node                        NodePort    <none>        5000:30000/SCTP 
```
And this IP: 
```
dric_e2_intf Link encap:Ethernet 
          inet addr:10.10.20.162
```
When I am executing the config file without E2 section, I am able to see that is attached to the AMF and the gNB keep working and reporting logs. And I can even attach UEs and make some testing.
But when I add the E2 configuration described here:
````
e2:
  enable_du_e2: true
  e2sm_kpm_enabled: true
  e2sm_rc_enabled: true
  addr: 10.10.20.162
  bind_addr: 10.10.20.171          
  port: 30000
````

It reports the following message: 
````
--== srsRAN gNB (commit e73b46182) ==--
2024-11-19T10:05:18.146254 [GNB     ] [I] Built in Release mode using commit e73b46182 on branch main
Connecting to NearRT-RIC on 10.10.20.162:30000
Failed to connect SCTP socket to 10.10.20.162:30000. error="Connection refused" timeout=0ms
2024-11-19T10:05:19.447524 [SCTP-GW ] [E] Failed to connect SCTP socket to 10.10.20.162:30000. error="Connection refused" timeout=0ms
srsRAN ERROR: Failed to create SCTP gateway.
````

It seems it is not able to configure the interface somehow. Also, what I am expecting in the nearRT-RIC is some message from the gNB in order to reject in a first try but with the values in order to admit it in the database, something like this:
````
 ERROR         E2T_E2AP     E2T_DATA_01 fd 8 E2node Global Id Not Provisioned: {
   gNB {
      global_gNB_ID {
         plmn_id = 0xXXXXXX
         gnb_id {
            gnb-ID = { 32, 0xXX 0xXX 0xXX 0xXX }
         }
      }
   }
}

ERROR         E2T_E2AP     E2T_DATA_01 Provisioned E2node missing in e2_htab table
INFO         E2T_E2AP     E2T_DATA_01 E2node fd 8 sent invalid e2setup
````

But even this message does not appear in the nearRT-RIC. Also something I am missing is the bind_interface variable for this interface, as it is not the same as the AMF or RU FHI.

Here I attach my configuration file I am currently using and as I said, I am running on commit e73b46182. 
[gnb-2x2-50-fdd-nearRT.txt](https://github.com/user-attachments/files/17813662/gnb-2x2-50-fdd-nearRT.txt)

Thanks and regards.

pgawlowicz, 2024-11-20:
could you share logs and e2ap pcap file?

EspiMCF, 2024-11-20:
Hi, do you need any specific level/layers of log? 
Here I have attached a zip file where you can find the pcap of the E2 interface from the srsRAN solution.
[DUinterfaceE2.zip](https://github.com/user-attachments/files/17832540/DUinterfaceE2.zip)

EspiMCF, 2024-11-21:
@s21sm As far as I know, the port that the container of the Near-RT RIC is exposing is 30000, I have tried also the default value port 36421 but it is the same behavior.

pgawlowicz, 2024-11-21:
Can you ping the srsgnb host machine from the `dric-service-e2node` contanier?

Is this interface (and IP) part of the docker container or the server running docker?
```
dric_e2_intf Link encap:Ethernet 
          inet addr:10.10.20.162
```

pgawlowicz, 2024-11-22:
could you show the routing table of Near-RT RIC ?

EspiMCF, 2024-11-22:
```
$ ip route show
default via 169.254.1.1 dev eth0
169.254.1.1 dev eth0 scope link

$ ifconfig
lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope: Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          

eth0      Link encap:Ethernet  HWaddr fa:35:7e:51:6e:fa
          inet addr:100.106.143.159  Bcast:0.0.0.0  Mask:255.255.255.255
          inet6 addr: fe80::f835:7eff:fe51:6efa/64 Scope: Link
          UP BROADCAST RUNNING MULTICAST  MTU:1480  Metric:1

dric_e2_intf Link encap:Ethernet  HWaddr 00:50:56:ae:1f:c2  Driver iavf
          inet addr:10.10.20.162  Bcast:10.10.20.191  Mask:255.255.255.224
          inet6 addr: fe80::250:56ff:feae:1fc2/64 Scope: Link
          UP BROADCAST RUNNING PROMISC MULTICAST  MTU:1500  Metric:1
```

pgawlowicz, 2024-11-22:
and also the routing table of the DU server?

EspiMCF, 2024-11-25:
````
$ ip route show
default via 10.10.1.1 dev enp81s0f3 proto static
10.10.1.0/24 dev enp81s0f3 proto kernel scope link src 10.10.1.215
10.10.2.0/24 dev eno8303np0.1002 proto kernel scope link src 10.10.2.50
10.10.20.160/27 dev eno8403np1 proto kernel scope link src 10.10.20.171
````

EspiMCF, 2024-11-25:
Update: I have already ping between both nodes without any lost packet (problem was related with an internal firewall in the docker compose environment denying the echo reply).
Problem now, it is that I have checked with `nmap -sY -p 30000 10.10.20.162` and that port (30000) for SCTP connections are closed and the port 5000 is open: 
```
$nmap -sY -p 5000 10.10.20.162
Starting Nmap 7.80 ( https://nmap.org ) at 2024-11-25 11:28 UTC
Nmap scan report for 10.10.20.162
Host is up (0.000037s latency).
PORT      STATE SERVICE
5000/sctp open  unknown
Nmap done: 1 IP address (1 host up) scanned in 3.74 seconds
```
So, what I need now is to force the E2 service from the srsRAN to attack the port 5000 because when I am changing to that port and run the configuration file, it says:
```
--== srsRAN gNB (commit e73b46182) ==--
--port: Value 5000 not in range [20000 - 40000]
Run with --help for more information.
```
I know that currently is not supported and maybe is easier to change the port in the NearRT-RIC but for the moment, I am not able to do that, so any suggest?

Thanks and regards.

pgawlowicz, 2024-11-25:
you might try to translate the port using iptables.

EspiMCF, 2024-12-02:
Still having same error:
In srsRAN:
```
--== srsRAN gNB (commit e73b46182) ==--

2024-12-02T11:05:16.746042 [GNB     ] [I] Built in Release mode using commit e73b46182 on branch main
2024-12-02T11:05:18.044618 [E2AP    ] [I] Connecting to E2 Agent RIC (10.10.20.162)..
Connecting to NearRT-RIC on 10.10.20.162:36421
2024-12-02T11:05:18.044911 [E2AP    ] [I] E2 connection established
Initializing the Open Fronthaul Interface for sector#0: ul_compr=[BFP,9], dl_compr=[BFP,9], prach_compr=[BFP,9], prach_cp_enabled=true, downlink_broadcast=false
Cell pci=3, bw=50 MHz, 2T2R, dl_arfcn=536020 (n7), dl_freq=2680.1 MHz, dl_ssb_arfcn=531610, ul_freq=2560.1 MHz

N2: Connection to AMF on 10.10.2.10:38412 completed
==== gNB started ===
Type <h> to view help
2024-12-02T11:05:18.106895 [E2      ] [I] Generate RAN function definition for OID: 1.3.6.1.4.1.53148.1.2.2.2
2024-12-02T11:05:18.107000 [E2      ] [I] Generate RAN function definition for OID: 1.3.6.1.4.1.53148.1.1.2.3
2024-12-02T11:05:18.107055 [E2      ] [I] Added RAN function OID 1.3.6.1.4.1.53148.1.2.2.2 to candidate list under RAN Function ID 2
2024-12-02T11:05:18.107056 [E2      ] [I] Added RAN function OID 1.3.6.1.4.1.53148.1.1.2.3 to candidate list under RAN Function ID 3
2024-12-02T11:05:18.107311 [E2-ASN1-PCK] [I] Received PDU of 18 bytes
2024-12-02T11:05:18.107348 [E2      ] [I] Handling E2 PDU of type unsuccessfulOutcome
2024-12-02T11:05:18.107350 [E2      ] [I] E2AP msg, "unsuccessfulOutcome.E2setupFailure", transaction id=0
2024-12-02T11:05:18.107353 [E2      ] [E] E2 Setup procedure failed.
2024-12-02T11:05:18.107361 [E2      ] [I] Transmitting E2 Setup Failure message
```

And in the NearRT-RIC:
```
Dec  2 13:05:13 847725     INFO         E2T_E2AP     E2T_DATA_01 E2node fd 10 sent invalid e2setup
Dec  2 13:05:15 458065     INFO           E2T_IO     E2T_DATA_01 new conn fdtype 2 ip b3140a0affff0000 0000000000000000 port 44982 fd 11
Dec  2 13:05:15 458078     INFO           E2T_IO     E2T_DATA_01 dric/dric_data.c:902 E2node IP table not provisioned
Dec  2 13:05:15 458086     INFO        E2T_E2NIO     E2T_DATA_01 yippee! new e2node connection family 2 ip 0xb3140a0affff0000 0x0000000000000000 port 44982 fd 11
Dec  2 13:05:15 458091     INFO         CLIENT_1     E2T_DATA_01 SCTP status for fd 11 assoc_id 50 state 0x4 local rwnd 16777216 unackdata 0 penddata 0 instrms 2 outstrms 2 frag_point 1432
Dec  2 13:05:15 458094     INFO         CLIENT_1     E2T_DATA_01 SCTP assoc for fd 11 assoc_id 0 maxrxt 10 peer dst 1 peer_rwnd 16777216 local_rwnd 212992 cookie_life 60000
Dec  2 13:05:15 520409    ERROR         E2T_E2AP     E2T_DATA_01 fd 11 E2node Global Id Not Provisioned: {
   gNB {
      global_gNB_ID {
         plmn_id = 0x12f410
         gnb_id {
            gnb-ID = { 32, 0x00 0x00 0x00 0x03 }
         }
      }
      gNB_DU_ID = 1
   }
}
```

pgawlowicz, 2024-12-02:
Could you share e2ap pcap file?

EspiMCF, 2024-12-02:
[e2interface.zip](https://github.com/user-attachments/files/17978215/e2interface.zip)

pgawlowicz, 2024-12-02:
The E2 Setup Request looks correct, but still the RIC sends Setup Failure. 
What does the following log line mean?
```
Dec  2 13:05:15 458078     INFO           E2T_IO     E2T_DATA_01 dric/dric_data.c:902 E2node IP table not provisioned
```

Maybe you need to preconfigure a whitelist of E2Nodes?

pgawlowicz, 2024-12-16:
@EspiMCF any update on this issue?

EspiMCF, 2024-12-16:
Sorry, I haven't replied because I don't know exactly the meaning of ‘E2node IP table not provisioned’ and I have asked but no response from the NearRT-RIC provider so, the status is still the same...
I have already done everything that is required from the NearRT-RIC provider to accept the connection, but it seems that it still does not accept, so I am not sure what I should do next

pgawlowicz, 2024-12-16:
Please try the following:
1. Change KPM module RAN_FUNC_ID to 1 [(here)](https://github.com/srsran/srsRAN_Project/blob/main/lib/e2/e2sm/e2sm_kpm/e2sm_kpm_asn1_packer.cpp#L32C1-L32C62) and recompile the code.

2. Enable only e2sm_kpm in gnb config:
```
e2:
  enable_du_e2: true                # Enable DU E2 agent (one for each DU instance)
  enable_cu_cp_e2: false            # Enables the CU E2 agent for CU-CP
  enable_cu_up_e2: false            # Enables the CU E2 agent for CU-UP
  e2sm_kpm_enabled: true            # Enable KPM service module
  e2sm_rc_enabled: false             # Enable RC service module
```
4. Try to connect to RIC.

EspiMCF, 2024-12-16:
I had a problem with that because I can not add this lines due to the commit version:
```
enable_cu_cp_e2: false            # Enables the CU E2 agent for CU-CP
  enable_cu_up_e2: false            # Enables the CU E2 agent for CU-UP
```
The output says:
```
INI was not able to parse e2.enable_cu_cp_e2
```
But I have re-compiled and executed with the RAN_FUNC_ID to 1 and this section for e2:
```
e2:
  enable_du_e2: true                # Enable DU E2 agent (one for each DU instance)
#  enable_cu_cp_e2: false            # Enables the CU E2 agent for CU-CP
#  enable_cu_up_e2: false            # Enables the CU E2 agent for CU-UP
  e2sm_kpm_enabled: true            # Enable KPM service module
  e2sm_rc_enabled: false             # Enable RC service module
  addr: 10.10.20.162
  bind_addr: 10.10.20.179
  port: 36421
```
And still having same output and this is the capture file I got: 
[e2interface_ran_func_id_1.zip](https://github.com/user-attachments/files/18152820/e2interface_ran_func_id_1.zip)

pgawlowicz, 2024-12-16:
could you update your gnb to the newest version? and try again?

The ASN1 printable string in the E2 Setup Request is wrong in the version you use currently:

It should be: https://github.com/srsran/srsRAN_Project/blob/main/lib/e2/common/e2ap_asn1_helpers.h#L117C5-L117C118