srsran/srsRAN_Project #943: gnb_zmq.yaml failed to bind SCTP socket to 10.53.1.1:0. Cause: Cannot assign requested address

mcmtwjyy, 2024-11-21:
I didn't find a solution that really worked

## Issue Description ##
I follow the srsRAN_Project file requirements, using USRP B200, based on 
sudo./gnb -c gnb_zmq.yam
l directive want to be able to run.
But here's the problem
[image]

Failed to bind SCTP socket to 10.53.1.1:0. Cause：Cannot assign requested address
srsRAN ERROR: Failed to create SCTP gateway client.

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
[What happens instead e.g. error message]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

pgawlowicz, 2024-11-21:
Did you start the dockerized version of Open5GS? if not, then most probably you do not have an interface with `ip=10.53.1.1`,

mcmtwjyy, 2024-11-22:
thank you! I will try again to verify.

closed 2024-11-25 by pgawlowicz